RinCat/RTL88x2BU-Linux-Driver #179: Strange things (lost upload) with TP-Link Archer T4U Plus

dipol0, 2023-02-17:
Hi. Sorri for my english ))

I have TP-Link Archer T4U Plus. Use you driver as most stable then iam think, but... . I have SAMBA shares, FTP...
And linux and Win clients. When i try upload/download big file (~100Mb and more....) over SMB/FTP connecton...hmm... wifi on Archer T4U freezed/sleep... after few time connection restored (i see webmin panel on machine with TL-Link T4U) but file transmission lost. Same speedtest_cli sometimes fails... (https://www.speedtest.net/apps/cli)

HW:
1. Router - _Microtik RB962UiGS-5HacT2HnT rev. r2 with RouterOS v7.8rc2_
2. MAIN PC with connected _TP-Link Archer T4U Plus_ - MoBo - _ASUS PRIME A320I-K_
3. OS: _Ubuntu 22.04. Kernel 5.15.94 with UKSM patch - https://github.com/dolohow/uksm/tree/master/v5.x_
4. lsusb out:
`Bus 002 Device 003: ID 2357:0115 TP-Link Archer T4U ver.3`
5. I can connect TL-Link Archer T4U to 5GHz wifi on Mikrotik router load 88x2bu module with optins:
`options 88x2bu rtw_drv_log_level=5 rtw_led_ctrl=1 rtw_switch_usb_mode=1 rtw_power_mgnt=0 rtw_wireless_mode=84 rtw_vht_enable=1 rtw_beaform_cap=11 rtw_qos_opt_enable=1 rtw_channel=48`
6. Internal mobo wire NIC i dont use and dont start it from /etc/network/interfaces (i disable netplan) - T4U a single connected over USB3 port on mobo.
7. Use _wpa_supplicant_ (v.2.10)

in **dmesg** out i see many entries:
`[45136.801978] RTW: [HALMAC][TRACE]chksum disable
[45136.801980] RTW: [HALMAC][TRACE]get_usb_bulkout_id_88xx ===>
[45136.801981] RTW: [HALMAC][TRACE]get_usb_bulkout_id_88xx <===
[45136.805291] RTW: [HALMAC][TRACE]chksum disable
[45136.805295] RTW: [HALMAC][TRACE]get_usb_bulkout_id_88xx ===>
[45136.805297] RTW: [HALMAC][TRACE]get_usb_bulkout_id_88xx <===
[45136.805391] RTW: [HALMAC][TRACE]chksum disable
[45136.805393] RTW: [HALMAC][TRACE]get_usb_bulkout_id_88xx ===>
[45136.805394] RTW: [HALMAC][TRACE]get_usb_bulkout_id_88xx <===
[45136.805446] RTW: [HALMAC][TRACE]chksum disable
[45136.805448] RTW: [HALMAC][TRACE]get_usb_bulkout_id_88xx ===>
[45136.805449] RTW: [HALMAC][TRACE]get_usb_bulkout_id_88xx <===
`
sometimes:

`[45136.167969] RTW: [HALMAC][TRACE]get_usb_bulkout_id_88xx <===
[45136.167998] RTW: [HALMAC][TRACE]chksum disable
[45136.168001] RTW: [HALMAC][TRACE]get_usb_bulkout_id_88xx ===>
[45136.168004] RTW: [HALMAC][TRACE]get_usb_bulkout_id_88xx <===
[45136.186610] RTW: unknown vendor specific information element ignored (vendor OUI 00:0c:42 len=42)
[45136.186618] RTW: Unknown Broadcom information element ignored (type=52 len=26)
[45136.188980] RTW: xmit_status_check REG_TXDMA_STATUS:0x00000401
[45136.190464] RTW: [HALMAC][TRACE]cfg_usb_rx_agg_88xx ===>
[45136.196971] RTW: [HALMAC][TRACE]cfg_usb_rx_agg_88xx <===
[45136.200180] RTW: rtl8822b_fillh2ccmd(wlx3460f95c4249): id=0x42 buf= 0x00 0x00 0x34 0x00 0x00 0x00 0x04
[45136.264330] RTW: [HALMAC][TRACE]chksum disable
[45136.264345] RTW: [HALMAC][TRACE]get_usb_bulkout_id_88xx ===>
[45136.264349] RTW: [HALMAC][TRACE]get_usb_bulkout_id_88xx <===
[45136.279802] RTW: unknown vendor specific information element ignored (vendor OUI 00:0c:42 len=42)
[45136.279811] RTW: Unknown Broadcom information element ignored (type=52 len=26)
[45136.392428] RTW: unknown vendor specific information element ignored (vendor OUI 00:0c:42 len=42)
[45136.392457] RTW: Unknown Broadcom information element ignored (type=52 len=26)
[45136.484378] RTW: unknown vendor specific information element ignored (vendor OUI 00:0c:42 len=42)
[45136.484402] RTW: Unknown Broadcom information element ignored (type=52 len=26)
[45136.597204] RTW: unknown vendor specific information element ignored (vendor OUI 00:0c:42 len=42)
[45136.597229] RTW: Unknown Broadcom information element ignored (type=52 len=26)
[45136.689163] RTW: unknown vendor specific information element ignored (vendor OUI 00:0c:42 len=42)
[45136.689171] RTW: Unknown Broadcom information element ignored (type=52 len=26)
[45136.738080] RTW: [HALMAC][TRACE]chksum disable
[45136.738087] RTW: [HALMAC][TRACE]get_usb_bulkout_id_88xx ===>
[45136.738090] RTW: [HALMAC][TRACE]get_usb_bulkout_id_88xx <===
[45136.738112] RTW: [HALMAC][TRACE]chksum disable
[45136.738116] RTW: [HALMAC][TRACE]get_usb_bulkout_id_88xx ===>
`
Sometimes after fail connection not restore automatically and i run:
`systemctl restart networking` for restore  wifi connection or little more hard solution with reload 88x2bu module

FILE - [syslog.zip](https://mega.nz/file/691BDTxR#ZN_zxI0oY4ynbSO_RtCnj9EuLu9AYIGzjSaa_N40iy4)

If you need wpa_supplicant log and mikrotik log - wrtie me - i send it for you.

This iisue happens on Linux and Win clients (i try use connect SMB share as network drive or use FTP client - Filezilla...).

May be you can see problem and fins solution. Thnx.

RinCat, 2023-02-17:
Hi, I don't see any error here. But can you try to disable wifi background scan?

`RTW: cfg80211_rtw_scan (wlx3460f95c4249) : scan deny`

https://wiki.archlinux.org/title/NetworkManager#Regular_network_disconnects,_latency_and_lost_packets_(WiFi)

dipol0, 2023-02-17:
@RinCat 
I not profi but i dont use GUI on this machine. And my 5GHz network is hidden. I fix BSSID in wpa_supplicant.conf

******************
ctrl_interface_group=0
ctrl_interface=/var/run/wpa_supplicant
update_config=1
ap_scan=1
country=<my_counrty>
network={
    ssid="MYNET-5G"
    bssid=<mac_5ghz_on_router>
    scan_ssid=1
    key_mgmt=WPA-PSK
    pairwise=CCMP
    proto=RSN
    group=CCMP
    psk=<my_psk>
 }
******************
Or may be i need do something else?

RinCat, 2023-02-17:
You can wait and see if it happened again.

dipol0, 2023-02-18:
> You can wait and see if it happened again.

Yes. It repeat every time when i try upload/download big file to/from machime with Archer T4U over SMB/FTP... to/from Linux/Win clients. I try use enother driver (https://github.com/morrownr/88x2bu-20210702) but have same issue, but you driver look more stable.

May be logs my router and wpa_supplicant tell more for you?

RinCat, 2023-02-18:
Maybe you can try move the Archer T4U to a different USB port?
It can also be caused by insufficient power supply.

dipol0, 2023-02-18:
> Maybe you can try move the Archer T4U to a different USB port? It can also be caused by insufficient power supply.

))) Iam read about this in another issue but not can try it now. I onnect Archer to front USB but without another connected dev like as mouse/keyboard etc...

I try play connect Archer to back side ports tomorrow and write about result.

Thnx.

dipol0, 2023-02-18:
> Maybe you can try move the Archer T4U to a different USB port? It can also be caused by insufficient power supply.

Hi. I connect Archer to mobo backside USB port but not have positive effect. For exampe i move/copy from my local PC to remote (with Archer) file have 3.5Gb, uploading stop and fail ~40% and Archer lost connection with router and try recconect.

I check /sys/class/net/wlx3460f95c4249/statistics and found after try big file move have dropped pakets in

/sys/class/net/wlx3460f95c4249/statistic/rx_dropped  - ~80+ value
and
/sys/class/net/wlx3460f95c4249/statistic/tx_dropped - ~2..4 value

parameters with "error" - clear = 0

You driver have option for try change tx/rx buffer size, tso.... ? ethtool cant change this parameters ((

I try use driver from another maintainer (https://github.com/fastoe/RTL8812BU , v5.13.1 ) but have same result/

nikallian, 2023-02-18:
No offense to the guys here, they do great work maintaining the driver which I used when I bought my T4U but in the process I found several issues especially with hostapd so I tested a bunch of other drivers and finally I found one which in my case appeared to have almost no issues. 
You can give it a try. 

https://github.com/ivanovborislav/rtl88x2bu

dipol0, 2023-02-20:
@nikallian tnkx

I try few drivers:
1. https://github.com/RinCat/RTL88x2BU-Linux-Driver
2. https://github.com/ivanovborislav/rtl88x2bu
3. https://github.com/morrownr/88x2bu-20210702

Use - wpa_supplicant.

But all this drivers with default settings disconnect me from PC with Archer T4U if i try upload/download big size files ( from 500...800mb and more).

_Experimental i found and get more stability if use module option **rtw_pwrtrim_enable=1** with RinCat driver. I dont know what is it, but it works form me. At now i can upload/download files with size 10Gb (more i dont try)._

+ )) i found:

1. In RinCat driver option **rtw_led_ctrl=0** - dont turn LED off on my Archer.
2. If i set wireless 5GHz on my Mikrotik router in 5GH-only-AC and set driver option **rtw_wireless_mode=64** and try connect - not have connection and register station on router (all thre driver - RinCat, ivanovborislav, morrown). If use  **rtw_wireless_mode=84** and set in router 5GHz-A/N/AC - all work.

At now i set in router 5GHz-A/N/AC, Freq=5240, Frequency Mode = superchannel. Start RinCat module with options:

`options 88x2bu rtw_drv_log_level=1 rtw_led_ctrl=1 rtw_switch_usb_mode=1 rtw_power_mgnt=0 rtw_wireless_mode=84 rtw_dfs_region_domain=0 rtw_country_code=MY rtw_vht_enable=1 rtw_ht_enable=1 rtw_beaform_cap=11 rtw_qos_opt_enable=1 rtw_channel=48 rtw_pwrtrim_enable=1 rtw_ampdu_enable=1 rtw_ch_switch_offload=1 rtw_gro_en=1`

Test (IP 192.168.0.50 - PC with TP-Link Archer T4U Plus / AC1300):
**********************
root@pc:/var/log# iperf3 -c 192.168.0.50 -4
Connecting to host 192.168.0.50, port 5201
[  5] local 192.168.0.2 port 55236 connected to 192.168.0.50 port 5201
[ ID] Interval           Transfer     Bitrate         Retr  Cwnd
[  5]   0.00-1.00   sec   112 MBytes   943 Mbits/sec  755    283 KBytes       
[  5]   1.00-2.00   sec   110 MBytes   921 Mbits/sec  856    269 KBytes       
[  5]   2.00-3.00   sec   107 MBytes   900 Mbits/sec  908    260 KBytes       
[  5]   3.00-4.00   sec   111 MBytes   927 Mbits/sec  747    255 KBytes       
[  5]   4.00-5.00   sec   110 MBytes   921 Mbits/sec  916    252 KBytes       
[  5]   5.00-6.00   sec   109 MBytes   912 Mbits/sec  1314    256 KBytes       
[  5]   6.00-7.00   sec   110 MBytes   923 Mbits/sec  1105    262 KBytes       
[  5]   7.00-8.00   sec   108 MBytes   902 Mbits/sec  1408    283 KBytes       
[  5]   8.00-9.00   sec   109 MBytes   912 Mbits/sec  1251    266 KBytes       
[  5]   9.00-10.00  sec   105 MBytes   881 Mbits/sec  1908   5.66 KBytes       

[ ID] Interval           Transfer     Bitrate         Retr
[  5]   0.00-10.00  sec  1.06 GBytes   914 Mbits/sec  11168             sender
[  5]   0.00-10.04  sec  1.06 GBytes   908 Mbits/sec                  receiver

iperf Done.
root@pc:/var/log# iperf3 -c 192.168.0.50 -4 -R
Connecting to host 192.168.0.50, port 5201
Reverse mode, remote host 192.168.0.50 is sending
[  5] local 192.168.0.2 port 42636 connected to 192.168.0.50 port 5201
[ ID] Interval           Transfer     Bitrate
[  5]   0.00-1.00   sec   106 MBytes   889 Mbits/sec                  
[  5]   1.00-2.00   sec   107 MBytes   901 Mbits/sec                  
[  5]   2.00-3.00   sec   106 MBytes   885 Mbits/sec                  
[  5]   3.00-4.00   sec   105 MBytes   881 Mbits/sec                  
[  5]   4.00-5.00   sec   106 MBytes   885 Mbits/sec                  
[  5]   5.00-6.00   sec   106 MBytes   893 Mbits/sec                  
[  5]   6.00-7.00   sec   108 MBytes   904 Mbits/sec                  
[  5]   7.00-8.00   sec   106 MBytes   891 Mbits/sec                  
[  5]   8.00-9.00   sec   108 MBytes   904 Mbits/sec                  
[  5]   9.00-10.00  sec   104 MBytes   870 Mbits/sec                  

[ ID] Interval           Transfer     Bitrate         Retr
[  5]   0.00-10.04  sec  1.04 GBytes   889 Mbits/sec  597             sender
[  5]   0.00-10.00  sec  1.04 GBytes   890 Mbits/sec                  receiver

iperf Done.
**********************

**Worked!!!**

I see too many retries. I play with **tcp_congestion_control** but not. May be it normal situation for WiFI - noises...etc.

I think we can close this issue but may be RinCat be improve driver quality, and stability :+1:

dipol0, 2023-02-21:
Nope.  It happens again... ((

dipol0, 2023-02-22:
I try kernel 5.17.9 - no good result.

Stop on big size file upload to PC with Archer T4U Plus (sometimes and download from too). I try FTP/SAMBA
Low upload speed max 40MB/s

[rtw.log](https://github.com/RinCat/RTL88x2BU-Linux-Driver/files/10805496/rtw.log)
[mikrotik-main.log](https://github.com/RinCat/RTL88x2BU-Linux-Driver/files/10805499/mikrotik-main.log)

Check MAC Archer T4U - 34:60:F9:5C:42:49
Router Mikrotik 5GHz wireless MAC - 08:55:31:CF:7C:B7

@RinCat, may be log give more for you. Thnx.

dubhater, 2024-11-09:
Solution: rtw_switch_usb_mode=0.

By the way @dipol0 how can you get 900 Mbps? This chip can't go that fast.