RinCat/RTL88x2BU-Linux-Driver #180: Trying to eliminate warnings as errors in make

flyingpreacher, 2023-02-23:
Here is the error message I keep getting when trying to compile on EuroLinux (compatible with RHEL 9):
cc1: some warnings being treated as errors
make[2]: *** [scripts/Makefile.build:295: /home/junior/RTL88x2BU-Linux-Driver/os_dep/linux/rtw_proc.o] Error 1
make[1]: *** [Makefile:1915: /home/junior/RTL88x2BU-Linux-Driver] Error 2
make[1]: Leaving directory '/usr/src/kernels/5.14.0-162.12.1.el9_1.x86_64'
make: *** [Makefile:2455: modules] Error 2

I have tried adding this to the makefile but it doesn't work:
EXTRA_CFLAGS += -Wno-error

What could I be doing wrong? 
Thanks!

RinCat, 2023-02-23:
Hi, please post full logs.

flyingpreacher, 2023-02-23:
Sorry. What path would the full log be in? I am new to compiling software, but I am trying to learn. Thanks.

RinCat, 2023-02-23:
Try run `make` in the `RTL88x2BU-Linux-Driver`, and copy all the output.

flyingpreacher, 2023-02-23:
make: Warning: File 'Makefile' has modification time 16089 s in the future
make ARCH=x86_64 CROSS_COMPILE= -C /lib/modules/5.14.0-162.12.1.el9_1.x86_64/build M=/home/junior/RTL88x2BU-Linux-Driver  modules
make[1]: Entering directory '/usr/src/kernels/5.14.0-162.12.1.el9_1.x86_64'
make[2]: Warning: File '/home/junior/RTL88x2BU-Linux-Driver/Makefile' has modification time 16088 s in the future
  CC [M]  /home/junior/RTL88x2BU-Linux-Driver/core/rtw_cmd.o
  CC [M]  /home/junior/RTL88x2BU-Linux-Driver/core/rtw_security.o
  CC [M]  /home/junior/RTL88x2BU-Linux-Driver/core/rtw_debug.o
  CC [M]  /home/junior/RTL88x2BU-Linux-Driver/core/rtw_io.o
  CC [M]  /home/junior/RTL88x2BU-Linux-Driver/core/rtw_ioctl_query.o
  CC [M]  /home/junior/RTL88x2BU-Linux-Driver/core/rtw_ioctl_set.o
  CC [M]  /home/junior/RTL88x2BU-Linux-Driver/core/rtw_ieee80211.o
  CC [M]  /home/junior/RTL88x2BU-Linux-Driver/core/rtw_mlme.o
  CC [M]  /home/junior/RTL88x2BU-Linux-Driver/core/rtw_mlme_ext.o
/home/junior/RTL88x2BU-Linux-Driver/core/rtw_mlme_ext.c: In function ‘rtw_txpwr_init_regd’:
/home/junior/RTL88x2BU-Linux-Driver/core/rtw_mlme_ext.c:299:20: warning: this statement may fall through [-Wimplicit-fallthrough=]
  299 |                 if (rfctl->regd_name)
      |                    ^
/home/junior/RTL88x2BU-Linux-Driver/core/rtw_mlme_ext.c:302:9: note: here
  302 |         default:
      |         ^~~~~~~
/home/junior/RTL88x2BU-Linux-Driver/core/rtw_mlme_ext.c: In function ‘mgt_dispatcher’:
/home/junior/RTL88x2BU-Linux-Driver/core/rtw_mlme_ext.c:1670:38: warning: this statement may fall through [-Wimplicit-fallthrough=]
 1670 |                         ptable->func = &OnAuthClient;
      |                         ~~~~~~~~~~~~~^~~~~~~~~~~~~~~
/home/junior/RTL88x2BU-Linux-Driver/core/rtw_mlme_ext.c:1672:9: note: here
 1672 |         case WIFI_ASSOCREQ:
      |         ^~~~
  CC [M]  /home/junior/RTL88x2BU-Linux-Driver/core/rtw_mi.o
  CC [M]  /home/junior/RTL88x2BU-Linux-Driver/core/rtw_wlan_util.o
  CC [M]  /home/junior/RTL88x2BU-Linux-Driver/core/rtw_vht.o
  CC [M]  /home/junior/RTL88x2BU-Linux-Driver/core/rtw_pwrctrl.o
  CC [M]  /home/junior/RTL88x2BU-Linux-Driver/core/rtw_rf.o
  CC [M]  /home/junior/RTL88x2BU-Linux-Driver/core/rtw_chplan.o
  CC [M]  /home/junior/RTL88x2BU-Linux-Driver/core/monitor/rtw_radiotap.o
  CC [M]  /home/junior/RTL88x2BU-Linux-Driver/core/rtw_recv.o
  CC [M]  /home/junior/RTL88x2BU-Linux-Driver/core/rtw_sta_mgt.o
  CC [M]  /home/junior/RTL88x2BU-Linux-Driver/core/rtw_ap.o
  CC [M]  /home/junior/RTL88x2BU-Linux-Driver/core/wds/rtw_wds.o
  CC [M]  /home/junior/RTL88x2BU-Linux-Driver/core/mesh/rtw_mesh.o
  CC [M]  /home/junior/RTL88x2BU-Linux-Driver/core/mesh/rtw_mesh_pathtbl.o
  CC [M]  /home/junior/RTL88x2BU-Linux-Driver/core/mesh/rtw_mesh_hwmp.o
  CC [M]  /home/junior/RTL88x2BU-Linux-Driver/core/rtw_xmit.o
  CC [M]  /home/junior/RTL88x2BU-Linux-Driver/core/rtw_p2p.o
  CC [M]  /home/junior/RTL88x2BU-Linux-Driver/core/rtw_rson.o
  CC [M]  /home/junior/RTL88x2BU-Linux-Driver/core/rtw_tdls.o
  CC [M]  /home/junior/RTL88x2BU-Linux-Driver/core/rtw_br_ext.o
  CC [M]  /home/junior/RTL88x2BU-Linux-Driver/core/rtw_iol.o
  CC [M]  /home/junior/RTL88x2BU-Linux-Driver/core/rtw_sreset.o
  CC [M]  /home/junior/RTL88x2BU-Linux-Driver/core/rtw_btcoex_wifionly.o
  CC [M]  /home/junior/RTL88x2BU-Linux-Driver/core/rtw_btcoex.o
  CC [M]  /home/junior/RTL88x2BU-Linux-Driver/core/rtw_beamforming.o
  CC [M]  /home/junior/RTL88x2BU-Linux-Driver/core/rtw_odm.o
  CC [M]  /home/junior/RTL88x2BU-Linux-Driver/core/rtw_rm.o
  CC [M]  /home/junior/RTL88x2BU-Linux-Driver/core/rtw_rm_fsm.o
  CC [M]  /home/junior/RTL88x2BU-Linux-Driver/core/rtw_ft.o
  CC [M]  /home/junior/RTL88x2BU-Linux-Driver/core/rtw_wnm.o
  CC [M]  /home/junior/RTL88x2BU-Linux-Driver/core/rtw_mbo.o
  CC [M]  /home/junior/RTL88x2BU-Linux-Driver/core/rtw_rm_util.o
  CC [M]  /home/junior/RTL88x2BU-Linux-Driver/core/efuse/rtw_efuse.o
  CC [M]  /home/junior/RTL88x2BU-Linux-Driver/core/rtw_roch.o
  CC [M]  /home/junior/RTL88x2BU-Linux-Driver/core/crypto/aes-internal.o
  CC [M]  /home/junior/RTL88x2BU-Linux-Driver/core/crypto/aes-internal-enc.o
  CC [M]  /home/junior/RTL88x2BU-Linux-Driver/core/crypto/aes-gcm.o
  CC [M]  /home/junior/RTL88x2BU-Linux-Driver/core/crypto/aes-ccm.o
  CC [M]  /home/junior/RTL88x2BU-Linux-Driver/core/crypto/aes-omac1.o
  CC [M]  /home/junior/RTL88x2BU-Linux-Driver/core/crypto/ccmp.o
  CC [M]  /home/junior/RTL88x2BU-Linux-Driver/core/crypto/gcmp.o
  CC [M]  /home/junior/RTL88x2BU-Linux-Driver/core/crypto/aes-siv.o
  CC [M]  /home/junior/RTL88x2BU-Linux-Driver/core/crypto/aes-ctr.o
  CC [M]  /home/junior/RTL88x2BU-Linux-Driver/core/crypto/sha256-internal.o
[... 11 lines cut ...]
  CC [M]  /home/junior/RTL88x2BU-Linux-Driver/os_dep/linux/recv_linux.o
  CC [M]  /home/junior/RTL88x2BU-Linux-Driver/os_dep/linux/ioctl_cfg80211.o
/home/junior/RTL88x2BU-Linux-Driver/os_dep/linux/ioctl_cfg80211.c: In function ‘rtw_get_chbwoff_from_cfg80211_chan_def’:
/home/junior/RTL88x2BU-Linux-Driver/os_dep/linux/ioctl_cfg80211.c:6828:21: warning: this statement may fall through [-Wimplicit-fallthrough=]
 6828 |                 *ht = 0;
      |                 ~~~~^~~
/home/junior/RTL88x2BU-Linux-Driver/os_dep/linux/ioctl_cfg80211.c:6830:9: note: here
 6830 |         case NL80211_CHAN_WIDTH_20:
      |         ^~~~
  CC [M]  /home/junior/RTL88x2BU-Linux-Driver/os_dep/linux/rtw_cfgvendor.o
  CC [M]  /home/junior/RTL88x2BU-Linux-Driver/os_dep/linux/wifi_regd.o
[... cut ...]
/home/junior/RTL88x2BU-Linux-Driver/os_dep/linux/rtw_proc.c:41:25: note: in definition of macro ‘pde_data’
   41 | #define pde_data(inode) PDE_DATA(inode)
      |                         ^~~~~~~~
cc1: some warnings being treated as errors
make[2]: *** [scripts/Makefile.build:295: /home/junior/RTL88x2BU-Linux-Driver/os_dep/linux/rtw_proc.o] Error 1
make[1]: *** [Makefile:1915: /home/junior/RTL88x2BU-Linux-Driver] Error 2
make[1]: Leaving directory '/usr/src/kernels/5.14.0-162.12.1.el9_1.x86_64'
make: *** [Makefile:2455: modules] Error 2

RinCat, 2023-02-23:
This is not a warning. Redhat back-ported some new kennel path to the old one.
Try this https://github.com/RinCat/RTL88x2BU-Linux-Driver/issues/168

flyingpreacher, 2023-02-23:
Ok. I uncommented lines 37-42 and I changed the kernel to 5.14. Here is the newlog:

make ARCH=x86_64 CROSS_COMPILE= -C /lib/modules/5.14.0-162.12.1.el9_1.x86_64/build M=/home/junior/RTL88x2BU-Linux-Driver  modules
make[1]: Entering directory '/usr/src/kernels/5.14.0-162.12.1.el9_1.x86_64'
  CC [M]  /home/junior/RTL88x2BU-Linux-Driver/core/rtw_cmd.o
  CC [M]  /home/junior/RTL88x2BU-Linux-Driver/core/rtw_security.o
  CC [M]  /home/junior/RTL88x2BU-Linux-Driver/core/rtw_debug.o
  CC [M]  /home/junior/RTL88x2BU-Linux-Driver/core/rtw_io.o
  CC [M]  /home/junior/RTL88x2BU-Linux-Driver/core/rtw_ioctl_query.o
  CC [M]  /home/junior/RTL88x2BU-Linux-Driver/core/rtw_ioctl_set.o
  CC [M]  /home/junior/RTL88x2BU-Linux-Driver/core/rtw_ieee80211.o
  CC [M]  /home/junior/RTL88x2BU-Linux-Driver/core/rtw_mlme.o
  CC [M]  /home/junior/RTL88x2BU-Linux-Driver/core/rtw_mlme_ext.o
/home/junior/RTL88x2BU-Linux-Driver/core/rtw_mlme_ext.c: In function ‘rtw_txpwr_init_regd’:
/home/junior/RTL88x2BU-Linux-Driver/core/rtw_mlme_ext.c:299:20: warning: this statement may fall through [-Wimplicit-fallthrough=]
  299 |                 if (rfctl->regd_name)
      |                    ^
/home/junior/RTL88x2BU-Linux-Driver/core/rtw_mlme_ext.c:302:9: note: here
  302 |         default:
      |         ^~~~~~~
/home/junior/RTL88x2BU-Linux-Driver/core/rtw_mlme_ext.c: In function ‘mgt_dispatcher’:
/home/junior/RTL88x2BU-Linux-Driver/core/rtw_mlme_ext.c:1670:38: warning: this statement may fall through [-Wimplicit-fallthrough=]
 1670 |                         ptable->func = &OnAuthClient;
      |                         ~~~~~~~~~~~~~^~~~~~~~~~~~~~~
/home/junior/RTL88x2BU-Linux-Driver/core/rtw_mlme_ext.c:1672:9: note: here
 1672 |         case WIFI_ASSOCREQ:
      |         ^~~~
  CC [M]  /home/junior/RTL88x2BU-Linux-Driver/core/rtw_mi.o
  CC [M]  /home/junior/RTL88x2BU-Linux-Driver/core/rtw_wlan_util.o
  CC [M]  /home/junior/RTL88x2BU-Linux-Driver/core/rtw_vht.o
  CC [M]  /home/junior/RTL88x2BU-Linux-Driver/core/rtw_pwrctrl.o
  CC [M]  /home/junior/RTL88x2BU-Linux-Driver/core/rtw_rf.o
  CC [M]  /home/junior/RTL88x2BU-Linux-Driver/core/rtw_chplan.o
  CC [M]  /home/junior/RTL88x2BU-Linux-Driver/core/monitor/rtw_radiotap.o
  CC [M]  /home/junior/RTL88x2BU-Linux-Driver/core/rtw_recv.o
  CC [M]  /home/junior/RTL88x2BU-Linux-Driver/core/rtw_sta_mgt.o
  CC [M]  /home/junior/RTL88x2BU-Linux-Driver/core/rtw_ap.o
  CC [M]  /home/junior/RTL88x2BU-Linux-Driver/core/wds/rtw_wds.o
  CC [M]  /home/junior/RTL88x2BU-Linux-Driver/core/mesh/rtw_mesh.o
  CC [M]  /home/junior/RTL88x2BU-Linux-Driver/core/mesh/rtw_mesh_pathtbl.o
  CC [M]  /home/junior/RTL88x2BU-Linux-Driver/core/mesh/rtw_mesh_hwmp.o
  CC [M]  /home/junior/RTL88x2BU-Linux-Driver/core/rtw_xmit.o
  CC [M]  /home/junior/RTL88x2BU-Linux-Driver/core/rtw_p2p.o
  CC [M]  /home/junior/RTL88x2BU-Linux-Driver/core/rtw_rson.o
  CC [M]  /home/junior/RTL88x2BU-Linux-Driver/core/rtw_tdls.o
  CC [M]  /home/junior/RTL88x2BU-Linux-Driver/core/rtw_br_ext.o
  CC [M]  /home/junior/RTL88x2BU-Linux-Driver/core/rtw_iol.o
  CC [M]  /home/junior/RTL88x2BU-Linux-Driver/core/rtw_sreset.o
  CC [M]  /home/junior/RTL88x2BU-Linux-Driver/core/rtw_btcoex_wifionly.o
  CC [M]  /home/junior/RTL88x2BU-Linux-Driver/core/rtw_btcoex.o
  CC [M]  /home/junior/RTL88x2BU-Linux-Driver/core/rtw_beamforming.o
  CC [M]  /home/junior/RTL88x2BU-Linux-Driver/core/rtw_odm.o
  CC [M]  /home/junior/RTL88x2BU-Linux-Driver/core/rtw_rm.o
  CC [M]  /home/junior/RTL88x2BU-Linux-Driver/core/rtw_rm_fsm.o
  CC [M]  /home/junior/RTL88x2BU-Linux-Driver/core/rtw_ft.o
  CC [M]  /home/junior/RTL88x2BU-Linux-Driver/core/rtw_wnm.o
  CC [M]  /home/junior/RTL88x2BU-Linux-Driver/core/rtw_mbo.o
  CC [M]  /home/junior/RTL88x2BU-Linux-Driver/core/rtw_rm_util.o
  CC [M]  /home/junior/RTL88x2BU-Linux-Driver/core/efuse/rtw_efuse.o
  CC [M]  /home/junior/RTL88x2BU-Linux-Driver/core/rtw_roch.o
  CC [M]  /home/junior/RTL88x2BU-Linux-Driver/core/crypto/aes-internal.o
  CC [M]  /home/junior/RTL88x2BU-Linux-Driver/core/crypto/aes-internal-enc.o
  CC [M]  /home/junior/RTL88x2BU-Linux-Driver/core/crypto/aes-gcm.o
  CC [M]  /home/junior/RTL88x2BU-Linux-Driver/core/crypto/aes-ccm.o
  CC [M]  /home/junior/RTL88x2BU-Linux-Driver/core/crypto/aes-omac1.o
  CC [M]  /home/junior/RTL88x2BU-Linux-Driver/core/crypto/ccmp.o
  CC [M]  /home/junior/RTL88x2BU-Linux-Driver/core/crypto/gcmp.o
  CC [M]  /home/junior/RTL88x2BU-Linux-Driver/core/crypto/aes-siv.o
  CC [M]  /home/junior/RTL88x2BU-Linux-Driver/core/crypto/aes-ctr.o
  CC [M]  /home/junior/RTL88x2BU-Linux-Driver/core/crypto/sha256-internal.o
[... 23 lines cut ...]
  CC [M]  /home/junior/RTL88x2BU-Linux-Driver/os_dep/linux/rtw_proc.o
/home/junior/RTL88x2BU-Linux-Driver/os_dep/linux/rtw_proc.c:37:1: error: expected identifier or ‘(’ before ‘if’
   37 | if (LINUX_VERSION_CODE < KERNEL_VERSION(3, 10, 0))
      | ^~
/home/junior/RTL88x2BU-Linux-Driver/os_dep/linux/rtw_proc.c: In function ‘rtw_drv_proc_init’:
/home/junior/RTL88x2BU-Linux-Driver/os_dep/linux/rtw_proc.c:333:20: error: implicit declaration of function ‘rtw_proc_create_dir’; did you mean ‘rtw_proc_create_entry’? [-Werror=implicit-function-declaration]
  333 |         rtw_proc = rtw_proc_create_dir(RTW_PROC_NAME, get_proc_net, NULL);
      |                    ^~~~~~~~~~~~~~~~~~~
      |                    rtw_proc_create_entry
[... cut ...]
/home/junior/RTL88x2BU-Linux-Driver/os_dep/linux/rtw_proc.c:6200:17: warning: assignment to ‘struct proc_dir_entry *’ from ‘int’ makes pointer from integer without a cast [-Wint-conversion]
 6200 |         dir_dev = rtw_proc_create_dir(dev->name, drv_proc, dev);
      |                 ^
cc1: some warnings being treated as errors
make[2]: *** [scripts/Makefile.build:295: /home/junior/RTL88x2BU-Linux-Driver/os_dep/linux/rtw_proc.o] Error 1
make[1]: *** [Makefile:1915: /home/junior/RTL88x2BU-Linux-Driver] Error 2
make[1]: Leaving directory '/usr/src/kernels/5.14.0-162.12.1.el9_1.x86_64'
make: *** [Makefile:2455: modules] Error 2

RinCat, 2023-02-23:
NO, you do not uncomment it. They are not comment.

just find that code block and only change the `KERNEL_VERSION(5, 17, 0))` to `KERNEL_VERSION(5, 14, 0)) `

flyingpreacher, 2023-02-23:
Ok. That worked! 

Thank you for being so patient with me and helping me through this.

RinCat, 2023-02-23:
You're welcome

closed 2023-02-23 by RinCat