RinCat/RTL88x2BU-Linux-Driver #181: Memory Leak and Errors (Kernel 5.17)

chrisvollorenew, 2023-03-03:
Hello, 

It looks like I am seeing a memory leak for this driver and other warnings for the kernel. Please advise. 

[    9.805937] RTW: module init start
[    9.812787] RTW: rtl88x2bu v5.13.1-20-gbd7c7eb9d.20210702_COEX20210316-18317b7b
[    9.823611] RTW: rtl88x2bu BT-Coex version = COEX20210316-18317b7b
[    9.823982] RTW: [HALMAC]55772M
               HALMAC_MAJOR_VER = 1
               HALMAC_PROTOTYPE_VER = 6
               HALMAC_MINOR_VER = 6
               HALMAC_PATCH_VER = 23
[    9.842372] systemd-journald[127]: Received client request to flush runtime journal.
[    9.879059] RTW: ERROR [HALMAC][ERR]Dump efuse in suspend
[   10.521225] RTW: HW EFUSE
[   10.528593] RTW: 0x000: 29 81 00 3C  09 00 21 00  B6 04 64 04  00 00 A3 00  
[   10.538908] RTW: 0x010: 28 28 28 28  28 28 29 29  29 29 29 02  00 00 FF FF  
[   10.549237] RTW: 0x020: FF FF 29 29  29 29 29 29  29 29 29 29  2A 2A 2A 2A  
[   10.559589] RTW: 0x030: 02 00 FF FF  00 FF 10 00  FF FF 2E 2E  2E 2E 2E 2E  
[   10.559661] RTW: 0x040: 2F 2F 2F 2F  2F 02 00 00  FF FF FF FF  28 28 28 28  
[   10.559708] RTW: 0x050: 28 28 28 28  28 28 2A 2A  2A 2A 02 00  FF FF 00 FF  
[   10.559754] RTW: 0x060: 10 00 FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF  
[   10.559802] RTW: 0x070: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF  
[   10.559846] RTW: 0x080: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF  
[   10.559915] RTW: 0x090: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF  
[   10.559962] RTW: 0x0A0: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF  
[   10.560007] RTW: 0x0B0: FF FF FF FF  FF FF FF FF  7F 10 1B 00  FF FF FF FF  
[   10.560052] RTW: 0x0C0: FF 21 00 10  00 00 00 00  00 FF 03 FF  FF FF FF FF  
[   10.560096] RTW: 0x0D0: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF  
[   10.560141] RTW: 0x0E0: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF  
[   10.560186] RTW: 0x0F0: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF  
[   10.560231] RTW: 0x100: DA 0B 2C B8  E7 67 82 40  24 B2 A9 E6  19 09 03 52  
[   10.560279] RTW: 0x110: 65 61 6C 74  65 6B 0E 03  38 30 32 2E  31 31 61 63  
[   10.560324] RTW: 0x120: 20 4E 49 43  08 03 31 32  33 34 35 36  FF FF FF FF  
[   10.560369] RTW: 0x130: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF  
[   10.560415] RTW: 0x140: 33 0A 1F 01  00 00 21 0F  FF FF FF FF  FF FF FF FF  
[   10.560460] RTW: 0x150: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF  
[   10.560505] RTW: 0x160: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF  
[   10.560550] RTW: 0x170: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF  
[   10.560595] RTW: 0x180: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF  
[   10.560639] RTW: 0x190: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF  
[   10.560684] RTW: 0x1A0: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF  
[   10.560730] RTW: 0x1B0: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF  
[   10.560774] RTW: 0x1C0: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF  
[   10.560819] RTW: 0x1D0: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF  
[   10.560864] RTW: 0x1E0: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF  
[   10.560909] RTW: 0x1F0: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF  
[   10.560954] RTW: 0x200: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF  
[   10.560998] RTW: 0x210: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF  
[   10.561042] RTW: 0x220: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF  
[   10.561087] RTW: 0x230: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF  
[   10.561133] RTW: 0x240: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF  
[   10.561180] RTW: 0x250: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF  
[   10.561225] RTW: 0x260: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF  
[   10.561269] RTW: 0x270: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF  
[   10.561314] RTW: 0x280: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF  
[   10.561359] RTW: 0x290: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF  
[   10.561404] RTW: 0x2A0: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF  
[   10.561449] RTW: 0x2B0: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF  
[   10.561494] RTW: 0x2C0: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF  
[   10.561539] RTW: 0x2D0: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF  
[   10.561584] RTW: 0x2E0: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF  
[   10.561676] RTW: 0x2F0: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF  
[   10.561744] RTW: hal_com_config_channel_plan chplan:0x7F
[   10.561770] RTW: ERROR [HALMAC][ERR]Dump efuse in suspend
[   10.592913] RTW: [HALMAC][ALWAYS]shall R reg twice!!
[   10.755555] RTW: WARN Invalid rate 0x0 in MRateToHwRate
[   11.087686] RTW: WARN Invalid rate 0x0 in MRateToHwRate
[   11.094040] RTW: WARN Invalid rate 0x0 in MRateToHwRate
[   11.104831] RTW: WARN Invalid rate 0x0 in MRateToHwRate
[... 46 lines cut ...]
[   23.759998]  ehci_work.part.0 from ehci_irq+0x200/0x360
[   23.765447]  ehci_irq from usb_hcd_irq+0x24/0x38
[   23.770230]  usb_hcd_irq from __handle_irq_event_percpu+0x5c/0x124
[   23.776550]  __handle_irq_event_percpu from handle_irq_event+0x44/0x8c
[   23.783188]  handle_irq_event from handle_fasteoi_irq+0x98/0x18c
[   23.789297]  handle_fasteoi_irq from generic_handle_domain_irq+0x40/0x84
[   23.796109]  generic_handle_domain_irq from gic_handle_irq+0x74/0x88
[   23.802576]  gic_handle_irq from generic_handle_arch_irq+0x34/0x44
[   23.808863]  generic_handle_arch_irq from __irq_usr+0x60/0x80
[   23.814707] Exception stack(0xc454dfb0 to 0xc454dff8)
[   23.819844] dfa0:                                     00000001 a8ae1c9c 00000000 00000000
[   23.828148] dfc0: bedc99f0 bedc62d4 bedc7214 bedc6290 00000000 bedc89f0 00000002 bedc9e08
[   23.836451] dfe0: 00000001 bedc6128 bedc62d4 b675fa80 000f0030 ffffffff
[   23.843168] ---[ end trace 0000000000000000 ]---
[   24.116075] RTW: WARN _beamforming_leave: 00:00:00:00:00:00 is neither beamforming ee or er!!
[   24.125039] RTW: ERROR Free disconnecting network of scanned_queue failed due to pwlan == NULL

[   24.326725] RTW: assoc success
[   35.047911] vcc-lcd: disabling
[  134.615000] systemd-journald[127]: Failed to send WATCHDOG=1 notification message: Connection refused
[  204.615006] systemd-journald[127]: Failed to send WATCHDOG=1 notification message: Transport endpoint is not connected
[  272.567988] mmc1: card aaaa removed
[  280.417555] systemd-journald[127]: Failed to send WATCHDOG=1 notification message: Transport endpoint is not connected
[  384.615002] systemd-journald[127]: Failed to send WATCHDOG=1 notification message: Transport endpoint is not connected
[  494.614622] systemd-journald[127]: Failed to send WATCHDOG=1 notification message: Transport endpoint is not connected

RinCat, 2023-03-06:
I don't see any memory leak. The error is `RTW: ERROR rtl8822b_c2h_handler_no_io: C2H(9) enqueue FAIL!` which seems a I/O error.

chrisvollorenew, 2023-03-06:
Memory leak won't show up in the logs, but I do appear to be running out of memory - I suspect this module. Is there any way I can check for memory leak easily?

RinCat, 2023-03-06:
Check this if your kernel enabled https://docs.kernel.org/dev-tools/kmemleak.html

chrisvollorenew, 2023-03-06:
Thank you - question: Is there plans for this driver to go mainline?

RinCat, 2023-03-06:
No, the mainline will be the RTW88 driver, which already in mainline but not very stable.