RinCat/RTL88x2BU-Linux-Driver #182: Linux 6.2 with 8822BU overrides RTL88x2BU module

ogasser, 2023-03-05:
The Linux kernel 6.2 has (finally) [added support for Realtek 8822BU](https://lwn.net/Articles/917733/). Unfortunately, this newly added in-tree module has a lower networking performance, at least on my system: 30 Mbps with 8822BU compared to 100 Mpbs (which is the maximum of my ISP contract) with RTL88x2BU.

After I added the following lines to `/etc/modprobe.d/blacklist.conf` Linux again loads the more performant RTL88x2BU module:

```
blacklist rtw88_core
blacklist rtw88_8822b
blacklist rtw88_usb
blacklist rtw88_8822bu
```

Maybe this should be added to the main README file @RinCat , in case other users are stumbling across this issue as well?

RinCat, 2023-03-06:
This warning already in the README for 10 months. 
I don't know if we want to blacklist `rtw88_core` and `rtw88_usb`, since other devices may use them.

ogasser, 2023-03-06:
Yep, you are right, blacklisting `rtw88_8822bu` is enough and more specific!

I somehow missed the entry in the README, maybe I was searching for "Linux 6.2". It might be useful to mention that explicitly, as it has found its way into the mainline kernel with this release.