RinCat/RTL88x2BU-Linux-Driver #183: New Arch Linux kernel causing strange issues with SMB and SSH (linux-6.2.5.arch1-1)

RAIACAD0564, 2023-03-13:
Ever since I updated my system to Linux kernel 6.2.5, I've been experiencing a strange issue specifically with the WiFi adapter that uses this driver. Everything with the internet functions as normal on the PC, but if I try to connect to it through my SMB or SSH setup from another device, it will not be able to connect. I've tried with my other WiFi adapter and it worked with no problem, same if I have both connected at the same time for some odd reason (Connecting using specifically the IP of the WiFi adapter that uses this driver).

I think it's down to this driver and the kernel as SSH and SMB works fine on other adapters. I did update networkmanager about two days ago but didn't experience this issue until today once I installed the kernel update, and that's the only thing that updated today that seems like it would cause any issue.

Hopefully a kernel update or something else will fix this in the future itself, but I thought I'd mention it if there is a way to fix it in the driver or if others are experiencing this issue. 

And also, before anyone asks the RTW88 8822bu USB driver is already blacklisted.

Here is the pastebin logs:
networkmanager: https://pastebin.com/uGki9FsP
Kernel Logs (demsg): https://pastebin.com/ns3YW2DH

RinCat, 2023-03-14:
Hi, you cannot connect to other SMB/SSH to local system or they cannot connect to your WiFi adapter's system?
Have you checked any firewalls? Like the usb wifi interface name maybe changed.

RAIACAD0564, 2023-03-14:
Other devices cannot connect to my system when I use the WiFi adapter that uses this driver. Other devices will only connect to it if I connect my PC to the WiFi network through it's built in adapter (separate to the one I usually use that uses this driver). I also don't have any firewall installed, so idk how that could cause an issue. As for the interface name changing, idk how that would effect what I'm doing as I just connect to the device through it's IP address directly. 

Also, I have tried resetting my connection settings, manually changing my local IP address to a different one, and unplugging and reconnecting the adapter. None of those fixed the issue.

RinCat, 2023-03-14:
This weird, do you able to ping it?

RAIACAD0564, 2023-03-15:
Okay, I did a system update yesterday and it seems the issue is thankfully fixed. I saw this driver was updated so I don't know if it was that or the fact the Kernel updated to 6.2.6. Looking through what updated, I think these are some possible updates that might have fixed it:

linux-6.2.6.arch1-1
rtl88x2bu-dkms-git 5.13.1.r181.2812290-1
plasma-nm-5.27.3-1
samba-4.18.0-1-x86_64
smbclient-4.18.0-1

Also, while I was still in the older version of the kernel (6.2.5) and this driver, I did try pinging my computer with the adapter from my phone and it wasn't able to ping it before I did this update. I thought I'd leave that info here in case people in the older kernel are still facing the issue. As it is already fixed now and I've already updated, I don't really wanna do any further testing of that.
[image]

RinCat, 2023-03-15:
The update is for linux 6.3. So this probably just the kernel issue.