RinCat/RTL88x2BU-Linux-Driver #184: Error with DKMS install (6.1.18-1-lts)

xtalax, 2023-03-13:
I have installed the package on an Endeavor os install with the AUR package, but am getting the following error:
```
DKMS make.log for rtl88x2bu-5.13.1.r166.3d25a90 for kernel 6.1.18-1-lts (x86_64)
Mon 13 Mar 15:49:18 GMT 2023
make ARCH=x86_64 CROSS_COMPILE= -C /lib/modules/6.1.18-1-lts/build M=/var/lib/dkms/rtl88x2bu/5.13.1.r166.3d25a90/build  modules
  CC [M]  /var/lib/dkms/rtl88x2bu/5.13.1.r166.3d25a90/build/core/rtw_cmd.o
  CC [M]  /var/lib/dkms/rtl88x2bu/5.13.1.r166.3d25a90/build/core/rtw_security.o
  CC [M]  /var/lib/dkms/rtl88x2bu/5.13.1.r166.3d25a90/build/core/rtw_debug.o
  CC [M]  /var/lib/dkms/rtl88x2bu/5.13.1.r166.3d25a90/build/core/rtw_io.o
  CC [M]  /var/lib/dkms/rtl88x2bu/5.13.1.r166.3d25a90/build/core/rtw_ioctl_query.o
  CC [M]  /var/lib/dkms/rtl88x2bu/5.13.1.r166.3d25a90/build/core/rtw_ioctl_set.o
  CC [M]  /var/lib/dkms/rtl88x2bu/5.13.1.r166.3d25a90/build/core/rtw_ieee80211.o
  CC [M]  /var/lib/dkms/rtl88x2bu/5.13.1.r166.3d25a90/build/core/rtw_mlme.o
  CC [M]  /var/lib/dkms/rtl88x2bu/5.13.1.r166.3d25a90/build/core/rtw_mlme_ext.o
  CC [M]  /var/lib/dkms/rtl88x2bu/5.13.1.r166.3d25a90/build/core/rtw_mi.o
  CC [M]  /var/lib/dkms/rtl88x2bu/5.13.1.r166.3d25a90/build/core/rtw_wlan_util.o
  CC [M]  /var/lib/dkms/rtl88x2bu/5.13.1.r166.3d25a90/build/core/rtw_vht.o
  CC [M]  /var/lib/dkms/rtl88x2bu/5.13.1.r166.3d25a90/build/core/rtw_pwrctrl.o
/var/lib/dkms/rtl88x2bu/5.13.1.r166.3d25a90/build/core/rtw_mlme_ext.c: In function ‘rtw_txpwr_init_regd’:
/var/lib/dkms/rtl88x2bu/5.13.1.r166.3d25a90/build/core/rtw_mlme_ext.c:299:20: warning: this statement may fall through [-Wimplicit-fallthrough=]
  299 |                 if (rfctl->regd_name)
      |                    ^
/var/lib/dkms/rtl88x2bu/5.13.1.r166.3d25a90/build/core/rtw_mlme_ext.c:302:9: note: here
  302 |         default:
      |         ^~~~~~~
/var/lib/dkms/rtl88x2bu/5.13.1.r166.3d25a90/build/core/rtw_mlme_ext.c: In function ‘mgt_dispatcher’:
/var/lib/dkms/rtl88x2bu/5.13.1.r166.3d25a90/build/core/rtw_mlme_ext.c:1670:38: warning: this statement may fall through [-Wimplicit-fallthrough=]
 1670 |                         ptable->func = &OnAuthClient;
      |                         ~~~~~~~~~~~~~^~~~~~~~~~~~~~~
/var/lib/dkms/rtl88x2bu/5.13.1.r166.3d25a90/build/core/rtw_mlme_ext.c:1672:9: note: here
 1672 |         case WIFI_ASSOCREQ:
      |         ^~~~
  CC [M]  /var/lib/dkms/rtl88x2bu/5.13.1.r166.3d25a90/build/core/rtw_rf.o
  CC [M]  /var/lib/dkms/rtl88x2bu/5.13.1.r166.3d25a90/build/core/rtw_chplan.o
  CC [M]  /var/lib/dkms/rtl88x2bu/5.13.1.r166.3d25a90/build/core/monitor/rtw_radiotap.o
  CC [M]  /var/lib/dkms/rtl88x2bu/5.13.1.r166.3d25a90/build/core/rtw_recv.o
  CC [M]  /var/lib/dkms/rtl88x2bu/5.13.1.r166.3d25a90/build/core/rtw_sta_mgt.o
  CC [M]  /var/lib/dkms/rtl88x2bu/5.13.1.r166.3d25a90/build/core/rtw_ap.o
  CC [M]  /var/lib/dkms/rtl88x2bu/5.13.1.r166.3d25a90/build/core/wds/rtw_wds.o
  CC [M]  /var/lib/dkms/rtl88x2bu/5.13.1.r166.3d25a90/build/core/mesh/rtw_mesh.o
  CC [M]  /var/lib/dkms/rtl88x2bu/5.13.1.r166.3d25a90/build/core/mesh/rtw_mesh_pathtbl.o
/var/lib/dkms/rtl88x2bu/5.13.1.r166.3d25a90/build/core/rtw_sta_mgt.c: In function ‘rtw_mfree_stainfo’:
/var/lib/dkms/rtl88x2bu/5.13.1.r166.3d25a90/build/core/rtw_sta_mgt.c:391:27: warning: the comparison will always evaluate as ‘true’ for the address of ‘lock’ will never be NULL [-Waddress]
  391 |         if (&(psta->lock) != NULL)
      |                           ^~
In file included from /var/lib/dkms/rtl88x2bu/5.13.1.r166.3d25a90/build/include/drv_types.h:112,
                 from /var/lib/dkms/rtl88x2bu/5.13.1.r166.3d25a90/build/core/rtw_sta_mgt.c:17:
/var/lib/dkms/rtl88x2bu/5.13.1.r166.3d25a90/build/include/sta_info.h:267:17: note: ‘lock’ declared here
  267 |         _lock   lock;
      |                 ^~~~
  CC [M]  /var/lib/dkms/rtl88x2bu/5.13.1.r166.3d25a90/build/core/mesh/rtw_mesh_hwmp.o
  CC [M]  /var/lib/dkms/rtl88x2bu/5.13.1.r166.3d25a90/build/core/rtw_xmit.o
  CC [M]  /var/lib/dkms/rtl88x2bu/5.13.1.r166.3d25a90/build/core/rtw_p2p.o
  CC [M]  /var/lib/dkms/rtl88x2bu/5.13.1.r166.3d25a90/build/core/rtw_rson.o
  CC [M]  /var/lib/dkms/rtl88x2bu/5.13.1.r166.3d25a90/build/core/rtw_tdls.o
  CC [M]  /var/lib/dkms/rtl88x2bu/5.13.1.r166.3d25a90/build/core/rtw_br_ext.o
  CC [M]  /var/lib/dkms/rtl88x2bu/5.13.1.r166.3d25a90/build/core/rtw_iol.o
  CC [M]  /var/lib/dkms/rtl88x2bu/5.13.1.r166.3d25a90/build/core/rtw_sreset.o
  CC [M]  /var/lib/dkms/rtl88x2bu/5.13.1.r166.3d25a90/build/core/rtw_btcoex_wifionly.o
  CC [M]  /var/lib/dkms/rtl88x2bu/5.13.1.r166.3d25a90/build/core/rtw_btcoex.o
  CC [M]  /var/lib/dkms/rtl88x2bu/5.13.1.r166.3d25a90/build/core/rtw_beamforming.o
  CC [M]  /var/lib/dkms/rtl88x2bu/5.13.1.r166.3d25a90/build/core/rtw_odm.o
  CC [M]  /var/lib/dkms/rtl88x2bu/5.13.1.r166.3d25a90/build/core/rtw_rm.o
  CC [M]  /var/lib/dkms/rtl88x2bu/5.13.1.r166.3d25a90/build/core/rtw_rm_fsm.o
  CC [M]  /var/lib/dkms/rtl88x2bu/5.13.1.r166.3d25a90/build/core/rtw_ft.o
  CC [M]  /var/lib/dkms/rtl88x2bu/5.13.1.r166.3d25a90/build/core/rtw_wnm.o
  CC [M]  /var/lib/dkms/rtl88x2bu/5.13.1.r166.3d25a90/build/core/rtw_mbo.o
  CC [M]  /var/lib/dkms/rtl88x2bu/5.13.1.r166.3d25a90/build/core/rtw_rm_util.o
  CC [M]  /var/lib/dkms/rtl88x2bu/5.13.1.r166.3d25a90/build/core/efuse/rtw_efuse.o
  CC [M]  /var/lib/dkms/rtl88x2bu/5.13.1.r166.3d25a90/build/core/rtw_roch.o
  CC [M]  /var/lib/dkms/rtl88x2bu/5.13.1.r166.3d25a90/build/core/crypto/aes-internal.o
  CC [M]  /var/lib/dkms/rtl88x2bu/5.13.1.r166.3d25a90/build/core/crypto/aes-internal-enc.o
  CC [M]  /var/lib/dkms/rtl88x2bu/5.13.1.r166.3d25a90/build/core/crypto/aes-gcm.o
  CC [M]  /var/lib/dkms/rtl88x2bu/5.13.1.r166.3d25a90/build/core/crypto/aes-ccm.o
  CC [M]  /var/lib/dkms/rtl88x2bu/5.13.1.r166.3d25a90/build/core/crypto/aes-omac1.o
  CC [M]  /var/lib/dkms/rtl88x2bu/5.13.1.r166.3d25a90/build/core/crypto/ccmp.o
  CC [M]  /var/lib/dkms/rtl88x2bu/5.13.1.r166.3d25a90/build/core/crypto/gcmp.o
  CC [M]  /var/lib/dkms/rtl88x2bu/5.13.1.r166.3d25a90/build/core/crypto/aes-siv.o
  CC [M]  /var/lib/dkms/rtl88x2bu/5.13.1.r166.3d25a90/build/core/crypto/aes-ctr.o
  CC [M]  /var/lib/dkms/rtl88x2bu/5.13.1.r166.3d25a90/build/core/crypto/sha256-internal.o
  CC [M]  /var/lib/dkms/rtl88x2bu/5.13.1.r166.3d25a90/build/core/crypto/sha256.o
  CC [M]  /var/lib/dkms/rtl88x2bu/5.13.1.r166.3d25a90/build/core/crypto/sha256-prf.o
  CC [M]  /var/lib/dkms/rtl88x2bu/5.13.1.r166.3d25a90/build/core/crypto/rtw_crypto_wrap.o
  CC [M]  /var/lib/dkms/rtl88x2bu/5.13.1.r166.3d25a90/build/core/rtw_swcrypto.o
  CC [M]  /var/lib/dkms/rtl88x2bu/5.13.1.r166.3d25a90/build/os_dep/osdep_service.o
  CC [M]  /var/lib/dkms/rtl88x2bu/5.13.1.r166.3d25a90/build/os_dep/linux/os_intfs.o
  CC [M]  /var/lib/dkms/rtl88x2bu/5.13.1.r166.3d25a90/build/os_dep/linux/usb_intf.o
  CC [M]  /var/lib/dkms/rtl88x2bu/5.13.1.r166.3d25a90/build/os_dep/linux/usb_ops_linux.o
  CC [M]  /var/lib/dkms/rtl88x2bu/5.13.1.r166.3d25a90/build/os_dep/linux/ioctl_linux.o
  CC [M]  /var/lib/dkms/rtl88x2bu/5.13.1.r166.3d25a90/build/os_dep/linux/xmit_linux.o
  CC [M]  /var/lib/dkms/rtl88x2bu/5.13.1.r166.3d25a90/build/os_dep/linux/mlme_linux.o
/var/lib/dkms/rtl88x2bu/5.13.1.r166.3d25a90/build/os_dep/osdep_service.c: In function ‘rtw_random32’:
/var/lib/dkms/rtl88x2bu/5.13.1.r166.3d25a90/build/os_dep/osdep_service.c:2919:16: error: implicit declaration of function ‘prandom_u32’; did you mean ‘prandom_u32_max’? [-Werror=implicit-function-declaration]
 2919 |         return prandom_u32();
      |                ^~~~~~~~~~~
      |                prandom_u32_max
/var/lib/dkms/rtl88x2bu/5.13.1.r166.3d25a90/build/os_dep/linux/os_intfs.c: In function ‘rtw_os_ndev_register’:
/var/lib/dkms/rtl88x2bu/5.13.1.r166.3d25a90/build/os_dep/linux/os_intfs.c:2144:9: error: too many arguments to function ‘netif_napi_add’
 2144 |         netif_napi_add(ndev, &adapter->napi, rtw_recv_napi_poll, RTL_NAPI_WEIGHT);
      |         ^~~~~~~~~~~~~~
In file included from /var/lib/dkms/rtl88x2bu/5.13.1.r166.3d25a90/build/include/osdep_service_linux.h:31,
                 from /var/lib/dkms/rtl88x2bu/5.13.1.r166.3d25a90/build/include/osdep_service.h:54,
                 from /var/lib/dkms/rtl88x2bu/5.13.1.r166.3d25a90/build/include/drv_types.h:27,
                 from /var/lib/dkms/rtl88x2bu/5.13.1.r166.3d25a90/build/os_dep/linux/os_intfs.c:17:
./include/linux/netdevice.h:2569:1: note: declared here
 2569 | netif_napi_add(struct net_device *dev, struct napi_struct *napi,
      | ^~~~~~~~~~~~~~
make[2]: *** [scripts/Makefile.build:250: /var/lib/dkms/rtl88x2bu/5.13.1.r166.3d25a90/build/os_dep/linux/os_intfs.o] Error 1
make[2]: *** Waiting for unfinished jobs....
cc1: some warnings being treated as errors
make[2]: *** [scripts/Makefile.build:250: /var/lib/dkms/rtl88x2bu/5.13.1.r166.3d25a90/build/os_dep/osdep_service.o] Error 1
make[1]: *** [Makefile:2012: /var/lib/dkms/rtl88x2bu/5.13.1.r166.3d25a90/build] Error 2
make: *** [Makefile:2454: modules] Error 2

```

RinCat, 2023-03-14:
Hi, your local package / source code is outdated. Update it will fix this.

xtalax, 2023-03-15:
What is the current version? looks like the AUR package is behind, any idea who maintains?

RinCat, 2023-03-15:
The AUR is git version, you only need to reinstall it to update.

closed 2023-03-15 by xtalax