RinCat/RTL88x2BU-Linux-Driver #185: DKMS installation error with kernel 6.3 RC2

TheJags, 2023-03-14:
DKMS installation is working fine with kernel 6.2.6 but getting these errors with kernel 6.3 RC2:

OS: Ubuntu MATE 22.10

**Make.log**
````
DKMS make.log for rtl88x2bu-git for kernel 6.3.0-060300rc2-generic (x86_64)

Mon Mar 13 11:45:24 PM EDT 2023

make ARCH=x86_64 CROSS_COMPILE= -C /lib/modules/6.3.0-060300rc2-generic/build M=/var/lib/dkms/rtl88x2bu/git/build  modules
make[1]: Entering directory '/usr/src/linux-headers-6.3.0-060300rc2-generic'
warning: the compiler differs from the one used to build the kernel

  The kernel was built by: x86_64-linux-gnu-gcc-12 (Ubuntu 12.2.0-16ubuntu1) 12.2.0
  You are using:           gcc-12 (Ubuntu 12.2.0-16ubuntu1) 12.2.0

  CC [M]  /var/lib/dkms/rtl88x2bu/git/build/core/rtw_cmd.o
  CC [M]  /var/lib/dkms/rtl88x2bu/git/build/core/rtw_security.o
  CC [M]  /var/lib/dkms/rtl88x2bu/git/build/core/rtw_debug.o
  CC [M]  /var/lib/dkms/rtl88x2bu/git/build/core/rtw_io.o
  CC [M]  /var/lib/dkms/rtl88x2bu/git/build/core/rtw_ioctl_query.o
  CC [M]  /var/lib/dkms/rtl88x2bu/git/build/core/rtw_ioctl_set.o
  CC [M]  /var/lib/dkms/rtl88x2bu/git/build/core/rtw_ieee80211.o
  CC [M]  /var/lib/dkms/rtl88x2bu/git/build/core/rtw_mlme.o
  CC [M]  /var/lib/dkms/rtl88x2bu/git/build/core/rtw_mlme_ext.o

/var/lib/dkms/rtl88x2bu/git/build/core/rtw_mlme_ext.c: In function ‘rtw_txpwr_init_regd’:
/var/lib/dkms/rtl88x2bu/git/build/core/rtw_mlme_ext.c:299:20: warning: this statement may fall through [-Wimplicit-fallthrough=]
  299 |                 if (rfctl->regd_name)
      |                    ^
/var/lib/dkms/rtl88x2bu/git/build/core/rtw_mlme_ext.c:302:9: note: here
  302 |         default:
      |         ^~~~~~~
/var/lib/dkms/rtl88x2bu/git/build/core/rtw_mlme_ext.c: In function ‘mgt_dispatcher’:
/var/lib/dkms/rtl88x2bu/git/build/core/rtw_mlme_ext.c:1670:38: warning: this statement may fall through [-Wimplicit-fallthrough=]
 1670 |                         ptable->func = &OnAuthClient;
      |                         ~~~~~~~~~~~~~^~~~~~~~~~~~~~~
/var/lib/dkms/rtl88x2bu/git/build/core/rtw_mlme_ext.c:1672:9: note: here
 1672 |         case WIFI_ASSOCREQ:
      |         ^~~~

  CC [M]  /var/lib/dkms/rtl88x2bu/git/build/core/rtw_mi.o
  CC [M]  /var/lib/dkms/rtl88x2bu/git/build/core/rtw_wlan_util.o
  CC [M]  /var/lib/dkms/rtl88x2bu/git/build/core/rtw_vht.o
  CC [M]  /var/lib/dkms/rtl88x2bu/git/build/core/rtw_pwrctrl.o
  CC [M]  /var/lib/dkms/rtl88x2bu/git/build/core/rtw_rf.o
  CC [M]  /var/lib/dkms/rtl88x2bu/git/build/core/rtw_chplan.o
  CC [M]  /var/lib/dkms/rtl88x2bu/git/build/core/monitor/rtw_radiotap.o
  CC [M]  /var/lib/dkms/rtl88x2bu/git/build/core/rtw_recv.o
  CC [M]  /var/lib/dkms/rtl88x2bu/git/build/core/rtw_sta_mgt.o

/var/lib/dkms/rtl88x2bu/git/build/core/rtw_sta_mgt.c: In function ‘rtw_mfree_stainfo’:
/var/lib/dkms/rtl88x2bu/git/build/core/rtw_sta_mgt.c:391:27: warning: the comparison will always evaluate as ‘true’ for the address of ‘lock’ will never be NULL [-Waddress]
  391 |         if (&(psta->lock) != NULL)
      |                           ^~
In file included from /var/lib/dkms/rtl88x2bu/git/build/include/drv_types.h:112,
                 from /var/lib/dkms/rtl88x2bu/git/build/core/rtw_sta_mgt.c:17:
/var/lib/dkms/rtl88x2bu/git/build/include/sta_info.h:267:17: note: ‘lock’ declared here
  267 |         _lock   lock;
      |                 ^~~~

  CC [M]  /var/lib/dkms/rtl88x2bu/git/build/core/rtw_ap.o
  CC [M]  /var/lib/dkms/rtl88x2bu/git/build/core/wds/rtw_wds.o
  CC [M]  /var/lib/dkms/rtl88x2bu/git/build/core/mesh/rtw_mesh.o
  CC [M]  /var/lib/dkms/rtl88x2bu/git/build/core/mesh/rtw_mesh_pathtbl.o
  CC [M]  /var/lib/dkms/rtl88x2bu/git/build/core/mesh/rtw_mesh_hwmp.o
  CC [M]  /var/lib/dkms/rtl88x2bu/git/build/core/rtw_xmit.o
  CC [M]  /var/lib/dkms/rtl88x2bu/git/build/core/rtw_p2p.o
  CC [M]  /var/lib/dkms/rtl88x2bu/git/build/core/rtw_rson.o
  CC [M]  /var/lib/dkms/rtl88x2bu/git/build/core/rtw_tdls.o
  CC [M]  /var/lib/dkms/rtl88x2bu/git/build/core/rtw_br_ext.o
  CC [M]  /var/lib/dkms/rtl88x2bu/git/build/core/rtw_iol.o
  CC [M]  /var/lib/dkms/rtl88x2bu/git/build/core/rtw_sreset.o
  CC [M]  /var/lib/dkms/rtl88x2bu/git/build/core/rtw_btcoex_wifionly.o
  CC [M]  /var/lib/dkms/rtl88x2bu/git/build/core/rtw_btcoex.o
  CC [M]  /var/lib/dkms/rtl88x2bu/git/build/core/rtw_beamforming.o
  CC [M]  /var/lib/dkms/rtl88x2bu/git/build/core/rtw_odm.o
  CC [M]  /var/lib/dkms/rtl88x2bu/git/build/core/rtw_rm.o
  CC [M]  /var/lib/dkms/rtl88x2bu/git/build/core/rtw_rm_fsm.o
  CC [M]  /var/lib/dkms/rtl88x2bu/git/build/core/rtw_ft.o
````

Many thanks in advance.

RinCat, 2023-03-14:
Hi, the log is not complete. And if that is 6.3 kernel, it probably had some API changes, which I will update this driver later.

TheJags, 2023-03-14:
hi @RinCat thanks for the quick reply. I have completely removed the kernel 6.3 RC2, and reinstalled it again. Here's the error and complete make.log:

**Installation error:**
````
Building module:
Cleaning build area...
'make' -j2 KVER=6.3.0-060300rc2-generic KSRC=/lib/modules/6.3.0-060300rc2-generic/build........................(bad exit status: 2)

ERROR (dkms apport): binary package for rtl88x2bu: git not found
Error! Bad return status for module build on kernel: 6.3.0-060300rc2-generic (x86_64)

Consult /var/lib/dkms/rtl88x2bu/git/build/make.log for more information.
Error! One or more modules failed to install during autoinstall.
Refer to previous errors for more information.
   ...fail!

run-parts: /etc/kernel/header_postinst.d/dkms exited with return code 11
dpkg: error processing package linux-headers-6.3.0-060300rc2-generic (--install):
 installed linux-headers-6.3.0-060300rc2-generic package post-installation script subprocess returned error exit status 1

Setting up linux-modules-6.3.0-060300rc2-generic (6.3.0-060300rc2.202303122031) ...
Setting up linux-image-unsigned-6.3.0-060300rc2-generic (6.3.0-060300rc2.202303122031) ...
I: /boot/vmlinuz is now a symlink to vmlinuz-6.3.0-060300rc2-generic
I: /boot/initrd.img is now a symlink to initrd.img-6.3.0-060300rc2-generic
Processing triggers for linux-image-unsigned-6.3.0-060300rc2-generic (6.3.0-060300rc2.202303122031) ...

/etc/kernel/postinst.d/dkms:
 * dkms: running auto installation service for kernel 6.3.0-060300rc2-generic
Sign command: /usr/bin/kmodsign
Signing key: /var/lib/shim-signed/mok/MOK.priv
Public certificate (MOK): /var/lib/shim-signed/mok/MOK.der

Building module:
Cleaning build area...
'make' -j2 KVER=6.3.0-060300rc2-generic KSRC=/lib/modules/6.3.0-060300rc2-generic/build........................(bad exit status: 2)

ERROR (dkms apport): binary package for rtl88x2bu: git not found
Error! Bad return status for module build on kernel: 6.3.0-060300rc2-generic (x86_64)
Consult /var/lib/dkms/rtl88x2bu/git/build/make.log for more information.
Error! One or more modules failed to install during autoinstall.
Refer to previous errors for more information.
   ...fail!

run-parts: /etc/kernel/postinst.d/dkms exited with return code 11
dpkg: error processing package linux-image-unsigned-6.3.0-060300rc2-generic (--install):
 installed linux-image-unsigned-6.3.0-060300rc2-generic package post-installation script subprocess returned error exit status 1

Errors were encountered while processing:
 linux-headers-6.3.0-060300rc2-generic
 linux-image-unsigned-6.3.0-060300rc2-generic
````

**make.log:**

````
DKMS make.log for rtl88x2bu-git for kernel 6.3.0-060300rc2-generic (x86_64)
Tue Mar 14 06:02:15 PM EDT 2023

make ARCH=x86_64 CROSS_COMPILE= -C /lib/modules/6.3.0-060300rc2-generic/build M=/var/lib/dkms/rtl88x2bu/git/build  modules
make[1]: Entering directory '/usr/src/linux-headers-6.3.0-060300rc2-generic'
warning: the compiler differs from the one used to build the kernel
  The kernel was built by: x86_64-linux-gnu-gcc-12 (Ubuntu 12.2.0-16ubuntu1) 12.2.0
  You are using:           gcc-12 (Ubuntu 12.2.0-16ubuntu1) 12.2.0
  CC [M]  /var/lib/dkms/rtl88x2bu/git/build/core/rtw_cmd.o
  CC [M]  /var/lib/dkms/rtl88x2bu/git/build/core/rtw_security.o
  CC [M]  /var/lib/dkms/rtl88x2bu/git/build/core/rtw_debug.o
  CC [M]  /var/lib/dkms/rtl88x2bu/git/build/core/rtw_io.o
  CC [M]  /var/lib/dkms/rtl88x2bu/git/build/core/rtw_ioctl_query.o
  CC [M]  /var/lib/dkms/rtl88x2bu/git/build/core/rtw_ioctl_set.o
  CC [M]  /var/lib/dkms/rtl88x2bu/git/build/core/rtw_ieee80211.o
[... 104 lines cut ...]

/var/lib/dkms/rtl88x2bu/git/build/os_dep/linux/ioctl_cfg80211.c:481:9: error: too few arguments to function ‘cfg80211_ch_switch_notify’
  481 |         cfg80211_ch_switch_notify(adapter->pnetdev, &chdef, 0);
      |         ^~~~~~~~~~~~~~~~~~~~~~~~~
./include/net/cfg80211.h:8375:6: note: declared here
 8375 | void cfg80211_ch_switch_notify(struct net_device *dev,
      |      ^~~~~~~~~~~~~~~~~~~~~~~~~

/var/lib/dkms/rtl88x2bu/git/build/os_dep/linux/ioctl_cfg80211.c: In function ‘rtw_get_chbwoff_from_cfg80211_chan_def’:

/var/lib/dkms/rtl88x2bu/git/build/os_dep/linux/ioctl_cfg80211.c:6828:21: warning: this statement may fall through [-Wimplicit-fallthrough=]
 6828 |                 *ht = 0;
      |                 ~~~~^~~
/var/lib/dkms/rtl88x2bu/git/build/os_dep/linux/ioctl_cfg80211.c:6830:9: note: here
 6830 |         case NL80211_CHAN_WIDTH_20:
      |         ^~~~

make[2]: *** [scripts/Makefile.build:252: /var/lib/dkms/rtl88x2bu/git/build/os_dep/linux/ioctl_cfg80211.o] Error 1
make[2]: *** Waiting for unfinished jobs....
make[1]: *** [Makefile:2034: /var/lib/dkms/rtl88x2bu/git/build] Error 2
make[1]: Leaving directory '/usr/src/linux-headers-6.3.0-060300rc2-generic'
make: *** [Makefile:2455: modules] Error 2
````

Many thanks.

RinCat, 2023-03-14:
I just updated the driver, try again.

TheJags, 2023-03-15:
@RinCat yes, it's working just fine again. Thank you so much.

closed 2023-03-15 by TheJags