RinCat/RTL88x2BU-Linux-Driver #186: This rtl88x2bu drivers do not work with Kali Linux 6.1.0 kernel

cyberhellcommunity, 2023-03-28:
I tried installing the drivers to my Kali Linux machine running on kernel 6.1.0, I encountered an error at the end of the installation saying the 88x2bu.ko doesn't exist. I've tried reinstalling over and over again, but still same problem, can someone help me.

RinCat, 2023-03-28:
Hi, please post install logs.

cyberhellcommunity, 2023-03-28:
> Hi, please post install logs.
`┌──(kali㉿kali)-[~/RTL88x2BU-Linux-Driver]
└─$ sudo make
/bin/sh: 1: bc: not found
make ARCH=x86_64 CROSS_COMPILE= -C /lib/modules/6.1.0-kali5-amd64/build M=/home/kali/RTL88x2BU-Linux-Driver  modules
make[1]: *** /lib/modules/6.1.0-kali5-amd64/build: No such file or directory.  Stop.
make: *** [Makefile:2455: modules] Error 2
`

thats the result i got

cyberhellcommunity, 2023-03-29:
> Hi, please post install logs.
I've figured it out and fixed all my issues, the adapter is now working fine. Thanks

thesis09, 2023-03-30:
Cyberhellcommunity,RinCat can you tell how to fix this issue as mime is also same

closed 2023-07-14 by cyberhellcommunity