RinCat/RTL88x2BU-Linux-Driver #187: how to fix (fatal error:sys/socket.h: No such file or directory)

wcwcwc2004, 2023-03-29:
my linux heard is Linux yun 5.15.0-handsomekernel+ 
debian  aarch64 system 

and i use gcc is debian  but the kernel was build by ubuntu 
i see i have the file (sys/socket.h) 

root@yun:/usr/src/rtl88x2bu-git# find /usr/include/ -name socket.h
/usr/include/asm-generic/socket.h
/usr/include/aarch64-linux-gnu/asm/socket.h
/usr/include/aarch64-linux-gnu/bits/socket.h
/usr/include/aarch64-linux-gnu/sys/socket.h
/usr/include/linux/socket.h

make ARCH=arm64 CROSS_COMPILE= -C /lib/modules/5.15.0-handsomekernel+/build M=/usr/src/rtl88x2bu-git  modules
make[1]: Entering directory '/usr/src/linux-headers-5.15.0-handsomekernel+'
warning: the compiler differs from the one used to build the kernel
  **The kernel was built by: aarch64-linux-gnu-gcc (Ubuntu 9.4.0-1ubuntu1~20.04) 9.4.0
  You are using:           gcc (Debian 10.2.1-6) 10.2.1 20210110**
  CC [M]  /usr/src/rtl88x2bu-git/core/rtw_cmd.o
In file included from ./include/linux/compat.h:16,
                 from ./arch/arm64/include/asm/ftrace.h:42,
                 from ./include/linux/ftrace.h:22,
                 from ./include/linux/kprobes.h:29,
                 from ./include/linux/kgdb.h:19,
                 from ./arch/arm64/include/asm/cacheflush.h:11,
                 from ./include/linux/highmem.h:12,
                 from ./include/linux/bvec.h:10,
                 from ./include/linux/skbuff.h:17,
                 from ./include/linux/netfilter.h:6,
                 from ./include/linux/netfilter/nf_conntrack_tuple_common.h:6,
                 from ./include/linux/netfilter/nf_conntrack_dccp.h:28,
                 from ./include/net/netns/conntrack.h:11,
                 from ./include/net/net_namespace.h:27,
                 from ./include/linux/netdevice.h:37,
                 from /usr/src/rtl88x2bu-git/include/osdep_service_linux.h:31,
                 from /usr/src/rtl88x2bu-git/include/osdep_service.h:54,
                 from /usr/src/rtl88x2bu-git/include/drv_types.h:27,
                 from /usr/src/rtl88x2bu-git/core/rtw_cmd.c:17:
./include/linux/if.h:28:10: fatal error: sys/socket.h: No such file or directory
   28 | #include <sys/socket.h>   /* for struct sockaddr.  */
      |          ^~~~~~~~~~~~~~
compilation terminated.
make[2]: *** [scripts/Makefile.build:277: /usr/src/rtl88x2bu-git/core/rtw_cmd.o] Error 1
make[1]: *** [Makefile:1868: /usr/src/rtl88x2bu-git] Error 2
make[1]: Leaving directory '/usr/src/linux-headers-5.15.0-handsomekernel+'
make: *** [Makefile:2455: modules] Error 2

wcwcwc2004, 2023-03-29:
i use this  UFI410 
[image]

RinCat, 2023-03-30:
Try `make ARCH=aarch64`?