RinCat/RTL88x2BU-Linux-Driver #188: Installation of RTL88x2BU-Linux-Driver with Secure Boot enabled

eurbah, 2023-04-01:
For the installation of RTL88x2BU-Linux-Driver with Secure Boot enabled, your README.md provides nothing.

So, I have developed the attached shell script :

If the Linux builtin module rtw88_8822bu is not available, or if forced by the -f option,
then only if necessary (using explicit versions),
download, compile and install the 88x2bu Linux module (RTL88x2BU) found on the web.

This script saves the module source in ~/Downloads.
Afterwards, if there is NO internet connection, this script looks for the module source in ~/Downloads, and if found, this script uses it.

If Secure Boot is enabled, this script enforces usage of a MOK (Machine Owner Key) to sign the compiled module.

If DKMS is available, this script tries to use it.

Using the constants defined at the beginning, this script may be adapted for other devices and modules.

In the hope that this script will be useful.

[install-88x2bu.sh.txt](https://github.com/RinCat/RTL88x2BU-Linux-Driver/files/11128367/install-88x2bu.sh.txt)

RinCat, 2023-04-01:
Wow, that's a really long script. I will take a look when I have free time. Thanks!

eurbah, 2023-04-02:
I have attached a new version with fixed and improved logic.
[install-88x2bu.sh.txt](https://github.com/RinCat/RTL88x2BU-Linux-Driver/files/11133165/install-88x2bu.sh.txt)