RinCat/RTL88x2BU-Linux-Driver #189: Error building module (possibly related to clang)

neuromante, 2023-04-03:
My system is Openmandriva ROME on x86_64 architecture.
When I try to build the module (sudo dkms autoinstall) I get an error.
This is the relevant output:

`DKMS make.log for rtl88x2bu-git for kernel 6.2.6-desktop-1omv2390 (x86_64)
lun 3 apr 2023, 11:46:21, CEST
make ARCH=x86_64 CROSS_COMPILE= -C /lib/modules/6.2.6-desktop-1omv2390/build M=/var/lib/dkms/rtl88x2bu/git/build  modules
warning: the compiler differs from the one used to build the kernel
  The kernel was built by: OpenMandriva 15.0.7-3 clang version 15.0.7 (/builddir/build/BUILD/llvm-project-15.0.7.src/clang c5b761db9d28a8cbbafbaa455d57066c5a0e42bc)
  You are using:           gcc (GCC) 12.2.0 20220819 (OpenMandriva)
  CC [M]  /var/lib/dkms/rtl88x2bu/git/build/core/rtw_cmd.o
  CC [M]  /var/lib/dkms/rtl88x2bu/git/build/core/rtw_security.o
  CC [M]  /var/lib/dkms/rtl88x2bu/git/build/core/rtw_debug.o
  CC [M]  /var/lib/dkms/rtl88x2bu/git/build/core/rtw_io.o
gcc: error: unrecognized command-line option ‘-Qunused-arguments’
gcc: error: unrecognized command-line option ‘-Qunused-arguments’
gcc: error: unrecognized command-line option ‘-Qunused-arguments’
gcc: error: unrecognized command-line option ‘-Qunused-arguments’
gcc: error: unrecognized command-line option ‘-mretpoline-external-thunk’
gcc: error: unrecognized command-line option ‘-mretpoline-external-thunk’
gcc: error: unrecognized command-line option ‘-mretpoline-external-thunk’
gcc: error: unrecognized command-line option ‘-mretpoline-external-thunk’
gcc: error: unrecognized command-line option ‘-fsplit-lto-unit’
gcc: error: unrecognized command-line option ‘-fsplit-lto-unit’
gcc: error: unrecognized command-line option ‘-fsplit-lto-unit’
gcc: error: unrecognized command-line option ‘-fsplit-lto-unit’
gcc: error: unrecognized command-line option ‘-Wformat-invalid-specifier’
gcc: error: unrecognized command-line option ‘-Wformat-invalid-specifier’
gcc: error: unrecognized command-line option ‘-Wformat-invalid-specifier’
gcc: error: unrecognized command-line option ‘-Wformat-invalid-specifier’
gcc: error: unrecognized command-line option ‘-Wformat-insufficient-args’
make[2]: *** [scripts/Makefile.build:252: /var/lib/dkms/rtl88x2bu/git/build/core/rtw_cmd.o] Error 1
make[2]: *** Attesa per i processi non terminati....
gcc: error: unrecognized command-line option ‘-Wformat-insufficient-args’
make[2]: *** [scripts/Makefile.build:252: /var/lib/dkms/rtl88x2bu/git/build/core/rtw_security.o] Error 1
gcc: error: unrecognized command-line option ‘-Wformat-insufficient-args’
make[2]: *** [scripts/Makefile.build:252: /var/lib/dkms/rtl88x2bu/git/build/core/rtw_debug.o] Error 1
gcc: error: unrecognized command-line option ‘-Wformat-insufficient-args’
make[2]: *** [scripts/Makefile.build:252: /var/lib/dkms/rtl88x2bu/git/build/core/rtw_io.o] Error 1
make[1]: *** [Makefile:2021: /var/lib/dkms/rtl88x2bu/git/build] Error 2
make: *** [Makefile:2455: modules] Error 2
`

Is there some change I have to make to makefile? 
Thanks

RinCat, 2023-04-03:
> warning: the compiler differs from the one used to build the kernel 
> The kernel was built by: OpenMandriva 15.0.7-3 clang version 15.0.7 (/builddir/build/BUILD/llvm-project-15.0.7.src/clang c5b761db9d28a8cbbafbaa455d57066c5a0e42bc) 
> You are using:           gcc (GCC) 12.2.0 20220819 (OpenMandriva) 

Hi, you must use same compiler and same version as the kernel. And set `make CC=clang ...`

neuromante, 2023-04-03:
Yes BUT if I want to use dkms for automatic module building?

RinCat, 2023-04-03:
Try add `CC = clang` at the top of `Makefile`

neuromante, 2023-04-03:
I tried but this is the output:
`make ARCH=x86_64 CROSS_COMPILE= -C /lib/modules/6.2.6-desktop-1omv2390/build M=/usr/src/rtl88x2bu-git  modules
warning: the compiler differs from the one used to build the kernel
  The kernel was built by: OpenMandriva 15.0.7-3 clang version 15.0.7 (/builddir/build/BUILD/llvm-project-15.0.7.src/clang c5b761db9d28a8cbbafbaa455d57066c5a0e42bc)
  You are using:           gcc (GCC) 12.2.0 20220819 (OpenMandriva)
  CC [M]  /usr/src/rtl88x2bu-git/core/rtw_cmd.o
clang-15: error: unknown argument: '-mpreferred-stack-boundary=3'
clang-15: error: unknown argument: '-mindirect-branch-cs-prefix'
clang-15: warning: optimization flag '-falign-jumps=1' is not supported [-Wignored-optimization-argument]
make[2]: *** [scripts/Makefile.build:252: /usr/src/rtl88x2bu-git/core/rtw_cmd.o] Error 1
make[1]: *** [Makefile:2021: /usr/src/rtl88x2bu-git] Error 2
`
It seems that the compiler will remain the same even with the variable at the top of the Makefile...

neuromante, 2023-04-03:
Update: it compiles only if I type "sudo make CC=clang" in the src directory. Dkms seems to have issues

RinCat, 2023-04-03:
Edit `dkms.conf` and change
`MAKE="'make' -j$(nproc) KVER=${kernelver} KSRC=/lib/modules/${kernelver}/build"`
to
`MAKE="'make' -j$(nproc) CC=clang KVER=${kernelver} KSRC=/lib/modules/${kernelver}/build"`