RinCat/RTL88x2BU-Linux-Driver #193: rate at 270mbit/s connecting to a new wifi6 router

wesleysanjose, 2023-05-06:
bought a new wifi6 router, tenda rx9 pro and my macbook can connect at 866mbit/s. but this rtl88x2bu only connects to 270mbit/s.

even for the old asus rx68u it can connect to 405mbit/s, not sure why??
i followed the instruction to enable the usb to 3.0 but it doens't help on the rate.

wesleysanjose, 2023-05-07:
[image]
top on the list is the one i talked about

RinCat, 2023-05-07:
270mbps sounds like 2.4G. Can you try split your SSID by 2.4 / 5 GHz?

wesleysanjose, 2023-05-09:
I tried, still not working. the card can detect both 2.4G and 5G wifi separately (2 line items in the nmcli d wifi list) even I combine it in the router. also i just bought a intel ax200, strange thing is it also shows the same 270mbps (although my iphone 12pro can connect the wifi at 866mbps). i am wondering if it's because the kernel version is too low (5.4)?

RinCat, 2023-05-09:
5.4 should have WiFi 6 (802.11ax) support. Notice this driver / hardware only support WiFi 5 (802.11ac). But if AX 200 also at 270 mbps, maybe there are some configuration issues in the WiFi router.