RinCat/RTL88x2BU-Linux-Driver #195: Unable to install module for T4U on Ubuntu 22.04

randyman99, 2023-05-09:
A warning to others… I bricked my system trying to install this. Granted, it wasn’t the instructions listed here per se, but those did not work for me. I followed this set as listed on the TP-Link website, copying the instructions into a script and running that. 

sudo apt update

sudo apt install -y dkms git

sudo git clone 'https://github.com/RinCat/RTL88x2BU-Linux-Driver.git' /usr/src/rtl88x2bu-git

sudo sed -i 's/PACKAGE_VERSION="@PKGVER@"/PACKAGE_VERSION="git"/g' /usr/src/rtl88x2bu-git/dkms.conf

sudo dkms add -m rtl88x2bu -v git

sudo dkms autoinstall

sudo reboot

When it rebooted, my T4U still did not work (I had just upgraded to 22.04, Linux 5.15.0-71-generic x86_64). I put pauses after each step in the script and ran it again, and saw there was a problem with the makefile. It said the version of gcc I was using was not the same one that was used to create the module, but it continued to run anyway. I looked at the log file, and there were quite a few errors. I had run “sudo update” as the first instruction in the script, so I didn’t understand why gcc was not the most current. You’ll forgive me if I don’t have the exact text, as my system was bricked and it’s all gone. I had gcc version 4.x.y (Ubuntu 22.04), and the version used to build the module was gcc 4.x.y (Ubuntu1 22.04). They both had the same version/revision number, the only difference being the Ubuntu vs. Ubuntu1. So how do I get the right version? I tried installing it, but apt said I already had the most current version. I uninstalled gcc and then reinstalled it, and that seemingly innocuous act may have been my big mistake. Now the computer complained that distro-watch-(something) was not the most recent. I ignored that (perhaps also a seemingly innocuous act that was a mistake), and tried rerunning the script to build the T4U module, but it again failed. I again got a warning that distro-watch-(something) was not current. I again ignored it and returned to the Tp-Link page to document my actions and what was not working, while I still had all the information. In the middle of typing, the page disappeared. Okay, I’ll write it in a document and copy and paste it. As I was writing the document, the system went into spontaneous reboot.

When the reboot progressed from Grub, Ubuntu never came back. I rebooted into Ubuntu maintenance mode, and selected to repair dkms. I looked at the details beforehand, and it said three distro-watch-(something) packages needed to be repaired. I selected to repair them, and when that finished, it still wouldn’t boot into Ubuntu. I went into maintenance mode again, and it said the same three distro-watch-(something) packages needed to be repaired. I selected to repair them. When it finished, I selected to repair dkms again, and it said the same three packages needed to be repaired. I went into the same loop several times until I was convinced this was going nowhere, i.e., this installation of Ubuntu was bricked.

Luckily, I had installed another installation of Ubuntu 20.04 in another partition set on this computer before I upgraded. I rebooted into that, and everything is fine. I’ve already deleted the offending 22.04 installation, reinstalled Grub, and also removed the 22.04 logical volumes, volume group, and physical volume. I’m good with staying with 20.04 on this computer with the T4U dongle. I was in the process of moving 22.04 to a new computer I had just purchased (with onboard WiFi), and thankfully I had finished the critical steps of that process before this happened.

RinCat, 2023-05-09:
Your system should not broken by remove `gcc`, in fact it is not installed by default. 
And `distro-watch-(something)` not exists in ubuntu packages. I am not sure what was that since google find nothing.

randyman99, 2023-05-10:
@RinCat , I agree that uninstalling and installing gcc should not have broken Ubuntu, and I can't say for sure that was the cause, but that was exactly when my problems started, and they went downhill from there within a matter of minutes.On the other hand, it could have been the faulty compile that stepped on some code it shouldn't have. But why was there that minor difference in the compiler and why should that make a difference? I've used a lot of C compilers over the decades and never had one be so finicky. I also don't know what the distro-watch was about, and that seemed bizarre, but at the Ubuntu maintenance mode level, that's what prevented me from continuing into a normal boot. Sorry I don't have exact notes, as I was copying  and pasting from the terminal window when it crashed.

RinCat, 2023-05-10:
@randyman99 Since different versions of compilers may generate binary with incompatible internal data structures, the use of different versions of compilers is prohibited in linux build scripts. This is not under the control of this driver, because the build uses the script that comes with the linux kernel source code instead of calling the compiler directly.

randyman99, 2023-05-11:
@RinCat, exactly. You make a very good argument for why the user should not be depended upon to compile the source code. I could not agree more.