RinCat/RTL88x2BU-Linux-Driver #197: Recent updates caused compatibility lost for adapter

BananaNomNom, 2023-05-31:
OS: Arch
version: 6.3.5-arch1-1

desc: When the driver was updated recently it removed compatibility for the wifi adapter ALFA (AWUS036ACH). This adapter was working until a recent update made the driver no longer recognize or assign the driver to it seen here. 

usb-devices:
[image]

The driver itself is loaded in dkms and modprobe.

thinking this was a problem of it needing to be added as an id I ran the "echo VID PID > new_id" in the driver's directory where I still failed. dmesg shows this

dmesg: 
```[  974.300211] usb 1-9: USB disconnect, device number 6
[  977.609527] usb 1-9: new high-speed USB device number 7 using xhci_hcd
[  977.769756] usb 1-9: New USB device found, idVendor=0bda, idProduct=8812, bcdDevice= 0.00
[  977.769763] usb 1-9: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[  977.769766] usb 1-9: Product: 802.11n NIC
[  977.769769] usb 1-9: Manufacturer: Realtek
[  977.769771] usb 1-9: SerialNumber: 123456
[  977.769942] ACPI Error: Aborting method \_SB.PCI0.GPP2.PTXH.RHUB.PO13._PLD due to previous error (AE_AML_UNINITIALIZED_ELEMENT) (20221020/psparse-529)
[  977.796867] ACPI Error: Aborting method \_SB.PCI0.GPP2.PTXH.RHUB.PO13._PLD due to previous error (AE_AML_UNINITIALIZED_ELEMENT) (20221020/psparse-529)
[  977.796984] RTW: ### rtw_hal_ops_check - Error : Please hook hal_func.read_chip_version ###
[  977.796989] RTW: ### rtw_hal_ops_check - Error : Please hook hal_func.init_default_value ###
[  977.796992] RTW: ### rtw_hal_ops_check - Error : Please hook hal_func.intf_chip_configure ###
[  977.796995] RTW: ### rtw_hal_ops_check - Error : Please hook hal_func.read_adapter_info ###
[  977.796997] RTW: ### rtw_hal_ops_check - Error : Please hook hal_func.hal_power_on ###
[  977.797000] RTW: ### rtw_hal_ops_check - Error : Please hook hal_func.hal_power_off ###
[  977.797002] RTW: ### rtw_hal_ops_check - Error : Please hook hal_func.hal_init ###
[  977.797004] RTW: ### rtw_hal_ops_check - Error : Please hook hal_func.hal_deinit ###
[  977.797007] RTW: ### rtw_hal_ops_check - Error : Please hook hal_func.init_xmit_priv ###
[  977.797009] RTW: ### rtw_hal_ops_check - Error : Please hook hal_func.free_xmit_priv ###
[  977.797012] RTW: ### rtw_hal_ops_check - Error : Please hook hal_func.hal_xmit ###
[  977.797014] RTW: ### rtw_hal_ops_check - Error : Please hook hal_func.mgnt_xmit ###
[  977.797016] RTW: ### rtw_hal_ops_check - Error : Please hook hal_func.hal_xmitframe_enqueue ###
[  977.797019] RTW: ### rtw_hal_ops_check - Error : Please hook hal_func.init_recv_priv ###
[  977.797021] RTW: ### rtw_hal_ops_check - Error : Please hook hal_func.free_recv_priv ###
[  977.797023] RTW: ### rtw_hal_ops_check - Error : Please hook hal_func.inirp_init ###
[  977.797026] RTW: ### rtw_hal_ops_check - Error : Please hook hal_func.inirp_deinit ###
[  977.797028] RTW: ### rtw_hal_ops_check - Error : Please hook hal_func.dm_init ###
[  977.797030] RTW: ### rtw_hal_ops_check - Error : Please hook hal_func.dm_deinit ###
[  977.797033] RTW: ### rtw_hal_ops_check - Error : Please hook hal_func.hal_dm_watchdog ###
[  977.797035] RTW: ### rtw_hal_ops_check - Error : Please hook hal_func.set_chnl_bw_handler ###
[  977.797037] RTW: ### rtw_hal_ops_check - Error : Please hook hal_func.set_hw_reg_handler ###
[  977.797040] RTW: ### rtw_hal_ops_check - Error : Please hook hal_func.GetHwRegHandler ###
[  977.797042] RTW: ### rtw_hal_ops_check - Error : Please hook hal_func.get_hal_def_var_handler ###
[  977.797044] RTW: ### rtw_hal_ops_check - Error : Please hook hal_func.SetHalDefVarHandler ###
[  977.797046] RTW: ### rtw_hal_ops_check - Error : Please hook hal_func.GetHalODMVarHandler ###
[  977.797049] RTW: ### rtw_hal_ops_check - Error : Please hook hal_func.SetHalODMVarHandler ###
[  977.797051] RTW: ### rtw_hal_ops_check - Error : Please hook hal_func.SetBeaconRelatedRegistersHandler ###
[  977.797053] RTW: ### rtw_hal_ops_check - Error : Please hook hal_func.fill_h2c_cmd ###
[  977.797055] RTW: ### rtw_hal_ops_check - Error : Please hook hal_func.hal_mac_c2h_handler ###
[  977.797057] RTW: ### rtw_hal_ops_check - Error : Please hook hal_func.fill_fake_txdesc ###
[  977.797060] RTW: ### rtw_hal_ops_check - Error : Please hook hal_func.fw_dl ###
[  977.797062] RTW: ### rtw_hal_ops_check - Error : Please hook hal_func.set_tx_power_level_handler ###
[  977.797064] RTW: ### rtw_hal_ops_check - Error : Please hook hal_func.set_tx_power_index_handler ###
[  977.797066] RTW: ### rtw_hal_ops_check - Error : Please hook hal_func.get_tx_power_index_handler ###
[  977.797068] RTW: ### rtw_hal_ops_check - Error : Please hook hal_func.sreset_init_value ###
[  977.797069] RTW: ### rtw_hal_ops_check - Error : Please hook hal_func.sreset_reset_value ###
[  977.797071] RTW: ### rtw_hal_ops_check - Error : Please hook hal_func.silentreset ###
[  977.797072] RTW: ### rtw_hal_ops_check - Error : Please hook hal_func.sreset_xmit_status_check ###
[  977.797074] RTW: ### rtw_hal_ops_check - Error : Please hook hal_func.sreset_linked_status_check ###
[  977.797076] RTW: ### rtw_hal_ops_check - Error : Please hook hal_func.sreset_get_wifi_status ###
[  977.797077] RTW: ### rtw_hal_ops_check - Error : Please hook hal_func.sreset_inprogress ###
[  977.797079] RTW: ### rtw_hal_ops_check - Error : Please hook hal_func.init_mac_register ###
[  977.797080] RTW: ### rtw_hal_ops_check - Error : Please hook hal_func.init_phy ###
```

RinCat, 2023-06-15:
I had no idea what happened here. The USB ID is the default one, so the driver recognize it. 

` ACPI Error: Aborting method \_SB.PCI0.GPP2.PTXH.RHUB.PO13._PLD due to previous error (AE_AML_UNINITIALIZED_ELEMENT) (20221020/psparse-529)`
is probably the issue here. 

https://bbs.archlinux.org/viewtopic.php?id=243991
https://forums.debian.net/viewtopic.php?t=147331

ohan2, 2023-08-02:
I can confirm that this hardware issue exists. I was able to connect to wifi after plugging my RTL8812AU-equipped network adapter into a rear USB 3.0 port on my computer (I had orignally plugged it into a port near the front of the case, thinking this would improve signal reception).