RinCat/RTL88x2BU-Linux-Driver #201: DKMS fails - (custom) kernel 6.5, Fedora

onlyreportingissues, 2023-06-29:
make.log:

DKMS make.log for rtl88x2bu-git for kernel 6.5.0-0.rc0.20230629git3a8a670eeeaa.3.fc38.x86_64 (x86_64)
Do 29. Jun 20:36:45 CEST 2023
make ARCH=x86_64 CROSS_COMPILE= -C /lib/modules/6.5.0-0.rc0.20230629git3a8a670eeeaa.3.fc38.x86_64/build M=/var/lib/dkms/rtl88x2bu/git/build  modules
make[1]: Verzeichnis „/usr/src/kernels/6.5.0-0.rc0.20230629git3a8a670eeeaa.3.fc38.x86_64“ wird betreten
  CC [M]  /var/lib/dkms/rtl88x2bu/git/build/core/rtw_cmd.o
  CC [M]  /var/lib/dkms/rtl88x2bu/git/build/core/rtw_security.o
  CC [M]  /var/lib/dkms/rtl88x2bu/git/build/core/rtw_debug.o
  CC [M]  /var/lib/dkms/rtl88x2bu/git/build/core/rtw_io.o
  CC [M]  /var/lib/dkms/rtl88x2bu/git/build/core/rtw_ioctl_query.o
  CC [M]  /var/lib/dkms/rtl88x2bu/git/build/core/rtw_ioctl_set.o
  CC [M]  /var/lib/dkms/rtl88x2bu/git/build/core/rtw_ieee80211.o
  CC [M]  /var/lib/dkms/rtl88x2bu/git/build/core/rtw_mlme.o
  CC [M]  /var/lib/dkms/rtl88x2bu/git/build/core/rtw_mlme_ext.o
  CC [M]  /var/lib/dkms/rtl88x2bu/git/build/core/rtw_mi.o
  CC [M]  /var/lib/dkms/rtl88x2bu/git/build/core/rtw_wlan_util.o
  CC [M]  /var/lib/dkms/rtl88x2bu/git/build/core/rtw_vht.o
  CC [M]  /var/lib/dkms/rtl88x2bu/git/build/core/rtw_pwrctrl.o
  CC [M]  /var/lib/dkms/rtl88x2bu/git/build/core/rtw_rf.o
  CC [M]  /var/lib/dkms/rtl88x2bu/git/build/core/rtw_chplan.o
  CC [M]  /var/lib/dkms/rtl88x2bu/git/build/core/monitor/rtw_radiotap.o
  CC [M]  /var/lib/dkms/rtl88x2bu/git/build/core/rtw_recv.o
/var/lib/dkms/rtl88x2bu/git/build/core/rtw_mlme_ext.c: In Funktion »rtw_txpwr_init_regd«:
/var/lib/dkms/rtl88x2bu/git/build/core/rtw_mlme_ext.c:299:20: Warnung: diese Anweisung könnte durchfallen [-Wimplicit-fallthrough=]
  299 |                 if (rfctl->regd_name)
      |                    ^
/var/lib/dkms/rtl88x2bu/git/build/core/rtw_mlme_ext.c:302:9: Anmerkung: hier
  302 |         default:
      |         ^~~~~~~
/var/lib/dkms/rtl88x2bu/git/build/core/rtw_mlme_ext.c: In Funktion »mgt_dispatcher«:
/var/lib/dkms/rtl88x2bu/git/build/core/rtw_mlme_ext.c:1670:38: Warnung: diese Anweisung könnte durchfallen [-Wimplicit-fallthrough=]
 1670 |                         ptable->func = &OnAuthClient;
      |                         ~~~~~~~~~~~~~^~~~~~~~~~~~~~~
/var/lib/dkms/rtl88x2bu/git/build/core/rtw_mlme_ext.c:1672:9: Anmerkung: hier
 1672 |         case WIFI_ASSOCREQ:
      |         ^~~~
  CC [M]  /var/lib/dkms/rtl88x2bu/git/build/core/rtw_sta_mgt.o
  CC [M]  /var/lib/dkms/rtl88x2bu/git/build/core/rtw_ap.o
/var/lib/dkms/rtl88x2bu/git/build/core/rtw_mlme_ext.c: In Funktion »rtw_txpwr_init_regd«:
/var/lib/dkms/rtl88x2bu/git/build/core/rtw_mlme_ext.c:201:21: Warnung: »strcmp« liest 1 oder mehr Bytes aus einer Region der Größe 0 [-Wstringop-overread]
  201 |                 if (strcmp(exc->regd_name, regd_str(TXPWR_LMT_NONE)) == 0)
      |                     ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
In Datei, eingebunden von /var/lib/dkms/rtl88x2bu/git/build/include/drv_types.h:62,
                 von /var/lib/dkms/rtl88x2bu/git/build/core/rtw_mlme_ext.c:17:
/var/lib/dkms/rtl88x2bu/git/build/include/rtw_rf.h:267:14: Anmerkung: Quellobjekt »regd_name« der Größe 0
  267 |         char regd_name[0];
      |              ^~~~~~~~~
/var/lib/dkms/rtl88x2bu/git/build/core/rtw_mlme_ext.c:203:26: Warnung: »strcmp« liest 1 oder mehr Bytes aus einer Region der Größe 0 [-Wstringop-overread]
  203 |                 else if (strcmp(exc->regd_name, regd_str(TXPWR_LMT_WW)) == 0)
      |                          ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
/var/lib/dkms/rtl88x2bu/git/build/include/rtw_rf.h:267:14: Anmerkung: Quellobjekt »regd_name« der Größe 0
  267 |         char regd_name[0];
      |              ^~~~~~~~~
  CC [M]  /var/lib/dkms/rtl88x2bu/git/build/core/wds/rtw_wds.o
  CC [M]  /var/lib/dkms/rtl88x2bu/git/build/core/mesh/rtw_mesh.o
  CC [M]  /var/lib/dkms/rtl88x2bu/git/build/core/mesh/rtw_mesh_pathtbl.o
  CC [M]  /var/lib/dkms/rtl88x2bu/git/build/core/mesh/rtw_mesh_hwmp.o
  CC [M]  /var/lib/dkms/rtl88x2bu/git/build/core/rtw_xmit.o
  CC [M]  /var/lib/dkms/rtl88x2bu/git/build/core/rtw_p2p.o
  CC [M]  /var/lib/dkms/rtl88x2bu/git/build/core/rtw_rson.o
  CC [M]  /var/lib/dkms/rtl88x2bu/git/build/core/rtw_tdls.o
/var/lib/dkms/rtl88x2bu/git/build/core/rtw_sta_mgt.c: In Funktion »rtw_mfree_stainfo«:
/var/lib/dkms/rtl88x2bu/git/build/core/rtw_sta_mgt.c:391:27: Warnung: der Vergleich wird immer als »true« ausgewertet, da die Adresse von »lock« niemals NULL sein wird [-Waddress]
  391 |         if (&(psta->lock) != NULL)
      |                           ^~
In Datei, eingebunden von /var/lib/dkms/rtl88x2bu/git/build/include/drv_types.h:112,
                 von /var/lib/dkms/rtl88x2bu/git/build/core/rtw_sta_mgt.c:17:
/var/lib/dkms/rtl88x2bu/git/build/include/sta_info.h:267:17: Anmerkung: »lock« ist hier deklariert
  267 |         _lock   lock;
[... 100 lines cut ...]
  CC [M]  /var/lib/dkms/rtl88x2bu/git/build/os_dep/linux/ioctl_cfg80211.o
  CC [M]  /var/lib/dkms/rtl88x2bu/git/build/os_dep/linux/rtw_cfgvendor.o
  CC [M]  /var/lib/dkms/rtl88x2bu/git/build/os_dep/linux/wifi_regd.o
  CC [M]  /var/lib/dkms/rtl88x2bu/git/build/os_dep/linux/rtw_android.o
  CC [M]  /var/lib/dkms/rtl88x2bu/git/build/os_dep/linux/rtw_proc.o
  CC [M]  /var/lib/dkms/rtl88x2bu/git/build/os_dep/linux/nlrtw.o
  CC [M]  /var/lib/dkms/rtl88x2bu/git/build/os_dep/linux/rtw_rhashtable.o
  CC [M]  /var/lib/dkms/rtl88x2bu/git/build/os_dep/linux/ioctl_mp.o
/var/lib/dkms/rtl88x2bu/git/build/os_dep/linux/wifi_regd.c: In Funktion »rtw_regd_init«:
/var/lib/dkms/rtl88x2bu/git/build/os_dep/linux/wifi_regd.c:409:36: Fehler: »REGULATORY_IGNORE_STALE_KICKOFF« nicht deklariert (erste Verwendung in dieser Funktion)
  409 |         wiphy->regulatory_flags |= REGULATORY_IGNORE_STALE_KICKOFF;
      |                                    ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
/var/lib/dkms/rtl88x2bu/git/build/os_dep/linux/wifi_regd.c:409:36: Anmerkung: jeder nicht deklarierte Bezeichner wird nur einmal für jede Funktion, in der er vorkommt, gemeldet
make[2]: *** [scripts/Makefile.build:252: /var/lib/dkms/rtl88x2bu/git/build/os_dep/linux/wifi_regd.o] Fehler 1
make[2]: *** Es wird auf noch nicht beendete Prozesse gewartet …
/var/lib/dkms/rtl88x2bu/git/build/os_dep/linux/ioctl_cfg80211.c: In Funktion »rtw_get_chbwoff_from_cfg80211_chan_def«:
/var/lib/dkms/rtl88x2bu/git/build/os_dep/linux/ioctl_cfg80211.c:6832:21: Warnung: diese Anweisung könnte durchfallen [-Wimplicit-fallthrough=]
 6832 |                 *ht = 0;
      |                 ~~~~^~~
/var/lib/dkms/rtl88x2bu/git/build/os_dep/linux/ioctl_cfg80211.c:6834:9: Anmerkung: hier
 6834 |         case NL80211_CHAN_WIDTH_20:
      |         ^~~~
make[1]: *** [Makefile:2050: /var/lib/dkms/rtl88x2bu/git/build] Fehler 2
make[1]: Verzeichnis „/usr/src/kernels/6.5.0-0.rc0.20230629git3a8a670eeeaa.3.fc38.x86_64“ wird verlassen
make: *** [Makefile:2470: modules] Fehler 2

RinCat, 2023-06-30:
6.5 just into merge window, please wait for a while.

closed 2023-07-11 by RinCat

onlyreportingissues, 2023-07-11:
It works now, thanks.