RinCat/RTL88x2BU-Linux-Driver #202: Chip id is undefined

Srednii, 2023-07-03:
Hello. Using D-Link DW-185 card, which is in supported devices list.
After install dmesg shows that my device is undefined.

```
[    7.647214] RTW: ERROR [HALMAC][ERR]Chip id is undefined
[    7.647216] RTW: ERROR rtw_halmac_init_adapter: halmac_init_adapter fail!(status=54)
[    7.647219] RTW: ### rtw_hal_ops_check - Error : Please hook hal_func.read_chip_version ###
[    7.647221] RTW: ### rtw_hal_ops_check - Error : Please hook hal_func.init_default_value ###
[    7.647222] RTW: ### rtw_hal_ops_check - Error : Please hook hal_func.intf_chip_configure ###
[    7.647223] RTW: ### rtw_hal_ops_check - Error : Please hook hal_func.read_adapter_info ###
[    7.647224] RTW: ### rtw_hal_ops_check - Error : Please hook hal_func.hal_power_on ###
[    7.647225] RTW: ### rtw_hal_ops_check - Error : Please hook hal_func.hal_power_off ###
[    7.647226] RTW: ### rtw_hal_ops_check - Error : Please hook hal_func.hal_init ###
[    7.647227] RTW: ### rtw_hal_ops_check - Error : Please hook hal_func.hal_deinit ###
[    7.647228] RTW: ### rtw_hal_ops_check - Error : Please hook hal_func.init_xmit_priv ###
[    7.647228] RTW: ### rtw_hal_ops_check - Error : Please hook hal_func.free_xmit_priv ###
[    7.647229] RTW: ### rtw_hal_ops_check - Error : Please hook hal_func.hal_xmit ###
[    7.647230] RTW: ### rtw_hal_ops_check - Error : Please hook hal_func.mgnt_xmit ###
[    7.647231] RTW: ### rtw_hal_ops_check - Error : Please hook hal_func.hal_xmitframe_enqueue ###
[    7.647232] RTW: ### rtw_hal_ops_check - Error : Please hook hal_func.init_recv_priv ###
[    7.647233] RTW: ### rtw_hal_ops_check - Error : Please hook hal_func.free_recv_priv ###
[    7.647234] RTW: ### rtw_hal_ops_check - Error : Please hook hal_func.inirp_init ###
[    7.647234] RTW: ### rtw_hal_ops_check - Error : Please hook hal_func.inirp_deinit ###
[    7.647235] RTW: ### rtw_hal_ops_check - Error : Please hook hal_func.dm_init ###
[    7.647236] RTW: ### rtw_hal_ops_check - Error : Please hook hal_func.dm_deinit ###
[    7.647237] RTW: ### rtw_hal_ops_check - Error : Please hook hal_func.hal_dm_watchdog ###
[    7.647237] RTW: ### rtw_hal_ops_check - Error : Please hook hal_func.set_chnl_bw_handler ###
[    7.647238] RTW: ### rtw_hal_ops_check - Error : Please hook hal_func.set_hw_reg_handler ###
[    7.647239] RTW: ### rtw_hal_ops_check - Error : Please hook hal_func.GetHwRegHandler ###
[    7.647240] RTW: ### rtw_hal_ops_check - Error : Please hook hal_func.get_hal_def_var_handler ###
[    7.647241] RTW: ### rtw_hal_ops_check - Error : Please hook hal_func.SetHalDefVarHandler ###
[    7.647241] RTW: ### rtw_hal_ops_check - Error : Please hook hal_func.GetHalODMVarHandler ###
[    7.647242] RTW: ### rtw_hal_ops_check - Error : Please hook hal_func.SetHalODMVarHandler ###
[    7.647243] RTW: ### rtw_hal_ops_check - Error : Please hook hal_func.SetBeaconRelatedRegistersHandler ###
[    7.647243] RTW: ### rtw_hal_ops_check - Error : Please hook hal_func.fill_h2c_cmd ###
[    7.647244] RTW: ### rtw_hal_ops_check - Error : Please hook hal_func.hal_mac_c2h_handler ###
[    7.647244] RTW: ### rtw_hal_ops_check - Error : Please hook hal_func.fill_fake_txdesc ###
[    7.647245] RTW: ### rtw_hal_ops_check - Error : Please hook hal_func.fw_dl ###
[    7.647246] RTW: ### rtw_hal_ops_check - Error : Please hook hal_func.set_tx_power_level_handler ###
[    7.647246] RTW: ### rtw_hal_ops_check - Error : Please hook hal_func.set_tx_power_index_handler ###
[    7.647247] RTW: ### rtw_hal_ops_check - Error : Please hook hal_func.get_tx_power_index_handler ###
[    7.647248] RTW: ### rtw_hal_ops_check - Error : Please hook hal_func.sreset_init_value ###
[    7.647248] RTW: ### rtw_hal_ops_check - Error : Please hook hal_func.sreset_reset_value ###
[    7.647249] RTW: ### rtw_hal_ops_check - Error : Please hook hal_func.silentreset ###
[    7.647250] RTW: ### rtw_hal_ops_check - Error : Please hook hal_func.sreset_xmit_status_check ###
[    7.647250] RTW: ### rtw_hal_ops_check - Error : Please hook hal_func.sreset_linked_status_check ###
[    7.647251] RTW: ### rtw_hal_ops_check - Error : Please hook hal_func.sreset_get_wifi_status ###
[    7.647252] RTW: ### rtw_hal_ops_check - Error : Please hook hal_func.sreset_inprogress ###
[    7.647252] RTW: ### rtw_hal_ops_check - Error : Please hook hal_func.init_mac_register ###
[    7.647253] RTW: ### rtw_hal_ops_check - Error : Please hook hal_func.init_phy ###
[    7.647280] usbcore: registered new interface driver rtl88x2bu
[    7.647281] RTW: module init ret=0
```

RinCat, 2023-07-03:
Hi, can you try reinsert it or change the USB port? The chip id is different than the USB ID, and it is hardcoded in the hardware.

Srednii, 2023-07-04:
Changed USB port and now it's working. Thank you. Don't know why i didn't that earlier.

closed 2023-07-04 by Srednii