RinCat/RTL88x2BU-Linux-Driver #206: Provide instructions for updating from git and rebuilding for DKMS

olberger, 2023-08-06:
AFAIU, when installed with dkms autoinstall from git following the README instructions, the module is rebuild fine when new kernel are installed/updated. However, there are no instructions for updating the module.

I guess a combination of `git rebase origin/master --autostash` and `dkms build rtl88x2bu/git --force` / `dkms install rtl88x2bu/git --force` would do the trick... but I'm by far no expert.

Hope you can improve the docs for updating, beyond initial install.

Thanks in advance.

MatthewForrester, 2023-08-15:
I would also like advice about this. 

> I guess a combination of `git rebase origin/master --autostash`

Why wouldn't you just use `git pull`?

olberger, 2023-08-17:
Well, as the initial install requires a modification of one of the files (`sed -i 's/PACKAGE_VERSION="@PKGVER@"/PACKAGE_VERSION="git"/g' /usr/src/rtl88x2bu-git/dkms.conf`), git pull would normally complain that a modification is there, and might block. Hence the use of rebase with the autostash option, that will undo and redo the sed.

Hop this helps,

PR #238 (2024-07-27): Provide instructions for updating from git and rebuilding for DKMS
Hope this helps, and closes #206
commit d4425c0b (2024-07-27): Provide instructions for updating from git and rebuilding for DKMS (#238)

Close #206
closed 2024-07-27 by RinCat