RinCat/RTL88x2BU-Linux-Driver #207: [Rocky Linux 8.8]: netif_napi_add error

vitduck, 2023-08-16:
Hi, 

I am trying to compile the driver on Rocky Linux with 4.18 kernel. 

Based on suggestion at #66, I changed to macro to 
```
#if LINUX_VERSION_CODE >= KERNEL_VERSION(4, 18, 0)
```
Compilation then proceeds with the following error instead: 
```
/opt/RTL88x2BU-Linux-Driver/os_dep/linux/os_intfs.c:2153:74: error: 
macro "netif_napi_add" passed 4 arguments, but takes just 3 
netif_napi_add(ndev, &adapter->napi, rtw_recv_napi_poll, RTL_NAPI_WEIGHT); 

/opt/RTL88x2BU-Linux-Driver/os_dep/linux/os_intfs.c:2153:2: error: 
"netif_napi_add" undeclared (first use in this situation); did you men 'netif_napi_del' ?  
netif_napi_add(ndev, &adapter->napi, rtw_recv_napi_poll, RTL_NAPI_WEIGHT); 
```

I understand that upgrading to 5.16 will likely fix the issue, but I am constrained with a clone of  RHEL 8. 

Thanks.

RinCat, 2023-08-16:
You will need to manually check what you kernel ported back patches, change all related error `KERNEL_VERSION` to you current kernel.

And it looks like also included this:

https://github.com/RinCat/RTL88x2BU-Linux-Driver/blob/12cfcd8cd8ec7115158df3d223510435541ddc32/os_dep/linux/os_intfs.c#L2150-L2154

vitduck, 2023-08-19:
Hi @RinCat 

Thanks for commenting. 

After trial and error, I was able to successfully compiler the kernel module. 
For future reference, below are the the macros need changing for RHEL 8 due to kernel back port. 

https://github.com/RinCat/RTL88x2BU-Linux-Driver/blob/12cfcd8cd8ec7115158df3d223510435541ddc32/os_dep/linux/os_intfs.c#L1722-L1726 

https://github.com/RinCat/RTL88x2BU-Linux-Driver/blob/12cfcd8cd8ec7115158df3d223510435541ddc32/os_dep/linux/os_intfs.c#L2150-L2154

https://github.com/RinCat/RTL88x2BU-Linux-Driver/blob/12cfcd8cd8ec7115158df3d223510435541ddc32/os_dep/linux/ioctl_cfg80211.c#L459-L460

https://github.com/RinCat/RTL88x2BU-Linux-Driver/blob/12cfcd8cd8ec7115158df3d223510435541ddc32/os_dep/linux/ioctl_cfg80211.c#L484-L485

https://github.com/RinCat/RTL88x2BU-Linux-Driver/blob/12cfcd8cd8ec7115158df3d223510435541ddc32/os_dep/linux/ioctl_cfg80211.c#L1152-L1154

https://github.com/RinCat/RTL88x2BU-Linux-Driver/blob/12cfcd8cd8ec7115158df3d223510435541ddc32/os_dep/linux/ioctl_cfg80211.c#L2081-L2083

https://github.com/RinCat/RTL88x2BU-Linux-Driver/blob/12cfcd8cd8ec7115158df3d223510435541ddc32/os_dep/linux/ioctl_cfg80211.c#L2272-L2274

https://github.com/RinCat/RTL88x2BU-Linux-Driver/blob/12cfcd8cd8ec7115158df3d223510435541ddc32/os_dep/linux/ioctl_cfg80211.c#L2295-L2297

https://github.com/RinCat/RTL88x2BU-Linux-Driver/blob/12cfcd8cd8ec7115158df3d223510435541ddc32/os_dep/linux/ioctl_cfg80211.c#L2346-L2348

https://github.com/RinCat/RTL88x2BU-Linux-Driver/blob/12cfcd8cd8ec7115158df3d223510435541ddc32/os_dep/linux/ioctl_cfg80211.c#L5407-L5409

https://github.com/RinCat/RTL88x2BU-Linux-Driver/blob/12cfcd8cd8ec7115158df3d223510435541ddc32/os_dep/linux/ioctl_cfg80211.c#L6788-L6790

https://github.com/RinCat/RTL88x2BU-Linux-Driver/blob/12cfcd8cd8ec7115158df3d223510435541ddc32/os_dep/linux/ioctl_cfg80211.c#L8328-L8329

https://github.com/RinCat/RTL88x2BU-Linux-Driver/blob/12cfcd8cd8ec7115158df3d223510435541ddc32/os_dep/linux/ioctl_cfg80211.c#L10510-L10511

https://github.com/RinCat/RTL88x2BU-Linux-Driver/blob/12cfcd8cd8ec7115158df3d223510435541ddc32/os_dep/linux/ioctl_cfg80211.c#L10729-L10731

https://github.com/RinCat/RTL88x2BU-Linux-Driver/blob/12cfcd8cd8ec7115158df3d223510435541ddc32/os_dep/linux/rtw_android.c#L666-L668

The module is loaded with ```modprobe``` 
```
cfg80211              905216  1 88x2bu
```

However, the wireless interface does not show up in ```ifconfig``` 
```
enp4s0: flags=4163<UP,BROADCAST,RUNNING,MULTICAST>  mtu 1500
        inet 192.168.0.9  netmask 255.255.255.0  broadcast 192.168.0.255
        inet6 fe80::aaa1:59ff:fe85:d835  prefixlen 64  scopeid 0x20<link>
        ether a8:a1:59:85:d8:35  txqueuelen 1000  (Ethernet)
        RX packets 749  bytes 58336 (56.9 KiB)
        RX errors 0  dropped 0  overruns 0  frame 0
        TX packets 1633  bytes 1781915 (1.6 MiB)
        TX errors 0  dropped 0 overruns 0  carrier 0  collisions 0

lo: flags=73<UP,LOOPBACK,RUNNING>  mtu 65536
        inet 127.0.0.1  netmask 255.0.0.0
        inet6 ::1  prefixlen 128  scopeid 0x10<host>
        loop  txqueuelen 1000  (Local Loopback)
        RX packets 12  bytes 1020 (1020.0 B)
        RX errors 0  dropped 0  overruns 0  frame 0
        TX packets 12  bytes 1020 (1020.0 B)
        TX errors 0  dropped 0 overruns 0  carrier 0  collisions 0
```
Is there some trivial thing that I miss ? 
Thanks.

RinCat, 2023-08-19:
You can run `dmesg` and see if there is any error.

vitduck, 2023-08-20:
I attached output of ```dmesg``` 
[dmesg.txt](https://github.com/RinCat/RTL88x2BU-Linux-Driver/files/12386928/dmesg.txt)

Please see the call trace at the end of the file.

RinCat, 2023-08-20:
Let me install Rocky Linux 8.8 and check, that one crashed outside of this driver.

RinCat, 2023-08-20:
```
diff --git a/core/monitor/rtw_radiotap.c b/core/monitor/rtw_radiotap.c
index e9ebc7d..e3b2e0b 100644
--- a/core/monitor/rtw_radiotap.c
+++ b/core/monitor/rtw_radiotap.c
@@ -21,7 +21,7 @@
 
 #define CHAN2FREQ(a) ((a < 14) ? (2407+5*a) : (5000+5*a))
 
-#if (LINUX_VERSION_CODE < KERNEL_VERSION(4, 20, 0))
+#if (LINUX_VERSION_CODE < KERNEL_VERSION(4, 18, 0))
 #define IEEE80211_RADIOTAP_ZERO_LEN_PSDU 26
 #define IEEE80211_RADIOTAP_LSIG 27
 #endif
diff --git a/include/osdep_service_linux.h b/include/osdep_service_linux.h
index fca4cf6..2bcf707 100644
--- a/include/osdep_service_linux.h
+++ b/include/osdep_service_linux.h
@@ -56,7 +56,7 @@
 #include <linux/list.h>
 #include <linux/vmalloc.h>
 
-#if (LINUX_VERSION_CODE >= KERNEL_VERSION(5, 4, 0))
+#if (LINUX_VERSION_CODE >= KERNEL_VERSION(4, 18, 0))
 	#include <uapi/linux/sched/types.h>
 #endif
 
diff --git a/os_dep/linux/ioctl_cfg80211.c b/os_dep/linux/ioctl_cfg80211.c
index f7b59a8..b44d319 100644
--- a/os_dep/linux/ioctl_cfg80211.c
+++ b/os_dep/linux/ioctl_cfg80211.c
@@ -456,9 +456,9 @@ u8 rtw_cfg80211_ch_switch_notify(_adapter *adapter, u8 ch, u8 bw, u8 offset,
 	if (started) {
 #if (LINUX_VERSION_CODE >= KERNEL_VERSION(6, 3, 0))
 		cfg80211_ch_switch_started_notify(adapter->pnetdev, &chdef, 0, 0, false, 0);
-#elif (LINUX_VERSION_CODE >= KERNEL_VERSION(6, 1, 0))
+#elif (LINUX_VERSION_CODE >= KERNEL_VERSION(4, 18, 0))
 		cfg80211_ch_switch_started_notify(adapter->pnetdev, &chdef, 0, 0, false);
-#elif (LINUX_VERSION_CODE >= KERNEL_VERSION(5, 11, 0))
+#elif (LINUX_VERSION_CODE >= KERNEL_VERSION(4, 18, 0))
 
 		/* --- cfg80211_ch_switch_started_notfiy() ---
 		 *  A new parameter, bool quiet, is added from Linux kernel v5.11,
@@ -481,7 +481,7 @@ u8 rtw_cfg80211_ch_switch_notify(_adapter *adapter, u8 ch, u8 bw, u8 offset,
 
 #if (LINUX_VERSION_CODE >= KERNEL_VERSION(6, 3, 0))
 	cfg80211_ch_switch_notify(adapter->pnetdev, &chdef, 0, 0);
-#elif (LINUX_VERSION_CODE >= KERNEL_VERSION(5, 19, 2))
+#elif (LINUX_VERSION_CODE >= KERNEL_VERSION(4, 18, 0))
 	cfg80211_ch_switch_notify(adapter->pnetdev, &chdef, 0);
 #else
 	cfg80211_ch_switch_notify(adapter->pnetdev, &chdef);
@@ -749,7 +749,7 @@ static int rtw_cfg80211_sync_iftype(_adapter *adapter)
 
 static u64 rtw_get_systime_us(void)
 {
-#if (LINUX_VERSION_CODE >= KERNEL_VERSION(4, 20, 0))
+#if (LINUX_VERSION_CODE >= KERNEL_VERSION(4, 18, 0))
 	return ktime_to_us(ktime_get_boottime());
 #elif (LINUX_VERSION_CODE >= KERNEL_VERSION(2, 6, 39))
 	struct timespec ts;
@@ -1149,7 +1149,7 @@ check_bss:
 		#endif
 
 		#if LINUX_VERSION_CODE >= KERNEL_VERSION(4, 12, 0)
-		#if LINUX_VERSION_CODE >= KERNEL_VERSION(6, 0, 0)
+		#if LINUX_VERSION_CODE >= KERNEL_VERSION(4, 18, 0)
 		roam_info.links[0].bssid = cur_network->network.MacAddress;
 		#else
 		roam_info.bssid = cur_network->network.MacAddress;
[... 477 lines cut ...]
 			set_fs(KERNEL_DS);
 			#else
 			set_fs(get_ds());
@@ -2619,7 +2615,7 @@ static int storeToFile(const char *path, u8 *buf, u32 sz)
 
 			ret = writeFile(fp, buf, sz);
 
-			#if (LINUX_VERSION_CODE < KERNEL_VERSION(5, 10, 0))
+			#if (LINUX_VERSION_CODE < KERNEL_VERSION(4, 18, 0))
 			set_fs(oldfs);
 			#endif
 			closeFile(fp);
@@ -2918,7 +2914,7 @@ u64 rtw_division64(u64 x, u64 y)
 inline u32 rtw_random32(void)
 {
 #ifdef PLATFORM_LINUX
-#if (LINUX_VERSION_CODE >= KERNEL_VERSION(6, 1, 0))
+#if (LINUX_VERSION_CODE >= KERNEL_VERSION(4, 18, 0))
 	return get_random_u32();
 #elif (LINUX_VERSION_CODE >= KERNEL_VERSION(3, 8, 0))
 	return prandom_u32();

```

I guess you need more than that. This patch works for Rocky 8.8 with kernel 4.18.0-477.10.1.el8_8.x86_64

vitduck, 2023-08-20:
Thanks for working the magic. 

After applying your patch, I am now able to connect to wifi network .
```
$ nmcli dev status 
DEVICE                TYPE      STATE         CONNECTION 
wlp9s0f3u2u3          wifi      connected     sazabi     
p2p-dev-wlp9s0f3u2u3  wifi-p2p  disconnected  --         
enp4s0                ethernet  unavailable   --         
lo                    loopback  unmanaged     --       
```
Please close the issue as you see fit. 
Once again, thanks very much for your time.

closed 2023-08-23 by RinCat