RinCat/RTL88x2BU-Linux-Driver #208: After the upgrade from kernel 5.15 to 6.2 my RTL8812BU does not work properly

rossanoparis, 2023-08-22:
**tp info:** TP-Link Archer T3U [Realtek RTL8812BU]
**System info:** Linux MINT 21.1 6.2.0-26-generic 26~22.04.1-Ubuntu SMP PREEMPT_DYNAMIC x86_64 x86_64 x86_64 GNU/Linux

Last week I've downloaded source code from master and built it
All went good apparently. I did not get any error message during compilation and its installation.

Despite that I have to report two unknown behaviours.
1) Before the upgrade the green LED was working and blinking while connected to the network, after the upgrade the LED is always OFF despite the TPLink is connected.
2) After let say 10min the connection drop down and the system reports 0% of available signal.

harykishan, 2023-08-31:
Hi Hari here. Me too am exactly facing this very same issue with my Tplink Archer T4U wifi adapter. Earlier it was working just fine with my Ubuntu 22.04 LTS. 
Kernel version -  6.2.0-31-generic
Wifi adapter - Tplink Archer T4U ver3
driver installed - this same one Realtek RTL8812BU

Any suggestions/fixes which work are highly appreciated.

rossanoparis, 2023-08-31:
kind @harykishan yesterday evening after an automatic update of "Linux firmware drivers" my problem has gone.
Right now the led is blincking again and the connection is stable.

Perhaps you should find a way to update your "Linux firmware drivers", I'm sorry but I can't help with that.

Koya90, 2023-09-07:
Wifi adapter cudy: Realtek RTL88x2BU
System info: Linux Ubuntu 22.04.3 LTS
Kernel version: 6.2.0-32-generic

I followed the instructions and successfully installed the driver.

Commands:
```
sudo apt update && sudo apt upgrade -y
sudo apt install -y git dkms build-essential
git clone https://github.com/cilynx/rtl88x2bu.git
cd rtl88x2bu
VER=$(sed -n 's/\PACKAGE_VERSION="\(.*\)"/\1/p' dkms.conf)
sudo rsync -rvhP ./ /usr/src/rtl88x2bu-${VER}
sudo dkms add -m rtl88x2bu -v ${VER}
sudo dkms build -m rtl88x2bu -v ${VER}
sudo dkms install -m rtl88x2bu -v ${VER}
sudo modprobe 88x2bu
```

After these steps, the Wi-Fi adapter showed up in the network manager and I could see the signals.
The problem is that I can't connect to any signal.

koya@koya-desktop: nmcli device
DEVICE TYPE STATE CONNECTION
wlxd03745f5093a wifi connected Koya optics
wlxf4a454819cb4 wifi connecting (configuring) Koya optics 1
enp4s0 ethernet unavailable
lo loopback unmanaged

Please help me how to solve problem with driver.

harykishan, 2023-09-14:
Hello @rossanoparis & all who are facing this issue with latest kernel versions of 6.0, 6.1 and above, 
this is how I was able to solve this issue on my system with following configuration:
Ubuntu 22.04 LTS.
Kernel version - 6.2.0-31-generic
Wifi adapter - Tplink Archer T4U ver3
driver installed - this same one Realtek RTL8812BU

For this TpLink Archer T4U (USB wi-fi adapter) new drivers for latest kernel version of 6.2 and above on Ubuntu 22.04
**Old version drivers have become unstable since Linux Kernel version 6.2 on Ubuntu 22.04.** 
Hence you need to install these latest version drivers for this TpLink Archer T4U. 

{{{**SOLVED, REQUIRES A DIFFERENT DRIVER SOURCE FOR 6.2 ONWARDS IN ORDER TO BE STABLE! INSTALL WITH INSTRUCTIONS BELOW:**}}}

**CODE TO RUN:**
sudo dkms remove rtl88x2bu/[5.8.7.1](http://5.8.7.1/) --all
sudo apt install git dkms
git clone https://github.com/morrownr/88x2bu-20210702.git
cd 88x2bu-20210702
sudo ./install-driver.sh

After installing these commands successfully, reboot and check if this driver is installed with:
**It must show the following output for this driver as "installed", like below:**
hk@hk-Ideapad-Z570:~$ sudo dkms status
[sudo] password for hk:
akvcam/1.2.3, 5.15.0-60-generic, x86_64: installed
akvcam/1.2.3, 5.15.0-71-generic, x86_64: installed
akvcam/1.2.3, 5.15.0-72-generic, x86_64: installed
akvcam/1.2.3, 5.15.0-73-generic, x86_64: installed
akvcam/1.2.3, 5.15.0-75-generic, x86_64: installed
akvcam/1.2.3, 5.15.0-76-generic, x86_64: installed
akvcam/1.2.3, 5.15.0-78-generic, x86_64: installed
akvcam/1.2.3, 5.15.0-79-generic, x86_64: installed
akvcam/1.2.3, 5.15.0-82-generic, x86_64: installed
akvcam/1.2.3, 5.19.0-50-generic, x86_64: installed
akvcam/1.2.3, 6.2.0-26-generic, x86_64: installed
akvcam/1.2.3, 6.2.0-31-generic, x86_64: installed
**rtl88x2bu/5.13.1, 6.2.0-31-generic, x86_64: installed**
v4l2loopback/0.12.7, 5.15.0-71-generic, x86_64: installed
v4l2loopback/0.12.7, 5.15.0-72-generic, x86_64: installed
v4l2loopback/0.12.7, 5.15.0-73-generic, x86_64: installed
v4l2loopback/0.12.7, 5.15.0-75-generic, x86_64: installed
v4l2loopback/0.12.7, 5.15.0-76-generic, x86_64: installed
v4l2loopback/0.12.7, 5.15.0-78-generic, x86_64: installed
v4l2loopback/0.12.7, 5.15.0-79-generic, x86_64: installed
v4l2loopback/0.12.7, 5.15.0-82-generic, x86_64: installed
v4l2loopback/0.12.7, 5.19.0-50-generic, x86_64: installed
v4l2loopback/0.12.7, 6.2.0-26-generic, x86_64: installed
v4l2loopback/0.12.7, 6.2.0-31-generic, x86_64: installed
hk@hk-Ideapad-Z570:~$

With this output, do a reboot of your system one more time. 
**NOW AFTER REBOOTING, PLUG IN YOUR ARCHER T4U, AND KA-CHING, THAT BEAUTIFUL GREEN LED ON IT WILL START GLOWING AND YOU GET TO ACCESS YOUR WIFI WITH FULL HIGH SPEED.**

**MAKE THIS AS AN ANSWER SO THAT THIS SOLUTION HELPS ALL.**

RinCat, 2023-10-01:
Read this please: https://github.com/RinCat/RTL88x2BU-Linux-Driver#linux-518-and-rtw88-driver