RinCat/RTL88x2BU-Linux-Driver #209: Mainline 8822 driver vs this repo differences

chrisvollorenew, 2023-09-12:
Hi, I'm not sure the best place to ask this, so I'll start here. 
What are the main difference between the mainline wifi driver for 8822bu and this one? 
Does this driver support features that are not included in the mainline? (such as concurrent mode?)

Duke-93, 2023-09-14:
On my AC1300 Archer T4U Wi-Fi adapter, the LED indicator does not work when using the main driver rtw88_8822bu.

RinCat, 2023-10-01:
Mainline is someone rewrite based on Realtek open sourced drivers that try to support all RTL88 WiFi devices. This one is I modified based on Realtek RTL88x2BU open sourced driver, which included some official proprietary binary content. However Realtek seems no longer supported this driver, so it's in maintenance mode. No new features will be added, only compatibility with the new version of the kernel will be made until the mainline driver is stable enough.