RinCat/RTL88x2BU-Linux-Driver #211: don't work with: ID 0bda:b812 Realtek Semiconductor Corp. RTL88x2bu [AC1200 Techkey]

zenbooster, 2023-09-26:
raspberry pi 2 model b v1.1

```
zenbooster@zen-bridge-0:~/git/RTL88x2BU-Linux-Driver$ uname -r
5.15.0-1037-raspi
```

```
zenbooster@zen-bridge-0:~/git/RTL88x2BU-Linux-Driver$ cat /etc/os-release
PRETTY_NAME="Ubuntu 22.04.3 LTS"
NAME="Ubuntu"
VERSION_ID="22.04"
VERSION="22.04.3 LTS (Jammy Jellyfish)"
VERSION_CODENAME=jammy
ID=ubuntu
ID_LIKE=debian
HOME_URL="https://www.ubuntu.com/"
SUPPORT_URL="https://help.ubuntu.com/"
BUG_REPORT_URL="https://bugs.launchpad.net/ubuntu/"
PRIVACY_POLICY_URL="https://www.ubuntu.com/legal/terms-and-policies/privacy-policy"
UBUNTU_CODENAME=jammy
```

```
zenbooster@zen-bridge-0:~/git/RTL88x2BU-Linux-Driver$ lsusb
Bus 001 Device 088: ID 0bda:b812 Realtek Semiconductor Corp. RTL88x2bu [AC1200 Techkey]
Bus 001 Device 110: ID 0424:ec00 Microchip Technology, Inc. (formerly SMSC) SMSC9512/9514 Fast Ethernet Adapter
Bus 001 Device 109: ID 0424:9514 Microchip Technology, Inc. (formerly SMSC) SMC9514 Hub
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
```

```
Sep 26 14:00:16 zen-bridge-0 kernel: [ 6814.703519] RTW: 0x2E0: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF
Sep 26 14:00:16 zen-bridge-0 kernel: [ 6814.703734] RTW: 0x2F0: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF
Sep 26 14:00:16 zen-bridge-0 kernel: [ 6814.703922] RTW: hal_com_config_channel_plan chplan:0x7F
Sep 26 14:00:16 zen-bridge-0 kernel: [ 6814.703955] RTW: ERROR [HALMAC][ERR]Dump efuse in suspend
Sep 26 14:00:16 zen-bridge-0 kernel: [ 6814.712718] RTW: WARN [HALMAC][WARN]power state unchange!!
Sep 26 14:00:16 zen-bridge-0 kernel: [ 6814.738153] RTW: [HALMAC][ALWAYS]shall R reg twice!!
Sep 26 14:00:16 zen-bridge-0 kernel: [ 6814.738185] RTW: WARN rtw_halmac_poweron: Power state abnormal, try to recover...OK
Sep 26 14:00:16 zen-bridge-0 kernel: [ 6814.859683] RTW: WARN Invalid rate 0x0 in MRateToHwRate
Sep 26 14:00:16 zen-bridge-0 kernel: [ 6814.859729] RTW: WARN Invalid rate 0x0 in MRateToHwRate
Sep 26 14:00:16 zen-bridge-0 kernel: [ 6814.859750] RTW: WARN Invalid rate 0x0 in MRateToHwRate
Sep 26 14:00:16 zen-bridge-0 kernel: [ 6814.859768] RTW: WARN Invalid rate 0x0 in MRateToHwRate
Sep 26 14:00:16 zen-bridge-0 kernel: [ 6814.893539] RTW: ERROR [HALMAC][ERR]Dump efuse in suspend
Sep 26 14:00:16 zen-bridge-0 kernel: [ 6814.893923] RTW: ERROR [HALMAC][ERR]Dump efuse in suspend
Sep 26 14:00:16 zen-bridge-0 kernel: [ 6814.894379] RTW: ERROR [HALMAC][ERR]Dump efuse in suspend
Sep 26 14:00:16 zen-bridge-0 kernel: [ 6814.895097] RTW: [RF_PATH] ver_id.RF_TYPE:RF_2T2R
Sep 26 14:00:16 zen-bridge-0 kernel: [ 6814.895126] RTW: [RF_PATH] HALSPEC's rf_reg_trx_path_bmp:0x33, rf_reg_path_avail_num:2, max_tx_cnt:2
Sep 26 14:00:16 zen-bridge-0 kernel: [ 6814.895147] RTW: [RF_PATH] PG's trx_path_bmp:0x00, max_tx_cnt:0
Sep 26 14:00:16 zen-bridge-0 kernel: [ 6814.895163] RTW: [RF_PATH] Registry's trx_path_bmp:0x00, tx_path_lmt:0, rx_path_lmt:0
Sep 26 14:00:16 zen-bridge-0 kernel: [ 6814.895179] RTW: [RF_PATH] HALDATA's trx_path_bmp:0x33, max_tx_cnt:2
Sep 26 14:00:16 zen-bridge-0 kernel: [ 6814.895195] RTW: [RF_PATH] HALDATA's rf_type:RF_2T2R, NumTotalRFPath:2
Sep 26 14:00:16 zen-bridge-0 kernel: [ 6814.895213] RTW: [TRX_Nss] HALSPEC - tx_nss:2, rx_nss:2
Sep 26 14:00:16 zen-bridge-0 kernel: [ 6814.895230] RTW: [TRX_Nss] Registry - tx_nss:0, rx_nss:0
Sep 26 14:00:16 zen-bridge-0 kernel: [ 6814.895244] RTW: [TRX_Nss] HALDATA - tx_nss:2, rx_nss:2
Sep 26 14:00:16 zen-bridge-0 kernel: [ 6814.895265] RTW: txpath=0x3, rxpath=0x3
Sep 26 14:00:16 zen-bridge-0 kernel: [ 6814.895280] RTW: txpath_1ss:0x1, num:1
Sep 26 14:00:16 zen-bridge-0 kernel: [ 6814.895295] RTW: txpath_2ss:0x3, num:2
Sep 26 14:00:16 zen-bridge-0 kernel: [ 6814.897459] RTW: rtw_regsty_chk_target_tx_power_valid return _FALSE for band:0, path:0, rs:0, t:-1
Sep 26 14:00:16 zen-bridge-0 kernel: [ 6814.903190] RTW: rtw_ndev_init(wlan0) if1 mac_addr=1c:bf:ce:56:38:57
Sep 26 14:00:16 zen-bridge-0 networkd-dispatcher[636]: WARNING:Unknown index 44 seen, reloading interface list
Sep 26 14:00:17 zen-bridge-0 kernel: [ 6815.043729] RTW: ERROR _halmac_reg_write_8: I/O FAIL!
Sep 26 14:00:17 zen-bridge-0 kernel: [ 6815.043768] RTW: ERROR _halmac_reg_write_8: I/O FAIL!
Sep 26 14:00:17 zen-bridge-0 kernel: [ 6815.043783] RTW: ERROR _halmac_reg_write_8: I/O FAIL!
Sep 26 14:00:17 zen-bridge-0 kernel: [ 6815.043795] RTW: ERROR _halmac_reg_write_8: I/O FAIL!
Sep 26 14:00:17 zen-bridge-0 kernel: [ 6815.043808] RTW: ERROR _halmac_reg_write_8: I/O FAIL!
Sep 26 14:00:17 zen-bridge-0 kernel: [ 6815.043827] RTW: ERROR _halmac_reg_write_32: I/O FAIL!
Sep 26 14:00:17 zen-bridge-0 kernel: [ 6815.043840] RTW: ERROR _halmac_reg_write_8: I/O FAIL!
Sep 26 14:00:17 zen-bridge-0 kernel: [ 6815.043865] RTW: WARN rtw_halmac_txfifo_wait_empty: Interrupted by I/O forbiden!
Sep 26 14:00:17 zen-bridge-0 kernel: [ 6815.043878] RTW: ERROR rtw_halmac_txfifo_wait_empty: Fail to wait txfifo empty!(cnt=1)
[... 317 lines cut ...]
Sep 26 14:00:21 zen-bridge-0 kernel: [ 6819.511664] RTW: txpath_2ss:0x3, num:2
Sep 26 14:00:21 zen-bridge-0 kernel: [ 6819.513620] RTW: rtw_regsty_chk_target_tx_power_valid return _FALSE for band:0, path:0, rs:0, t:-1
Sep 26 14:00:21 zen-bridge-0 kernel: [ 6819.518890] RTW: rtw_ndev_init(wlan0) if1 mac_addr=1c:bf:ce:56:38:57
Sep 26 14:00:21 zen-bridge-0 networkd-dispatcher[636]: WARNING:Unknown index 47 seen, reloading interface list
Sep 26 14:00:21 zen-bridge-0 kernel: [ 6819.684864] RTW: ERROR _halmac_reg_write_8: I/O FAIL!
Sep 26 14:00:21 zen-bridge-0 kernel: [ 6819.684903] RTW: ERROR _halmac_reg_write_8: I/O FAIL!
Sep 26 14:00:21 zen-bridge-0 kernel: [ 6819.684917] RTW: ERROR _halmac_reg_write_8: I/O FAIL!
Sep 26 14:00:21 zen-bridge-0 kernel: [ 6819.684930] RTW: ERROR _halmac_reg_write_8: I/O FAIL!
Sep 26 14:00:21 zen-bridge-0 kernel: [ 6819.684943] RTW: ERROR _halmac_reg_write_8: I/O FAIL!
Sep 26 14:00:21 zen-bridge-0 kernel: [ 6819.684963] RTW: ERROR _halmac_reg_write_32: I/O FAIL!
Sep 26 14:00:21 zen-bridge-0 kernel: [ 6819.684976] RTW: ERROR _halmac_reg_write_8: I/O FAIL!
Sep 26 14:00:21 zen-bridge-0 kernel: [ 6819.685001] RTW: WARN rtw_halmac_txfifo_wait_empty: Interrupted by I/O forbiden!
Sep 26 14:00:21 zen-bridge-0 kernel: [ 6819.685014] RTW: ERROR rtw_halmac_txfifo_wait_empty: Fail to wait txfifo empty!(cnt=1)
Sep 26 14:00:21 zen-bridge-0 kernel: [ 6819.685032] RTW: ERROR rtl8822b_hal_init Download Firmware from array failed
Sep 26 14:00:21 zen-bridge-0 kernel: [ 6819.685053] RTW: ERROR _halmac_reg_write_8: I/O FAIL!
Sep 26 14:00:21 zen-bridge-0 kernel: [ 6819.685066] RTW: ERROR _halmac_reg_write_8: I/O FAIL!
Sep 26 14:00:21 zen-bridge-0 kernel: [ 6819.685081] RTW: ERROR _halmac_reg_write_8: I/O FAIL!
Sep 26 14:00:21 zen-bridge-0 kernel: [ 6819.685098] RTW: ERROR _halmac_reg_write_32: I/O FAIL!
Sep 26 14:00:21 zen-bridge-0 kernel: [ 6819.685111] RTW: ERROR _halmac_reg_write_16: I/O FAIL!
Sep 26 14:00:21 zen-bridge-0 kernel: [ 6819.685161] RTW: ERROR _halmac_reg_write_8: I/O FAIL!
Sep 26 14:00:21 zen-bridge-0 systemd-networkd[612]: wlan0: Link UP
Sep 26 14:00:21 zen-bridge-0 kernel: [ 6819.787443] usb 1-1.5: USB disconnect, device number 48
Sep 26 14:00:21 zen-bridge-0 systemd-networkd[612]: wlan0: Link DOWN

```

RinCat, 2023-10-01:
Maybe power issue? It is I/O error.

ahmogit, 2023-12-27:
@zenbooster: Fwiw: Just got one of these from Amazon (0bda:b812, Techkey).  This driver built (with a few warnings) under gcc 13.2.1, and operates without issue for me. Distro: Arch, kernel 6.1.68 on an x86_64 commodity laptop.  @RinCat: Let me know if you want a list of the gcc warnings.

robinelvin, 2025-03-20:
Same or very similar issue here. Running on a desktop PC under Fedora 41. Compiled without error using auto DKMS installation. I've tried different slots/buses and it does the same thing.

Brand "Fideco"

```
[  380.307908] usb 1-14: new high-speed USB device number 5 using xhci_hcd
[  380.431598] usb 1-14: New USB device found, idVendor=0bda, idProduct=b812, bcdDevice= 2.10
[  380.431615] usb 1-14: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[  380.431622] usb 1-14: Product: USB3.0 802.11ac 1200M Adapter
[  380.431628] usb 1-14: Manufacturer: Realtek
[  380.431633] usb 1-14: SerialNumber: 123456
[  380.435330] RTW: [HALMAC]55772M
               HALMAC_MAJOR_VER = 1
               HALMAC_PROTOTYPE_VER = 6
               HALMAC_MINOR_VER = 6
               HALMAC_PATCH_VER = 23
[  380.435959] RTW: ERROR [HALMAC][ERR]Dump efuse in suspend
[  380.600186] RTW: HW EFUSE
[  380.600197] RTW: 0x000: 29 81 00 3C  09 00 A1 00  B6 04 64 10  00 00 A3 00  
[  380.600237] RTW: 0x010: 35 35 35 35  35 35 33 33  33 33 33 F2  00 00 FF FF  
[  380.600274] RTW: 0x020: FF FF 28 2A  2C 2E 25 25  24 24 2A 2A  29 29 28 28  
[  380.600311] RTW: 0x030: F0 00 FF FF  00 FF C0 00  FF FF 33 32  33 33 33 32  
[  380.600347] RTW: 0x040: 33 33 33 33  33 13 00 00  FF FF FF FF  29 29 28 28  
[  380.600382] RTW: 0x050: 25 27 28 2A  29 2E 2F 2F  2E 2E E0 00  FF FF 00 FF  
[  380.600418] RTW: 0x060: C0 00 FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF  
[  380.600454] RTW: 0x070: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF  
[  380.600490] RTW: 0x080: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF  
[  380.600526] RTW: 0x090: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF  
[  380.600562] RTW: 0x0A0: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF  
[  380.600598] RTW: 0x0B0: FF FF FF FF  FF FF FF FF  7F 20 21 00  FF FF FF FF  
[  380.600634] RTW: 0x0C0: FF 09 00 11  00 00 00 00  00 FF 03 FF  FF FF FF FF  
[  380.600670] RTW: 0x0D0: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF  
[  380.600706] RTW: 0x0E0: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF  
[  380.600742] RTW: 0x0F0: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF  
[  380.600778] RTW: 0x100: DA 0B 12 B8  C1 67 02 1C  BF CE 46 91  75 09 03 52  
[  380.600823] RTW: 0x110: 65 61 6C 74  65 6B 1F 03  55 53 42 33  2E 30 20 38  
[  380.600860] RTW: 0x120: 30 32 2E 31  31 61 63 20  31 32 30 30  4D 20 41 64  
[  380.600896] RTW: 0x130: 61 70 74 65  72 08 03 31  32 33 34 35  36 FF FF FF  
[  380.600933] RTW: 0x140: 33 0A 1F 01  00 00 21 0F  FF FF FF FF  FF FF FF FF  
[  380.600969] RTW: 0x150: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF  
[  380.601005] RTW: 0x160: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF  
[  380.601042] RTW: 0x170: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF  
[  380.601078] RTW: 0x180: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF  
[  380.601114] RTW: 0x190: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF  
[  380.601151] RTW: 0x1A0: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF  
[  380.601187] RTW: 0x1B0: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF  
[  380.601223] RTW: 0x1C0: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF  
[  380.601260] RTW: 0x1D0: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF  
[  380.601296] RTW: 0x1E0: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF  
[  380.601332] RTW: 0x1F0: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF  
[  380.601368] RTW: 0x200: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF  
[  380.601405] RTW: 0x210: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF  
[  380.601441] RTW: 0x220: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF  
[  380.601477] RTW: 0x230: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF  
[  380.601514] RTW: 0x240: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF  
[  380.601550] RTW: 0x250: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF  
[  380.601586] RTW: 0x260: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF  
[  380.601623] RTW: 0x270: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF  
[  380.601659] RTW: 0x280: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF  
[  380.601695] RTW: 0x290: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF  
[  380.601732] RTW: 0x2A0: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF  
[  380.601768] RTW: 0x2B0: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF  
[  380.601804] RTW: 0x2C0: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF  
[  380.601848] RTW: 0x2D0: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF  
[  380.601884] RTW: 0x2E0: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF  
[  380.601920] RTW: 0x2F0: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF  
[  380.601960] RTW: hal_com_config_channel_plan chplan:0x7F
[  380.601968] RTW: ERROR [HALMAC][ERR]Dump efuse in suspend
[  380.608061] RTW: [HALMAC][ALWAYS]shall R reg twice!!
[  380.647770] RTW: WARN Invalid rate 0x0 in MRateToHwRate
[... 95 lines cut ...]
[  381.157957]  ? do_syscall_64+0x8e/0x160
[  381.157960]  ? do_syscall_64+0x8e/0x160
[  381.157963]  entry_SYSCALL_64_after_hwframe+0x76/0x7e
[  381.157966] RIP: 0033:0x7fdf43703f8b
[  381.157974] Code: 48 89 e5 48 83 ec 20 89 55 ec 48 89 75 f0 89 7d f8 e8 29 61 f7 ff 8b 55 ec 48 8b 75 f0 41 89 c0 8b 7d f8 b8 2e 00 00 00 0f 05 <48> 3d 00 f0 ff ff 77 2d 44 89 c7 48 89 45 f8 e8 81 61 f7 ff 48 8b
[  381.157976] RSP: 002b:00007ffcffaa82c0 EFLAGS: 00000293 ORIG_RAX: 000000000000002e
[... cut ...]
[  381.158022] RTW: ERROR _halmac_reg_write_8: I/O FAIL!
[  381.158024] RTW: ERROR _halmac_reg_write_8: I/O FAIL!
[  381.158025] RTW: ERROR _halmac_reg_write_32: I/O FAIL!
[  381.158026] RTW: ERROR _halmac_reg_write_16: I/O FAIL!
[  381.158031] RTW: ERROR _halmac_reg_write_8: I/O FAIL!
[  381.158687] RTW: rtw_set_country_cmd country_code:"GB" mapping to chplan:0x26
[  381.158692] RTW: there is no any txpwr_regd
[  381.159268] RTW: ERROR _halmac_reg_write_32: I/O FAIL!
[  381.159271] RTW: ERROR _halmac_reg_write_16: I/O FAIL!
```

RinCat, 2025-03-20:
```
[  381.152238] RTW: ERROR _halmac_reg_write_8: I/O FAIL!
[  381.152251] RTW: ERROR _halmac_reg_write_16: I/O FAIL!
```

This indicates that it is disconnected from the system, have you enabled USB3.0?

robinelvin, 2025-03-21:
No matter which USB 3.0 port I plug into it always comes up as USB 2.0 480:

```
/:  Bus 001.Port 001: Dev 001, Class=root_hub, Driver=xhci_hcd/16p, 480M
    |__ Port 007: Dev 006, If 0, Class=Vendor Specific Class, Driver=rtl88x2bu, 480M
    |__ Port 013: Dev 003, If 0, Class=Human Interface Device, Driver=usbhid, 12M
    |__ Port 013: Dev 003, If 1, Class=Human Interface Device, Driver=usbhid, 12M
/:  Bus 002.Port 001: Dev 001, Class=root_hub, Driver=xhci_hcd/10p, 5000M
```

If I plug a USB 3.0 memory stick in to the same port it attaches to the USB 3.0 5000M bus:

```
/:  Bus 001.Port 001: Dev 001, Class=root_hub, Driver=xhci_hcd/16p, 480M
    |__ Port 013: Dev 003, If 0, Class=Human Interface Device, Driver=usbhid, 12M
    |__ Port 013: Dev 003, If 1, Class=Human Interface Device, Driver=usbhid, 12M
/:  Bus 002.Port 001: Dev 001, Class=root_hub, Driver=xhci_hcd/10p, 5000M
    |__ Port 007: Dev 002, If 0, Class=Mass Storage, Driver=usb-storage, 5000M
```

_Notice both were plugged into port 7._

Is this a udev thing or a problem with the driver?

RinCat, 2025-03-21:
Neither, the hardware by default will use USB 2.0. I just need to check if it caused by you enabled USB 3.0 mode in the driver.