RinCat/RTL88x2BU-Linux-Driver #219: [bug report] driver suddenly stopped working

Outspending, 2023-12-10:
I have a TP-Link Archer T2U PLUS [RTL8821AU]. And i updated my kernel and just stopped working.

My `lsusb` is showing my T2U but i can't connect to anything.
[image]

I've tried reinstalling the drivers, manually and through AUR and neither of them worked. 
I am using Arch Linux but they were working fine before i updated my kernel.

I can't see networks in my Networks tab which means its not being initialized or what. The green light on my USB dongle is also not blinking therefore isn't being registered. But i think that's just because the driver isn't working. 

Here's some more kernel information:
[image]

If anyone could help me that would be much appreciated. Thanks :+1:

Outspending, 2023-12-10:
A little update again, i downgraded my kernel to the previous version (6.6.4) and it worked again so its 100% the new kernel update. Hopefully this helps a bit.

neos1lver, 2023-12-10:
Having the same problem here after updating to 6.6.5

ashy1227, 2023-12-11:
Also having this problem, downgrading to 6.6.4 fixed it.

RinCat, 2023-12-11:
This is a known Linux kernel bug in 6.6.5 and 6.1.66 that has deadlock during the connection. Please downgrade the kernel for now.

Outspending, 2023-12-11:
> This is a known Linux kernel bug in 6.6.5 and 6.1.66 that has deadlock during the connection. Please downgrade the kernel for now.

Is there an ETA? Not a huge deal I already downgraded. Just wondering