RinCat/RTL88x2BU-Linux-Driver #222: NCI not appearing in linux 6.2.

fre3e3, 2023-12-31:
Hi, I'm using Xubuntu 22.04 LTS, kernel version 6.2.0-39-generic. I have a TP-LINK Archer T3U, theorically based on the chipset Realtek RTL8812BU.

Here's my output for 
```shell
lsusb
``` 
---
```shell
Bus 002 Device 002: ID 8087:0024 Intel Corp. Integrated Rate Matching Hub
Bus 002 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 001 Device 005: ID 0bda:58c2 Realtek Semiconductor Corp. Integrated Webcam HD
Bus 001 Device 004: ID 0bda:0129 Realtek Semiconductor Corp. RTS5129 Card Reader Controller
Bus 001 Device 003: ID 0a5c:21d7 Broadcom Corp. BCM43142 Bluetooth 4.0
Bus 001 Device 002: ID 8087:0024 Intel Corp. Integrated Rate Matching Hub
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 004 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 003 Device 002: ID 2357:011f TP-Link 802.11ac WLAN Adapter 
Bus 003 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub

``` 
---
I already blacklisted (as root user) the RTW88 8822bu USB driver introduced in 5.18+ kernels with the command in the README section, I installed build-essential, kernel-headers, etc.

I cloned the package via terminal, entered the folder, and made all of the installation steps, still it doesn't work.

During the "make" command procedure, it gave me some warnings, don't know if they are normal, but I think it all went good, 'cause it didn't give me any errors.

Thanks in advance.

RinCat, 2023-12-31:
`2357:011f`  seems Archer T**2**U, which is 8812**A**U.

fre3e3, 2023-12-31:
Right, sorry. Gonna search for that driver right now.