RinCat/RTL88x2BU-Linux-Driver #223: Support monitor,packet injection and AP mode

karthikvijay5227, 2024-01-06:
Do tp-link ac1300 with this driver support monitor,packet injection and AP mode in Kali Linux as i am confused with TP-Link Ac1300 and TP-Link AC600 which uses Realtek RTL8812AU.I want to use it for pentesting.Can anyone suggest the better option to consider.Please respond as soon as possible.

RinCat, 2024-01-08:
Hi, this driver only support BU, not AU. Also AC1300 and AC 600 are not hardware name but series that mixed different chips.

karthikvijay5227, 2024-01-08:
I know that.But I am in confusion of buying tp link ac600 with rtl8812au
and tp link ac133 with rtl8812bu for hacking purposes as I have seen very
few drivers for rtl8812bu linux. So i want to clear the confusion with each
chipset and their support for different modes.

RinCat, 2024-01-08:
This one should support those, but I never test it in Kali Linux. You can try if you want.

karthikvijay5227, 2024-01-08:
Is there any tested ones as I am not get in risk of not supporting it after
spending money.

RinCat, 2024-01-08:
You should able to find some information in the issues. I do remember some users played it with Kali.