RinCat/RTL88x2BU-Linux-Driver #224: Recompile issue on reboot

canoneod, 2024-01-08:
Hi, I'm using driver in Ubuntu 20.04.6 LTS and it just works fine with my device. 

But whenever I reboot my PC, my device is not found and I have to recompile this repo everytime. Does anyone went through 

similiar issue? What could be the possible solution to this?

RinCat, 2024-01-08:
Hi, do you have something like read-only root with overlay, which will discard changes in root filesystem? You should not need to recompile it unless kernel updated.

canoneod, 2024-01-14:
Hi, sorry for the late reply. I didn't have read-only root on the system, so I just assumed the problem is due to my user privilege issue. I solved by installing in super user. Thanks for the reply btw!

closed 2024-01-14 by canoneod