RinCat/RTL88x2BU-Linux-Driver #225: Does not compile against linux 6.7

jonathanjsimon, 2024-01-13:
[linux67-rtl88x2bu-5.13.1.r189.12cfcd8-build-make.log](https://github.com/RinCat/RTL88x2BU-Linux-Driver/files/13929549/linux67-rtl88x2bu-5.13.1.r189.12cfcd8-build-make.log)

DKMS fails to compile against Manjaro linux 6.7. You mention unofficial patches in readme but using distributed linux package does not compile. Here is the log. Thanks!

RinCat, 2024-01-13:
Hi, you need to use the latest driver version.

jonathanjsimon, 2024-01-13:
Understood. I'm using the AUR package, so I'll see why that's out of date. Thanks!

RinCat, 2024-01-13:
For AUR, you just need to reinstall it, it will pull the latest version.

jonathanjsimon, 2024-01-14:
Thank you. That worked. I might suggest bumping the AUR package version to trigger updates via AUR helpers etc, even if not technically necessary, to avoid having to respond to these types of queries, several of which I see in your issues list. Thanks for the driver :-)

closed 2024-01-14 by jonathanjsimon

RinCat, 2024-01-14:
Some AUR helper can do automatically check for such updates.
like `yay -Syu --devel`