RinCat/RTL88x2BU-Linux-Driver #226: DKMS fails - (custom) kernel 6.8, Fedora

onlyreportingissues, 2024-01-31:
DKMS make.log for rtl88x2bu-git for kernel 6.8.0-0.rc2.20240131gt2a6526c4.220.vanilla.fc39.x86_64 (x86_64)
Mi 31. Jan 19:22:31 CET 2024
make ARCH=x86_64 CROSS_COMPILE= -C /lib/modules/6.8.0-0.rc2.20240131gt2a6526c4.220.vanilla.fc39.x86_64/build M=/var/lib/dkms/rtl88x2bu/git/build  modules
make[1]: Verzeichnis „/usr/src/kernels/6.8.0-0.rc2.20240131gt2a6526c4.220.vanilla.fc39.x86_64“ wird betreten
  CC [M]  /var/lib/dkms/rtl88x2bu/git/build/core/rtw_cmd.o
  CC [M]  /var/lib/dkms/rtl88x2bu/git/build/core/rtw_security.o
  CC [M]  /var/lib/dkms/rtl88x2bu/git/build/core/rtw_debug.o
  CC [M]  /var/lib/dkms/rtl88x2bu/git/build/core/rtw_io.o
  CC [M]  /var/lib/dkms/rtl88x2bu/git/build/core/rtw_ioctl_query.o
  CC [M]  /var/lib/dkms/rtl88x2bu/git/build/core/rtw_ioctl_set.o
  CC [M]  /var/lib/dkms/rtl88x2bu/git/build/core/rtw_ieee80211.o
  CC [M]  /var/lib/dkms/rtl88x2bu/git/build/core/rtw_mlme.o
  CC [M]  /var/lib/dkms/rtl88x2bu/git/build/core/rtw_mlme_ext.o
  CC [M]  /var/lib/dkms/rtl88x2bu/git/build/core/rtw_mi.o
  CC [M]  /var/lib/dkms/rtl88x2bu/git/build/core/rtw_wlan_util.o
  CC [M]  /var/lib/dkms/rtl88x2bu/git/build/core/rtw_vht.o
In Datei, eingebunden von /var/lib/dkms/rtl88x2bu/git/build/include/drv_types.h:114,
                 von /var/lib/dkms/rtl88x2bu/git/build/core/rtw_cmd.c:17:
/var/lib/dkms/rtl88x2bu/git/build/include/rtw_mlme_ext.h:1181:4: Warnung: kein vorheriger Prototyp für »rtw_mesh_set_plink_state_cmd_hdl« [-Wmissing-prototypes]
 1181 | u8 rtw_mesh_set_plink_state_cmd_hdl(_adapter *adapter, u8 *parmbuf) { return H2C_CMD_FAIL; };
      |    ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
/var/lib/dkms/rtl88x2bu/git/build/core/rtw_ieee80211.c:489:6: Warnung: kein vorheriger Prototyp für »rtw_filter_suppport_rateie« [-Wmissing-prototypes]
  489 | void rtw_filter_suppport_rateie(WLAN_BSSID_EX *pbss_network, u8 keep)
      |      ^~~~~~~~~~~~~~~~~~~~~~~~~~
/var/lib/dkms/rtl88x2bu/git/build/core/rtw_ieee80211.c:766:5: Warnung: kein vorheriger Prototyp für »rtw_get_akm_suite_bitmap« [-Wmissing-prototypes]
  766 | u32 rtw_get_akm_suite_bitmap(u8 *s)
      |     ^~~~~~~~~~~~~~~~~~~~~~~~
/var/lib/dkms/rtl88x2bu/git/build/core/rtw_ieee80211.c:1824:6: Warnung: kein vorheriger Prototyp für »dump_ht_cap_ie« [-Wmissing-prototypes]
 1824 | void dump_ht_cap_ie(void *sel, const u8 *ie, u32 ie_len)
      |      ^~~~~~~~~~~~~~
/var/lib/dkms/rtl88x2bu/git/build/core/rtw_ieee80211.c:1843:6: Warnung: kein vorheriger Prototyp für »dump_ht_op_ie_content« [-Wmissing-prototypes]
 1843 | void dump_ht_op_ie_content(void *sel, const u8 *buf, u32 buf_len)
      |      ^~~~~~~~~~~~~~~~~~~~~
/var/lib/dkms/rtl88x2bu/git/build/core/rtw_ieee80211.c:1857:6: Warnung: kein vorheriger Prototyp für »dump_ht_op_ie« [-Wmissing-prototypes]
 1857 | void dump_ht_op_ie(void *sel, const u8 *ie, u32 ie_len)
      |      ^~~~~~~~~~~~~
  CC [M]  /var/lib/dkms/rtl88x2bu/git/build/core/rtw_pwrctrl.o
/var/lib/dkms/rtl88x2bu/git/build/core/rtw_mi.c:877:4: Warnung: kein vorheriger Prototyp für »_rtw_mi_busy_traffic_check« [-Wmissing-prototypes]
  877 | u8 _rtw_mi_busy_traffic_check(_adapter *padapter, void *data)
      |    ^~~~~~~~~~~~~~~~~~~~~~~~~~
/var/lib/dkms/rtl88x2bu/git/build/core/rtw_mi.c:1320:6: Warnung: kein vorheriger Prototyp für »rtw_dbg_skb_process« [-Wmissing-prototypes]
 1320 | void rtw_dbg_skb_process(_adapter *padapter, union recv_frame *precvframe, union recv_frame *pcloneframe)
      |      ^~~~~~~~~~~~~~~~~~~
/var/lib/dkms/rtl88x2bu/git/build/core/rtw_cmd.c:1634:6: Warnung: kein vorheriger Prototyp für »free_assoc_resources_hdl« [-Wmissing-prototypes]
 1634 | void free_assoc_resources_hdl(_adapter *padapter, u8 lock_scanned_queue)
      |      ^~~~~~~~~~~~~~~~~~~~~~~~
/var/lib/dkms/rtl88x2bu/git/build/core/rtw_mlme.c:26:6: Warnung: kein vorheriger Prototyp für »rtw_init_mlme_timer« [-Wmissing-prototypes]
   26 | void rtw_init_mlme_timer(_adapter *padapter)
      |      ^~~~~~~~~~~~~~~~~~~
/var/lib/dkms/rtl88x2bu/git/build/core/rtw_mlme.c:43:9: Warnung: kein vorheriger Prototyp für »_rtw_init_mlme_priv« [-Wmissing-prototypes]
   43 | sint    _rtw_init_mlme_priv(_adapter *padapter)
      |         ^~~~~~~~~~~~~~~~~~~
/var/lib/dkms/rtl88x2bu/git/build/core/rtw_cmd.c:2212:4: Warnung: kein vorheriger Prototyp für »rtw_ssmps_wk_hdl« [-Wmissing-prototypes]
 2212 | u8 rtw_ssmps_wk_hdl(_adapter *adapter, struct ssmps_cmd_parm *ssmp_param)
      |    ^~~~~~~~~~~~~~~~
/var/lib/dkms/rtl88x2bu/git/build/core/rtw_mlme.c:329:6: Warnung: kein vorheriger Prototyp für »_rtw_free_mlme_priv« [-Wmissing-prototypes]
  329 | void _rtw_free_mlme_priv(struct mlme_priv *pmlmepriv)
      |      ^~~~~~~~~~~~~~~~~~~
/var/lib/dkms/rtl88x2bu/git/build/core/rtw_mlme.c:357:9: Warnung: kein vorheriger Prototyp für »_rtw_enqueue_network« [-Wmissing-prototypes]
  357 | sint    _rtw_enqueue_network(_queue *queue, struct wlan_network *pnetwork)
      |         ^~~~~~~~~~~~~~~~~~~~
/var/lib/dkms/rtl88x2bu/git/build/core/rtw_cmd.c:2992:6: Warnung: kein vorheriger Prototyp für »rtw_dynamic_chk_wk_hdl« [-Wmissing-prototypes]
 2992 | void rtw_dynamic_chk_wk_hdl(_adapter *padapter)
      |      ^~~~~~~~~~~~~~~~~~~~~~
/var/lib/dkms/rtl88x2bu/git/build/core/rtw_cmd.c:3049:6: Warnung: kein vorheriger Prototyp für »lps_ctrl_wk_hdl« [-Wmissing-prototypes]
 3049 | void lps_ctrl_wk_hdl(_adapter *padapter, u8 lps_ctrl_type, u8 *buf)
      |      ^~~~~~~~~~~~~~~
/var/lib/dkms/rtl88x2bu/git/build/core/rtw_cmd.c:3228:6: Warnung: kein vorheriger Prototyp für »rtw_dm_in_lps_hdl« [-Wmissing-prototypes]
 3228 | void rtw_dm_in_lps_hdl(_adapter *padapter)
      |      ^~~~~~~~~~~~~~~~~
[... 680 lines cut ...]
 9927 | void rtw_cfg80211_update_wiphy_max_txpower(_adapter *adapter, struct wiphy *wiphy)
      |      ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
/var/lib/dkms/rtl88x2bu/git/build/os_dep/linux/ioctl_cfg80211.c:10326:5: Warnung: kein vorheriger Prototyp für »cfg80211_rtw_external_auth« [-Wmissing-prototypes]
10326 | int cfg80211_rtw_external_auth(struct wiphy *wiphy, struct net_device *dev,
      |     ^~~~~~~~~~~~~~~~~~~~~~~~~~
/var/lib/dkms/rtl88x2bu/git/build/os_dep/linux/ioctl_cfg80211.c: In Funktion »rtw_get_chbwoff_from_cfg80211_chan_def«:
/var/lib/dkms/rtl88x2bu/git/build/os_dep/linux/ioctl_cfg80211.c:6841:21: Warnung: diese Anweisung könnte durchfallen [-Wimplicit-fallthrough=]
 6841 |                 *ht = 0;
      |                 ~~~~^~~
/var/lib/dkms/rtl88x2bu/git/build/os_dep/linux/ioctl_cfg80211.c:6843:9: Anmerkung: hier
 6843 |         case NL80211_CHAN_WIDTH_20:
      |         ^~~~
/var/lib/dkms/rtl88x2bu/git/build/os_dep/linux/rtw_cfgvendor.c:252:5: Warnung: kein vorheriger Prototyp für »rtw_dev_get_feature_set« [-Wmissing-prototypes]
  252 | int rtw_dev_get_feature_set(struct net_device *dev)
      |     ^~~~~~~~~~~~~~~~~~~~~~~
/var/lib/dkms/rtl88x2bu/git/build/os_dep/linux/rtw_cfgvendor.c:291:6: Warnung: kein vorheriger Prototyp für »rtw_dev_get_feature_set_matrix« [-Wmissing-prototypes]
  291 | int *rtw_dev_get_feature_set_matrix(struct net_device *dev, int *num)
      |      ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
cc1: Einige Warnungen werden als Fehler behandelt
make[3]: *** [scripts/Makefile.build:243: /var/lib/dkms/rtl88x2bu/git/build/os_dep/linux/os_intfs.o] Fehler 1
/var/lib/dkms/rtl88x2bu/git/build/os_dep/linux/rtw_android.c:367:5: Warnung: kein vorheriger Prototyp für »rtw_android_get_rssi« [-Wmissing-prototypes]
  367 | int rtw_android_get_rssi(struct net_device *net, char *command, int total_len)
      |     ^~~~~~~~~~~~~~~~~~~~
/var/lib/dkms/rtl88x2bu/git/build/os_dep/linux/rtw_android.c:382:5: Warnung: kein vorheriger Prototyp für »rtw_android_get_link_speed« [-Wmissing-prototypes]
  382 | int rtw_android_get_link_speed(struct net_device *net, char *command, int total_len)

onlyreportingissues, 2024-01-31:
|     ^~~~~~~~~~~~~~~~~~~~~~~~~~
/var/lib/dkms/rtl88x2bu/git/build/os_dep/linux/rtw_android.c:394:5: Warnung: kein vorheriger Prototyp für »rtw_android_get_macaddr« [-Wmissing-prototypes]
  394 | int rtw_android_get_macaddr(struct net_device *net, char *command, int total_len)
      |     ^~~~~~~~~~~~~~~~~~~~~~~
/var/lib/dkms/rtl88x2bu/git/build/os_dep/linux/rtw_android.c:402:5: Warnung: kein vorheriger Prototyp für »rtw_android_set_country« [-Wmissing-prototypes]
  402 | int rtw_android_set_country(struct net_device *net, char *command, int total_len)
      |     ^~~~~~~~~~~~~~~~~~~~~~~
/var/lib/dkms/rtl88x2bu/git/build/os_dep/linux/rtw_android.c:413:5: Warnung: kein vorheriger Prototyp für »rtw_android_get_p2p_dev_addr« [-Wmissing-prototypes]
  413 | int rtw_android_get_p2p_dev_addr(struct net_device *net, char *command, int total_len)
      |     ^~~~~~~~~~~~~~~~~~~~~~~~~~~~
/var/lib/dkms/rtl88x2bu/git/build/os_dep/linux/rtw_android.c:424:5: Warnung: kein vorheriger Prototyp für »rtw_android_set_block_scan« [-Wmissing-prototypes]
  424 | int rtw_android_set_block_scan(struct net_device *net, char *command, int total_len)
      |     ^~~~~~~~~~~~~~~~~~~~~~~~~~
/var/lib/dkms/rtl88x2bu/git/build/os_dep/linux/rtw_android.c:436:5: Warnung: kein vorheriger Prototyp für »rtw_android_set_block« [-Wmissing-prototypes]
  436 | int rtw_android_set_block(struct net_device *net, char *command, int total_len)
      |     ^~~~~~~~~~~~~~~~~~~~~
/var/lib/dkms/rtl88x2bu/git/build/os_dep/linux/rtw_android.c:448:5: Warnung: kein vorheriger Prototyp für »rtw_android_setband« [-Wmissing-prototypes]
  448 | int rtw_android_setband(struct net_device *net, char *command, int total_len)
      |     ^~~~~~~~~~~~~~~~~~~
/var/lib/dkms/rtl88x2bu/git/build/os_dep/linux/rtw_android.c:461:5: Warnung: kein vorheriger Prototyp für »rtw_android_getband« [-Wmissing-prototypes]
  461 | int rtw_android_getband(struct net_device *net, char *command, int total_len)
      |     ^~~~~~~~~~~~~~~~~~~
/var/lib/dkms/rtl88x2bu/git/build/os_dep/linux/rtw_android.c:472:5: Warnung: kein vorheriger Prototyp für »rtw_android_set_miracast_mode« [-Wmissing-prototypes]
  472 | int rtw_android_set_miracast_mode(struct net_device *net, char *command, int total_len)
      |     ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~
/var/lib/dkms/rtl88x2bu/git/build/os_dep/linux/rtw_android.c:508:5: Warnung: kein vorheriger Prototyp für »get_int_from_command« [-Wmissing-prototypes]
  508 | int get_int_from_command(char *pcmd)
      |     ^~~~~~~~~~~~~~~~~~~~
/var/lib/dkms/rtl88x2bu/git/build/os_dep/linux/rtw_proc.c:739:5: Warnung: kein vorheriger Prototyp für »proc_get_led_config« [-Wmissing-prototypes]
  739 | int proc_get_led_config(struct seq_file *m, void *v)
      |     ^~~~~~~~~~~~~~~~~~~
/var/lib/dkms/rtl88x2bu/git/build/os_dep/linux/rtw_proc.c:749:9: Warnung: kein vorheriger Prototyp für »proc_set_led_config« [-Wmissing-prototypes]
  749 | ssize_t proc_set_led_config(struct file *file, const char __user *buffer, size_t count, loff_t *pos, void *data)
      |         ^~~~~~~~~~~~~~~~~~~
/var/lib/dkms/rtl88x2bu/git/build/os_dep/linux/rtw_proc.c:781:5: Warnung: kein vorheriger Prototyp für »proc_get_aid_status« [-Wmissing-prototypes]
  781 | int proc_get_aid_status(struct seq_file *m, void *v)
      |     ^~~~~~~~~~~~~~~~~~~
/var/lib/dkms/rtl88x2bu/git/build/os_dep/linux/rtw_proc.c:791:9: Warnung: kein vorheriger Prototyp für »proc_set_aid_status« [-Wmissing-prototypes]
  791 | ssize_t proc_set_aid_status(struct file *file, const char __user *buffer, size_t count, loff_t *pos, void *data)
      |         ^~~~~~~~~~~~~~~~~~~
/var/lib/dkms/rtl88x2bu/git/build/os_dep/linux/rtw_proc.c:824:5: Warnung: kein vorheriger Prototyp für »proc_get_ap_isolate« [-Wmissing-prototypes]
  824 | int proc_get_ap_isolate(struct seq_file *m, void *v)
      |     ^~~~~~~~~~~~~~~~~~~
/var/lib/dkms/rtl88x2bu/git/build/os_dep/linux/rtw_proc.c:834:9: Warnung: kein vorheriger Prototyp für »proc_set_ap_isolate« [-Wmissing-prototypes]
  834 | ssize_t proc_set_ap_isolate(struct file *file, const char __user *buffer, size_t count, loff_t *pos, void *data)
      |         ^~~~~~~~~~~~~~~~~~~
/var/lib/dkms/rtl88x2bu/git/build/os_dep/linux/rtw_proc.c:1693:9: Warnung: kein vorheriger Prototyp für »proc_set_macaddr_acl« [-Wmissing-prototypes]
 1693 | ssize_t proc_set_macaddr_acl(struct file *file, const char __user *buffer, size_t count, loff_t *pos, void *data)
      |         ^~~~~~~~~~~~~~~~~~~~
/var/lib/dkms/rtl88x2bu/git/build/os_dep/linux/rtw_proc.c:1994:9: Warnung: kein vorheriger Prototyp für »proc_set_update_non_ocp« [-Wmissing-prototypes]
 1994 | ssize_t proc_set_update_non_ocp(struct file *file, const char __user *buffer, size_t count, loff_t *pos, void *data)
      |         ^~~~~~~~~~~~~~~~~~~~~~~
/var/lib/dkms/rtl88x2bu/git/build/os_dep/linux/rtw_proc.c:2037:9: Warnung: kein vorheriger Prototyp für »proc_set_radar_detect« [-Wmissing-prototypes]
 2037 | ssize_t proc_set_radar_detect(struct file *file, const char __user *buffer, size_t count, loff_t *pos, void *data)
      |         ^~~~~~~~~~~~~~~~~~~~~
/var/lib/dkms/rtl88x2bu/git/build/os_dep/linux/rtw_proc.c:2199:5: Warnung: kein vorheriger Prototyp für »proc_get_rx_ampdu_size_limit« [-Wmissing-prototypes]
 2199 | int proc_get_rx_ampdu_size_limit(struct seq_file *m, void *v)
      |     ^~~~~~~~~~~~~~~~~~~~~~~~~~~~
[... cut ...]
/var/lib/dkms/rtl88x2bu/git/build/os_dep/linux/rtw_proc.c:5996:6: Warnung: kein vorheriger Prototyp für »rtw_odm_proc_deinit« [-Wmissing-prototypes]
 5996 | void rtw_odm_proc_deinit(_adapter       *adapter)
      |      ^~~~~~~~~~~~~~~~~~~
make[2]: *** [/usr/src/kernels/6.8.0-0.rc2.20240131gt2a6526c4.220.vanilla.fc39.x86_64/Makefile:1923: /var/lib/dkms/rtl88x2bu/git/build] Fehler 2
make[1]: *** [Makefile:242: __sub-make] Fehler 2
make[1]: Verzeichnis „/usr/src/kernels/6.8.0-0.rc2.20240131gt2a6526c4.220.vanilla.fc39.x86_64“ wird verlassen
make: *** [Makefile:2470: modules] Fehler 2

RinCat, 2024-02-01:
Hi, its now fixed. Can you try it again?

onlyreportingissues, 2024-02-01:
> Hi, its now fixed. Can you try it again?

Wow nice, you are the best! It works now.

closed 2024-02-01 by onlyreportingissues