RinCat/RTL88x2BU-Linux-Driver #229: DKMS fails - mainline kernel 6.9, Fedora

onlyreportingissues, 2024-03-15:
DKMS make.log for rtl88x2bu-git for kernel 6.9.0-0.rc0.20240314gt480e035f.204.vanilla.fc39.x86_64 (x86_64)
Fr 15. Mär 13:52:42 CET 2024
make ARCH=x86_64 CROSS_COMPILE= -C /lib/modules/6.9.0-0.rc0.20240314gt480e035f.204.vanilla.fc39.x86_64/build M=/var/lib/dkms/rtl88x2bu/git/build  modules
make[1]: Verzeichnis „/usr/src/kernels/6.9.0-0.rc0.20240314gt480e035f.204.vanilla.fc39.x86_64“ wird betreten
  CC [M]  /var/lib/dkms/rtl88x2bu/git/build/core/rtw_cmd.o
  CC [M]  /var/lib/dkms/rtl88x2bu/git/build/core/rtw_security.o
  CC [M]  /var/lib/dkms/rtl88x2bu/git/build/core/rtw_debug.o
  CC [M]  /var/lib/dkms/rtl88x2bu/git/build/core/rtw_io.o
  CC [M]  /var/lib/dkms/rtl88x2bu/git/build/core/rtw_ioctl_query.o
  CC [M]  /var/lib/dkms/rtl88x2bu/git/build/core/rtw_ioctl_set.o
  CC [M]  /var/lib/dkms/rtl88x2bu/git/build/core/rtw_ieee80211.o
  CC [M]  /var/lib/dkms/rtl88x2bu/git/build/core/rtw_mlme.o
  CC [M]  /var/lib/dkms/rtl88x2bu/git/build/core/rtw_mlme_ext.o
  CC [M]  /var/lib/dkms/rtl88x2bu/git/build/core/rtw_mi.o
  CC [M]  /var/lib/dkms/rtl88x2bu/git/build/core/rtw_wlan_util.o
  CC [M]  /var/lib/dkms/rtl88x2bu/git/build/core/rtw_vht.o
In Datei, eingebunden von /var/lib/dkms/rtl88x2bu/git/build/include/drv_types.h:114,
                 von /var/lib/dkms/rtl88x2bu/git/build/core/rtw_cmd.c:17:
/var/lib/dkms/rtl88x2bu/git/build/include/rtw_mlme_ext.h:1181:4: Warnung: kein vorheriger Prototyp für »rtw_mesh_set_plink_state_cmd_hdl« [-Wmissing-prototypes]
 1181 | u8 rtw_mesh_set_plink_state_cmd_hdl(_adapter *adapter, u8 *parmbuf) { return H2C_CMD_FAIL; };
      |    ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
/var/lib/dkms/rtl88x2bu/git/build/core/rtw_ieee80211.c:489:6: Warnung: kein vorheriger Prototyp für »rtw_filter_suppport_rateie« [-Wmissing-prototypes]
  489 | void rtw_filter_suppport_rateie(WLAN_BSSID_EX *pbss_network, u8 keep)
      |      ^~~~~~~~~~~~~~~~~~~~~~~~~~
/var/lib/dkms/rtl88x2bu/git/build/core/rtw_ieee80211.c:766:5: Warnung: kein vorheriger Prototyp für »rtw_get_akm_suite_bitmap« [-Wmissing-prototypes]
  766 | u32 rtw_get_akm_suite_bitmap(u8 *s)
      |     ^~~~~~~~~~~~~~~~~~~~~~~~
/var/lib/dkms/rtl88x2bu/git/build/core/rtw_ieee80211.c:1824:6: Warnung: kein vorheriger Prototyp für »dump_ht_cap_ie« [-Wmissing-prototypes]
 1824 | void dump_ht_cap_ie(void *sel, const u8 *ie, u32 ie_len)
      |      ^~~~~~~~~~~~~~
/var/lib/dkms/rtl88x2bu/git/build/core/rtw_ieee80211.c:1843:6: Warnung: kein vorheriger Prototyp für »dump_ht_op_ie_content« [-Wmissing-prototypes]
 1843 | void dump_ht_op_ie_content(void *sel, const u8 *buf, u32 buf_len)
      |      ^~~~~~~~~~~~~~~~~~~~~
/var/lib/dkms/rtl88x2bu/git/build/core/rtw_ieee80211.c:1857:6: Warnung: kein vorheriger Prototyp für »dump_ht_op_ie« [-Wmissing-prototypes]
 1857 | void dump_ht_op_ie(void *sel, const u8 *ie, u32 ie_len)
      |      ^~~~~~~~~~~~~
  CC [M]  /var/lib/dkms/rtl88x2bu/git/build/core/rtw_pwrctrl.o
/var/lib/dkms/rtl88x2bu/git/build/core/rtw_mlme.c:26:6: Warnung: kein vorheriger Prototyp für »rtw_init_mlme_timer« [-Wmissing-prototypes]
   26 | void rtw_init_mlme_timer(_adapter *padapter)
      |      ^~~~~~~~~~~~~~~~~~~
/var/lib/dkms/rtl88x2bu/git/build/core/rtw_vht.c:43:6: Warnung: kein vorheriger Prototyp für »dump_vht_cap_ie_content« [-Wmissing-prototypes]
   43 | void dump_vht_cap_ie_content(void *sel, const u8 *buf, u32 buf_len)
      |      ^~~~~~~~~~~~~~~~~~~~~~~
/var/lib/dkms/rtl88x2bu/git/build/core/rtw_mlme.c:43:9: Warnung: kein vorheriger Prototyp für »_rtw_init_mlme_priv« [-Wmissing-prototypes]
   43 | sint    _rtw_init_mlme_priv(_adapter *padapter)
      |         ^~~~~~~~~~~~~~~~~~~
/var/lib/dkms/rtl88x2bu/git/build/core/rtw_vht.c:83:6: Warnung: kein vorheriger Prototyp für »dump_vht_op_ie_content« [-Wmissing-prototypes]
   83 | void dump_vht_op_ie_content(void *sel, const u8 *buf, u32 buf_len)
      |      ^~~~~~~~~~~~~~~~~~~~~~
/var/lib/dkms/rtl88x2bu/git/build/core/rtw_mlme.c:329:6: Warnung: kein vorheriger Prototyp für »_rtw_free_mlme_priv« [-Wmissing-prototypes]
  329 | void _rtw_free_mlme_priv(struct mlme_priv *pmlmepriv)
      |      ^~~~~~~~~~~~~~~~~~~
/var/lib/dkms/rtl88x2bu/git/build/core/rtw_mlme.c:357:9: Warnung: kein vorheriger Prototyp für »_rtw_enqueue_network« [-Wmissing-prototypes]
  357 | sint    _rtw_enqueue_network(_queue *queue, struct wlan_network *pnetwork)
      |         ^~~~~~~~~~~~~~~~~~~~
/var/lib/dkms/rtl88x2bu/git/build/core/rtw_vht.c:370:6: Warnung: kein vorheriger Prototyp für »update_sta_vht_info_apmode_bf_cap« [-Wmissing-prototypes]
  370 | void update_sta_vht_info_apmode_bf_cap(_adapter *padapter, struct sta_info *psta)
      |      ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
/var/lib/dkms/rtl88x2bu/git/build/core/rtw_mlme_ext.c:627:5: Warnung: kein vorheriger Prototyp für »rtw_chset_get_ch_non_ocp_ms« [-Wmissing-prototypes]
  627 | u32 rtw_chset_get_ch_non_ocp_ms(RT_CHANNEL_INFO *ch_set, u8 ch, u8 bw, u8 offset)
      |     ^~~~~~~~~~~~~~~~~~~~~~~~~~~
/var/lib/dkms/rtl88x2bu/git/build/core/rtw_mi.c:877:4: Warnung: kein vorheriger Prototyp für »_rtw_mi_busy_traffic_check« [-Wmissing-prototypes]
  877 | u8 _rtw_mi_busy_traffic_check(_adapter *padapter, void *data)
      |    ^~~~~~~~~~~~~~~~~~~~~~~~~~
/var/lib/dkms/rtl88x2bu/git/build/core/rtw_cmd.c:1634:6: Warnung: kein vorheriger Prototyp für »free_assoc_resources_hdl« [-Wmissing-prototypes]
 1634 | void free_assoc_resources_hdl(_adapter *padapter, u8 lock_scanned_queue)
      |      ^~~~~~~~~~~~~~~~~~~~~~~~
/var/lib/dkms/rtl88x2bu/git/build/core/rtw_wlan_util.c:937:6: Warnung: kein vorheriger Prototyp für »_rtw_sec_camid_is_used« [-Wmissing-prototypes]
  937 | bool _rtw_sec_camid_is_used(struct cam_ctl_t *cam_ctl, u8 id)
      |      ^~~~~~~~~~~~~~~~~~~~~~
[... 444 lines cut ...]
 1245 | void dhcp_flag_bcast(_adapter *priv, struct sk_buff *skb)
      |      ^~~~~~~~~~~~~~~
/var/lib/dkms/rtl88x2bu/git/build/core/rtw_br_ext.c:1286:7: Warnung: kein vorheriger Prototyp für »scdb_findEntry« [-Wmissing-prototypes]
 1286 | void *scdb_findEntry(_adapter *priv, unsigned char *macAddr,
      |       ^~~~~~~~~~~~~~
  CC [M]  /var/lib/dkms/rtl88x2bu/git/build/core/rtw_rm_fsm.o
  CC [M]  /var/lib/dkms/rtl88x2bu/git/build/core/rtw_ft.o
/var/lib/dkms/rtl88x2bu/git/build/core/rtw_sreset.c:104:6: Warnung: kein vorheriger Prototyp für »sreset_restore_security_station« [-Wmissing-prototypes]
  104 | void sreset_restore_security_station(_adapter *padapter)
      |      ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
/var/lib/dkms/rtl88x2bu/git/build/core/rtw_sreset.c:140:6: Warnung: kein vorheriger Prototyp für »sreset_restore_network_station« [-Wmissing-prototypes]
  140 | void sreset_restore_network_station(_adapter *padapter)
      |      ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
/var/lib/dkms/rtl88x2bu/git/build/core/rtw_sreset.c:200:6: Warnung: kein vorheriger Prototyp für »sreset_restore_network_status« [-Wmissing-prototypes]
  200 | void sreset_restore_network_status(_adapter *padapter)
      |      ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~
  CC [M]  /var/lib/dkms/rtl88x2bu/git/build/core/rtw_wnm.o
  CC [M]  /var/lib/dkms/rtl88x2bu/git/build/core/rtw_mbo.o
/var/lib/dkms/rtl88x2bu/git/build/core/rtw_btcoex.c:409:6: Warnung: kein vorheriger Prototyp für »rtw_btcoex_SetBtPatchVersion« [-Wmissing-prototypes]
  409 | void rtw_btcoex_SetBtPatchVersion(PADAPTER padapter, u16 btHciVer, u16 btPatchVer)
      |      ^~~~~~~~~~~~~~~~~~~~~~~~~~~~
/var/lib/dkms/rtl88x2bu/git/build/core/rtw_btcoex.c:414:6: Warnung: kein vorheriger Prototyp für »rtw_btcoex_SetHciVersion« [-Wmissing-prototypes]
  414 | void rtw_btcoex_SetHciVersion(PADAPTER  padapter, u16 hciVersion)
      |      ^~~~~~~~~~~~~~~~~~~~~~~~
/var/lib/dkms/rtl88x2bu/git/build/core/rtw_btcoex.c:419:6: Warnung: kein vorheriger Prototyp für »rtw_btcoex_StackUpdateProfileInfo« [-Wmissing-prototypes]

onlyreportingissues, 2024-03-15:
419 | void rtw_btcoex_StackUpdateProfileInfo(void)
      |      ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
  CC [M]  /var/lib/dkms/rtl88x2bu/git/build/core/rtw_rm_util.o
  CC [M]  /var/lib/dkms/rtl88x2bu/git/build/core/efuse/rtw_efuse.o
  CC [M]  /var/lib/dkms/rtl88x2bu/git/build/core/rtw_roch.o
/var/lib/dkms/rtl88x2bu/git/build/core/rtw_odm.c:67:6: Warnung: kein vorheriger Prototyp für »rtw_odm_adaptivity_ver_msg« [-Wmissing-prototypes]
   67 | void rtw_odm_adaptivity_ver_msg(void *sel, _adapter *adapter)
      |      ^~~~~~~~~~~~~~~~~~~~~~~~~~
/var/lib/dkms/rtl88x2bu/git/build/core/rtw_odm.c:75:6: Warnung: kein vorheriger Prototyp für »rtw_odm_adaptivity_en_msg« [-Wmissing-prototypes]
   75 | void rtw_odm_adaptivity_en_msg(void *sel, _adapter *adapter)
      |      ^~~~~~~~~~~~~~~~~~~~~~~~~
/var/lib/dkms/rtl88x2bu/git/build/core/rtw_odm.c:92:6: Warnung: kein vorheriger Prototyp für »rtw_odm_adaptivity_mode_msg« [-Wmissing-prototypes]
   92 | void rtw_odm_adaptivity_mode_msg(void *sel, _adapter *adapter)
      |      ^~~~~~~~~~~~~~~~~~~~~~~~~~~
  CC [M]  /var/lib/dkms/rtl88x2bu/git/build/core/crypto/aes-internal.o
  CC [M]  /var/lib/dkms/rtl88x2bu/git/build/core/crypto/aes-internal-enc.o
  CC [M]  /var/lib/dkms/rtl88x2bu/git/build/core/crypto/aes-gcm.o
  CC [M]  /var/lib/dkms/rtl88x2bu/git/build/core/crypto/aes-ccm.o
  CC [M]  /var/lib/dkms/rtl88x2bu/git/build/core/crypto/aes-omac1.o
  CC [M]  /var/lib/dkms/rtl88x2bu/git/build/core/crypto/ccmp.o
  CC [M]  /var/lib/dkms/rtl88x2bu/git/build/core/crypto/gcmp.o
  CC [M]  /var/lib/dkms/rtl88x2bu/git/build/core/crypto/aes-siv.o
  CC [M]  /var/lib/dkms/rtl88x2bu/git/build/core/crypto/aes-ctr.o
/var/lib/dkms/rtl88x2bu/git/build/core/efuse/rtw_efuse.c:60:9: Warnung: kein vorheriger Prototyp für »efuse_IsBT_Masked« [-Wmissing-prototypes]
   60 | BOOLEAN efuse_IsBT_Masked(PADAPTER pAdapter, u16 Offset)
      |         ^~~~~~~~~~~~~~~~~
/var/lib/dkms/rtl88x2bu/git/build/core/efuse/rtw_efuse.c:985:6: Warnung: kein vorheriger Prototyp für »efuse_PreUpdateAction« [-Wmissing-prototypes]
  985 | void efuse_PreUpdateAction(
      |      ^~~~~~~~~~~~~~~~~~~~~
/var/lib/dkms/rtl88x2bu/git/build/core/efuse/rtw_efuse.c:1014:6: Warnung: kein vorheriger Prototyp für »efuse_PostUpdateAction« [-Wmissing-prototypes]
 1014 | void efuse_PostUpdateAction(
      |      ^~~~~~~~~~~~~~~~~~~~~~
  CC [M]  /var/lib/dkms/rtl88x2bu/git/build/core/crypto/sha256-internal.o
  CC [M]  /var/lib/dkms/rtl88x2bu/git/build/core/crypto/sha256.o
  CC [M]  /var/lib/dkms/rtl88x2bu/git/build/core/crypto/sha256-prf.o
  CC [M]  /var/lib/dkms/rtl88x2bu/git/build/core/crypto/rtw_crypto_wrap.o
  CC [M]  /var/lib/dkms/rtl88x2bu/git/build/core/rtw_swcrypto.o
  CC [M]  /var/lib/dkms/rtl88x2bu/git/build/os_dep/osdep_service.o
/var/lib/dkms/rtl88x2bu/git/build/core/crypto/ccmp.c:179:6: Warnung: kein vorheriger Prototyp für »ccmp_get_pn« [-Wmissing-prototypes]
  179 | void ccmp_get_pn(u8 *pn, const u8 *data)
      |      ^~~~~~~~~~~
  CC [M]  /var/lib/dkms/rtl88x2bu/git/build/os_dep/linux/os_intfs.o
  CC [M]  /var/lib/dkms/rtl88x2bu/git/build/os_dep/linux/usb_intf.o
  CC [M]  /var/lib/dkms/rtl88x2bu/git/build/os_dep/linux/usb_ops_linux.o
  CC [M]  /var/lib/dkms/rtl88x2bu/git/build/os_dep/linux/ioctl_linux.o
  CC [M]  /var/lib/dkms/rtl88x2bu/git/build/os_dep/linux/xmit_linux.o
  CC [M]  /var/lib/dkms/rtl88x2bu/git/build/os_dep/linux/mlme_linux.o
  CC [M]  /var/lib/dkms/rtl88x2bu/git/build/os_dep/linux/recv_linux.o
  CC [M]  /var/lib/dkms/rtl88x2bu/git/build/os_dep/linux/ioctl_cfg80211.o
  CC [M]  /var/lib/dkms/rtl88x2bu/git/build/os_dep/linux/rtw_cfgvendor.o
  CC [M]  /var/lib/dkms/rtl88x2bu/git/build/os_dep/linux/wifi_regd.o
/var/lib/dkms/rtl88x2bu/git/build/core/rtw_swcrypto.c:30:5: Warnung: kein vorheriger Prototyp für »_rtw_ccmp_encrypt« [-Wmissing-prototypes]
   30 | int _rtw_ccmp_encrypt(_adapter *padapter, u8 *key, u32 key_len, uint hdrlen, u8 *frame, uint plen)
      |     ^~~~~~~~~~~~~~~~~
/var/lib/dkms/rtl88x2bu/git/build/core/rtw_swcrypto.c:71:5: Warnung: kein vorheriger Prototyp für »_rtw_ccmp_decrypt« [-Wmissing-prototypes]
   71 | int _rtw_ccmp_decrypt(_adapter * padapter, u8 *key, u32 key_len, uint hdrlen, u8 *frame,
      |     ^~~~~~~~~~~~~~~~~
/var/lib/dkms/rtl88x2bu/git/build/core/rtw_swcrypto.c:134:5: Warnung: kein vorheriger Prototyp für »_rtw_gcmp_encrypt« [-Wmissing-prototypes]
  134 | int _rtw_gcmp_encrypt(_adapter * padapter, u8 *key, u32 key_len, uint hdrlen, u8 *frame, uint plen)
      |     ^~~~~~~~~~~~~~~~~
/var/lib/dkms/rtl88x2bu/git/build/core/rtw_swcrypto.c:165:5: Warnung: kein vorheriger Prototyp für »_rtw_gcmp_decrypt« [-Wmissing-prototypes]
  165 | int _rtw_gcmp_decrypt(_adapter *padapter, u8 *key, u32 key_len, uint hdrlen, u8 *frame, uint plen)
      |     ^~~~~~~~~~~~~~~~~
/var/lib/dkms/rtl88x2bu/git/build/core/rtw_swcrypto.c:196:4: Warnung: kein vorheriger Prototyp für »_bip_ccmp_protect« [-Wmissing-prototypes]
  196 | u8 _bip_ccmp_protect(const u8 *key, size_t key_len,
      |    ^~~~~~~~~~~~~~~~~
/var/lib/dkms/rtl88x2bu/git/build/core/rtw_swcrypto.c:220:4: Warnung: kein vorheriger Prototyp für »_bip_gcmp_protect« [-Wmissing-prototypes]
  220 | u8 _bip_gcmp_protect(u8 *whdr_pos, size_t len,
      |    ^~~~~~~~~~~~~~~~~
  CC [M]  /var/lib/dkms/rtl88x2bu/git/build/os_dep/linux/rtw_android.o
[... 394 lines cut ...]
/var/lib/dkms/rtl88x2bu/git/build/hal/hal_com_phycfg.c:4613:1: Warnung: kein vorheriger Prototyp für »phy_DecryptBBPgParaFile« [-Wmissing-prototypes]
[... cut ...]
/var/lib/dkms/rtl88x2bu/git/build/hal/hal_com_phycfg.c:5880:4: Warnung: kein vorheriger Prototyp für »phy_get_txpwr_tpc« [-Wmissing-prototypes]
 5880 | s8 phy_get_txpwr_tpc(_adapter *adapter, struct hal_spec_t *hal_spec)
      |    ^~~~~~~~~~~~~~~~~
make[2]: *** [/usr/src/kernels/6.9.0-0.rc0.20240314gt480e035f.204.vanilla.fc39.x86_64/Makefile:1923: /var/lib/dkms/rtl88x2bu/git/build] Fehler 2
make[1]: *** [Makefile:242: __sub-make] Fehler 2
make[1]: Verzeichnis „/usr/src/kernels/6.9.0-0.rc0.20240314gt480e035f.204.vanilla.fc39.x86_64“ wird verlassen
make: *** [Makefile:2470: modules] Fehler 2

RinCat, 2024-03-28:
https://github.com/RinCat/RTL88x2BU-Linux-Driver/commit/358f13d1749cac4b314c8fa4187f65dfa7ef1813

closed 2024-03-28 by RinCat