RinCat/RTL88x2BU-Linux-Driver #230: Unable to install driver on ubuntu 22.04

ivan-marroquin, 2024-03-19:
Many thanks for making this package available!

I have this version of ubuntu
Description:	Ubuntu 22.04.4 LTS
Release:	22.04
Codename:	jammy
kernel version 6.5.0-1018-oracle and architecture amd64

When I follow the instructions to install the driver with dkms command, I get this error message:
Error! Your kernel headers for kernel 6.5.0-1018-oracle cannot be found.
Please install the linux-headers-6.5.0-1018-oracle package or use the --kernelsourcedir option to tell DKMS where it's located.
Error! Your kernel headers for kernel 6.5.0-1018-oracle cannot be found.
Please install the linux-headers-6.5.0-1018-oracle package or use the --kernelsourcedir option to tell DKMS where it's located.

I searched on internet and found the packages shown on the attached screen shot. They did not solve the installation issue.

Any suggestions?

Kind regards,
Ivan
[image]

RinCat, 2024-03-19:
Hi, you installed both and DKMS still cannot found the headers?

ivan-marroquin, 2024-03-20:
Thanks for your quick response. I installed both debian files using this command; 

sudo dpkg -i package-file.deb

What I noticed is none of these deb files generated a 'build' folder under the linux headers for 6.5.0-1018-oracle. I attached a screen shot to show the folders available.

Thanks

[image]

ivan-marroquin, 2024-03-20:
Then, I looked to the folder 6.5.0-25-generic folder and it has a 'build' sub-folder (see screenshot). Should I use this folder to compile the driver? If so, how can I redirect dkms to search for this folder instead of the 'oracle' one?

[image]

RinCat, 2024-03-20:
Hi, can you try install from apt? Use the following commands:
`sudo apt update && sudo apt install linux-headers-oracle`

RinCat, 2024-03-20:
> Then, I looked to the folder 6.5.0-25-generic folder and it has a 'build' sub-folder (see screenshot). Should I use this folder to compile the driver? If so, how can I redirect dkms to search for this folder instead of the 'oracle' one?

No, you cannot use other kernel headers to build kernel modules, they are bonded.

ivan-marroquin, 2024-03-21:
Unfortunately, I don't have internet access from my linux PC

RinCat, 2024-03-21:
Do you have `/usr/src/linux-headers-6.5.0-1018-oracle/` ?

ivan-marroquin, 2024-03-21:
I do. The folder and its content are shown below. I hope these files are compatible with current linux headers

[image]

RinCat, 2024-03-21:
Try:
`sudo mkdir -p /lib/modules/6.5.0-1019-oracle && sudo ln -s /usr/src/linux-headers-6.5.0-1019-oracle/ /lib/modules/6.5.0-1019-oracle/build`

And install again

ivan-marroquin, 2024-03-21:
Since this folder already exists "/lib/modules/6.5.0-1018-oracle"

I just generated a symbolic link as follows:
sudo ln -s /usr/src/linux-oracle-6.5-headers-6.5.0-1018/ build

I ran the dkms autoinstall command, and got this error message

Kernel preparation unnecessary for this kernel. Skipping...

Building module:
cleaning build area...
unset ARCH; [ ! -h /usr/bin/cc ] && export CC=/usr/bin/gcc; env NV_VERBOSE=1 'make' -j4 NV_EXCLUDE_BUILD_MODULES='' KERNEL_UNAME=6.5.0-1018-oracle IGNORE_XEN_PRESENCE=1 IGNORE_CC_MISMATCH=1 SYSSRC=/lib/modules/6.5.0-1018-oracle/build LD=/usr/bin/ld.bfd CONFIG_X86_KERNEL_IBT= modules...(bad exit status: 2)
Error! Bad return status for module build on kernel: 6.5.0-1018-oracle (x86_64)
Consult /var/lib/dkms/nvidia/535.161.07/build/make.log for more information.

Kernel preparation unnecessary for this kernel. Skipping...

Building module:
cleaning build area...
'make' -j4 KVER=6.5.0-1018-oracle KSRC=/lib/modules/6.5.0-1018-oracle/build...(bad exit status: 2)
ERROR (dkms apport): binary package for rtl88x2bu: git not found
Error! Bad return status for module build on kernel: 6.5.0-1018-oracle (x86_64)
Consult /var/lib/dkms/rtl88x2bu/git/build/make.log for more information.

I attached the make.log file for your reference.

Thanks
[make.zip](https://github.com/RinCat/RTL88x2BU-Linux-Driver/files/14679950/make.zip)

RinCat, 2024-03-21:
Do you currently have to use the oracle version of the kernel? Use the standard kernel should allow you to use internet, then you will have more time to fix those issues / install packages.

ivan-marroquin, 2024-03-21:
How would you suggest to switch kernels? will the switch disrupt the grub configuration file?

RinCat, 2024-03-21:
It will coexists with your current kernels, so grub should works fine unless you modified it manually.

You will need to install `linux-image-6.5.0-26-generic` and  `linux-headers-6.5.0-26-generic` at least.

ivan-marroquin, 2024-03-26:
Switching to a generic version of the kernel seemed to solve the issue.

In case someone faces the same issue, I did the following:
- check there are generic version of the kernel available. find the latest generic version
- go to path '/lib/modules/latest generic version/' and confirm there is a 'build' sub-folder is not empty
- grub-customizer to replace the calls to kernel oracle version and replace by the latest generic version
- run 'sudo update-grub' and 'sudo reboot'
- compile the driver using the manual approach with dkms command

Thanks for your help

closed 2024-03-26 by ivan-marroquin