RinCat/RTL88x2BU-Linux-Driver #231: Installation to a non-current kernel

barryrowlingson, 2024-04-08:
Ubuntu has just installed a kernel update and I'm trying to recompile and set up the network module so it works on my next boot. 
Currently I'm on 5.15.0-100-generic but the next boot will go to 5.15.0-102-generic based on my latest modified grub.cfg file.

Compiling the kernel module with `make KSRC=/lib/modules/5.15.0-102-generic/build` works fine but the next step seems to want to install in the current kernel:

```
# make -n install
install -p -m 644 88x2bu.ko  /lib/modules/5.15.0-100-generic/kernel/drivers/net/wireless/
/sbin/depmod -a 5.15.0-100-generic
```

Looking at the Makefile it seems to get the KVER and MODDESTDIR from `uname -r` 

```
KVER  := $(shell uname -r)
KSRC := /lib/modules/$(KVER)/build
MODDESTDIR := /lib/modules/$(KVER)/kernel/drivers/net/wireless/
```
```
install:
	install -p -m 644 $(MODULE_NAME).ko  $(MODDESTDIR)
	/sbin/depmod -a ${KVER}
```
and I don't see an easy way to specify the new kernel when running make.

Obvious fix is to manually run the install and depmod commands manually with the correct settings, but can I assume depmod works for modules outside a running kernel? 

Or is dkms marvellous at fixing these things nowadays?

RinCat, 2024-04-11:
Hi, if you use `make install KVER=5.15.0-102-generic`, it should work?

barryrowlingson, 2024-04-11:
Yes, I can actually use `make clean; make KVER=5.15.0-100-generic` and `make install KVER=5.15.0-100-generic`, using `KVER` instead of specifying `KSRC` for the initial make build step as per the docs.

 I hadn't realised you could override these assignments with parameters on the command line like that, I did try `KVER=... make` thinking it might take the value from the environment but that only works if you have `-e` on the command line. I learn.