RinCat/RTL88x2BU-Linux-Driver #232: My ASUS USB-AC58 doesn't see 5Ghz networks with this driver

Mracobes9, 2024-05-15:
Hi. I installed this driver because i had disconnect issues with in-kenel. It driver solved it, but i can find and connect 5Ghz networks. Dmesg show the following logs:
```
[  723.212912] RTW: rtw_ndev_uninit(wlan0) if1
[  723.249439] RTW: rtw_dev_unload: driver in IPS-FWLPS
[  723.249445] rtl88x2bu 3-2:1.0: Runtime PM usage count underflow!
[  723.249559] RTW: WARN free_recv_skb_queue not empty, 1
[  733.216609] usb 3-2: new high-speed USB device number 4 using xhci_hcd
[  733.346344] usb 3-2: New USB device found, idVendor=0b05, idProduct=19aa, bcdDevice= 2.10
[  733.346349] usb 3-2: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[  733.346351] usb 3-2: Product: ASUS USB-AC58 USB Wireless adapter
[  733.346352] usb 3-2: Manufacturer: Asus
[  733.346353] usb 3-2: SerialNumber: 123
[  733.349008] RTW: [HALMAC]55772M
HALMAC_MAJOR_VER = 1
HALMAC_PROTOTYPE_VER = 6
HALMAC_MINOR_VER = 6
HALMAC_PATCH_VER = 23
[  733.354593] RTW: ERROR [HALMAC][ERR]Dump efuse in suspend
[  734.893604] RTW: HW EFUSE
[  734.893607] RTW: 0x000: 29 81 00 3C  09 00 A1 00  B6 04 64 10  00 00 A3 00
[  734.893616] RTW: 0x010: 2C 2C 2D 2D  2F 2F 2F 2F  2E 2F 2F 02  00 00 FF FF
[  734.893625] RTW: 0x020: FF FF 20 20  1C 1C 1D 1D  1D 1D 1D 1D  24 24 24 24
[  734.893633] RTW: 0x030: 02 00 FF FF  00 FF 00 00  FF FF 30 30  30 30 2F 2F
[  734.893641] RTW: 0x040: 2F 2F 30 2F  2F 02 00 00  FF FF FF FF  1E 1E 1E 1E
[  734.893650] RTW: 0x050: 20 20 20 1C  1C 1C 23 23  23 23 02 00  FF FF 00 FF
[  734.893658] RTW: 0x060: 00 00 FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF
[  734.893666] RTW: 0x070: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF
[  734.893674] RTW: 0x080: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF
[  734.893682] RTW: 0x090: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF
[  734.893691] RTW: 0x0A0: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF
[  734.893699] RTW: 0x0B0: FF FF FF FF  FF FF FF FF  26 29 19 00  FF FF FF FF
[  734.893707] RTW: 0x0C0: FF 09 00 11  00 00 00 00  00 FF 02 FF  FF FF FF FF
[  734.893715] RTW: 0x0D0: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF
[  734.893724] RTW: 0x0E0: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF
[  734.893732] RTW: 0x0F0: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF
[  734.893740] RTW: 0x100: 05 0B AA 19  E1 67 02 E8  9C 25 61 14  53 06 03 41
[  734.893749] RTW: 0x110: 73 75 73 24  03 41 53 55  53 20 55 53  42 2D 41 43
[  734.893757] RTW: 0x120: 35 38 20 55  53 42 20 57  69 72 65 6C  65 73 73 20
[  734.893765] RTW: 0x130: 61 64 61 70  74 65 72 05  03 31 32 33  FF FF FF 34
[  734.893773] RTW: 0x140: 33 0A 1F 01  00 00 21 0F  FF FF FF FF  FF FF FF FF
[  734.893782] RTW: 0x150: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF
[  734.893790] RTW: 0x160: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF
[  734.893798] RTW: 0x170: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF
[  734.893806] RTW: 0x180: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF
[  734.893815] RTW: 0x190: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF
[  734.893823] RTW: 0x1A0: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF
[  734.893831] RTW: 0x1B0: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF
[  734.893839] RTW: 0x1C0: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF
[  734.893847] RTW: 0x1D0: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF
[  734.893856] RTW: 0x1E0: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF
[  734.893864] RTW: 0x1F0: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF
[  734.893872] RTW: 0x200: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF
[  734.893880] RTW: 0x210: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF
[  734.893889] RTW: 0x220: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF
[  734.893897] RTW: 0x230: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF
[  734.893905] RTW: 0x240: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF
[  734.893913] RTW: 0x250: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF
[  734.893922] RTW: 0x260: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF
[  734.893930] RTW: 0x270: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF
[  734.893938] RTW: 0x280: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF
[  734.893946] RTW: 0x290: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF
[  734.893955] RTW: 0x2A0: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF
[  734.893963] RTW: 0x2B0: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF
[  734.893971] RTW: 0x2C0: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF
[  734.893979] RTW: 0x2D0: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF
[  734.893987] RTW: 0x2E0: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF
[  734.893996] RTW: 0x2F0: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF
[  734.894006] RTW: hal_com_config_channel_plan chplan:0x26
[  734.894008] RTW: ERROR [HALMAC][ERR]Dump efuse in suspend
[  734.938221] RTW: [HALMAC][ALWAYS]shall R reg twice!!
[... 201 lines cut ...]
ID-2: swap-2 type: zram size: 14.81 GiB used: 0 KiB (0.0%) priority: 100
comp: zstd avail: lzo,lzo-rle,lz4,lz4hc,842 max-streams: 12 dev: /dev/zram0
Sensors:
System Temperatures: cpu: 37.5 C mobo: N/A
Fan Speeds (rpm): N/A
GPU: device: nvidia screen: :0.0 temp: 38 C fan: 0% device: amdgpu
temp: 34.0 C
Info:
Memory: total: 16 GiB note: est. available: 14.82 GiB used: 3.93 GiB (26.5%)
Processes: 379 Power: uptime: 18m states: freeze,mem,disk suspend: deep
avail: s2idle wakeups: 0 hibernate: platform avail: shutdown, reboot,
suspend, test_resume image: 5.88 GiB services: org_kde_powerdevil,
power-profiles-daemon, upowerd Init: systemd v: 255 default: graphical
tool: systemctl
Packages: pm: pacman pkgs: 1942 libs: 574 tools: octopi,pamac,paru
Compilers: clang: 17.0.6 gcc: 14.1.1 alt: 13 Shell: garuda-inxi
default: fish v: 3.7.1 running-in: konsole inxi: 3.3.34
Garuda (2.6.26-1):
System install date:     2024-05-11
Last full system update: 2024-05-15
Is partially upgraded:   No
Relevant software:       snapper NetworkManager dracut nvidia-dkms
Windows dual boot:       Probably (Run as root to verify)
Failed units:            nmb.service
```

Mracobes9, 2024-05-15:
> Hi, I am not sure which distro you are using, but maybe you can try use iwd? https://wiki.debian.org/NetworkManager/iwd

My distro is garuda linux. It is arch based. I tried to use iwd, but it does not helped

Mracobes9, 2024-05-15:
I temporary fixed it by setting router channel to 40 for 5Ghz networks, but maybe it can to fix by driver settings ?

RinCat, 2024-05-16:
> I temporary fixed it by setting router channel to 40 for 5Ghz networks, but maybe it can to fix by driver settings ?

That seems you are try to use restricted channels? 

You can try `iw reg set [COUNTRY CODE]` like https://en.wikipedia.org/wiki/ISO_3166-1_alpha-2

fatihmete, 2024-06-22:
Same issue with tp-link archer t3u on fedora 40. When I installed module everything was good. But after reboot system, network manager (cinnamon applet) shows only 2.4 networks. I have changed NetworkManager backend to iwd. Now it looks working.
- Fedora iwd installation: https://nullr0ute.com/2023/01/using-iwd-for-wifi-in-fedora/ (option1 is safe)
- Convert known stations iwd format and fix auto connect issue: https://iwd.wiki.kernel.org/networkmanager#converting_network_profiles
- Fix Restarting iwd.service after boot: https://wiki.archlinux.org/title/iwd