RinCat/RTL88x2BU-Linux-Driver #233: Zorin OS 17.1 Core (Archer T2U Plus [RTL8821AU]) Cannot See Any Available Connections

Outspending, 2024-05-26:
So i did the installation as normal, and Ubuntu does detect the driver because before it said there's no available adapters when opening the wifi menu, now it tries to find connections although none appear.

Here's my `dkms status`:
[image]

Also my `lsusb`:
[image]

The Wifi Panel:
[image]

RinCat, 2024-05-26:
Hi, this driver only works for RTL88x2**B**U. You will need to find AU version for your T2U Plus. They are different chips.

Outspending, 2024-05-26:
OH that's my bad, i swear i've used this driver before with this chip. Totally my bad :skull: thank you!

closed 2024-05-26 by RinCat