RinCat/RTL88x2BU-Linux-Driver #236: Linux 6.1.92 - wlan0 losing it's IP address

chrisvollorenew, 2024-06-17:
Hello, I am running 6.1.92 and using commit f3df449fbfb5d41974fcb192c9eaf9fa43ffb1b0 (latest at time of writing).

I manually assign an IP address to the adapter using the command "ip addr ..." , then I launch hostapd so the adapter is in AP mode. 

After some time, the adapter loses it's IP address, and I see this in the logs:

```
[  307.095448] RTW: WARN cfg80211_rtw_scan (wlan0) : scan abort!! BusyTraffic
[  428.056906] RTW: ERROR Free disconnecting network of scanned_queue failed due to pwlan == NULL
[  428.056906] 
[  428.069413] RTW: WARN _beamforming_leave: d2:6d:ac:f7:33:63 is neither beamforming ee or er!!
[  428.082159] RTW: ERROR Free disconnecting network of scanned_queue failed due to pwlan == NULL
[  428.082159] 
[  428.095999] RTW: WARN _beamforming_leave: 76:ae:06:4f:30:12 is neither beamforming ee or er!!
[  428.106078] RTW: WARN _beamforming_leave: 76:ae:06:4f:30:12 is neither beamforming ee or er!!
[  428.114786] RTW: ERROR Free disconnecting network of scanned_queue failed due to pwlan == NULL
[  428.114786] 
[  428.202192] RTW: assoc success
[  428.206670] IPv6: ADDRCONF(NETDEV_CHANGE): wlan0: link becomes ready
```

Why is this happening?

chrisvollorenew, 2024-06-17:
Ok, I did a bit more testing and oddly this only appears to be the case when I have the `networkmanager` package installed. If I don't install networkmanager then the adapter keeps it's ip address. Any ideas why? I would still be nice to fix it if we can

chrisvollorenew, 2024-06-17:
Ok, it appears this is probably entirely related to the NetworkManager service. If I tell it to ignore wlan0 it fixes the issue. It is still a mystery to me why it happens exactly at the same timeout as the ap_max_inactivity=300 ...

closed 2024-06-17 by chrisvollorenew

RinCat, 2024-06-18:
Hi, that sounds like you did not disable DHCP in NetworkManager.