RinCat/RTL88x2BU-Linux-Driver #237: Mercusys MA30H

nicolascolla, 2024-07-11:
Hello, @RinCat and thank you very much for maintaining this fork.

I have a Mercusys MA30H, and the official drivers page (https://www.mercusys.com/en/download/ma30h/v2) offers the sources of the RTL88x2BU driver (that fail to build on newer kernels), so I'm pretty sure this is a RTL88x2BU device. However, after building the driver from this repo, the Wi-Fi dongle is still not recognized.

dmesg output:
```
[  642.056301] usb 1-1.2: new high-speed USB device number 4 using ehci-pci
[  642.166212] usb 1-1.2: New USB device found, idVendor=2c4e, idProduct=0107, bcdDevice= 2.10
[  642.166230] usb 1-1.2: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[  642.166237] usb 1-1.2: Product: 802.11ac NIC
[  642.166241] usb 1-1.2: Manufacturer: Realtek
[  642.166245] usb 1-1.2: SerialNumber: 123456
```

lsusb output:
```
Bus 001 Device 004: ID 2c4e:0107 Mercucys INC 802.11ac NIC

```
Do you know if there's any way in which I could get it working?
Thank you very much!

RinCat, 2024-07-11:
Hi @nicolascolla, this one is new USB device ID, so I will need to add it.

Also congratulations on finding a new driver version. I have not seen a new version of this driver for a long time. I will try to update it today.

RinCat, 2024-07-11:
Update the driver, now it should works. https://github.com/RinCat/RTL88x2BU-Linux-Driver/commit/8159d9825f645b4abf10f73fc25787155370efd2

nicolascolla, 2024-07-11:
Hello, @RinCat, and thanks a lot for the quick response. Unfortunately, it seems like my dongle is still not being detected. It still appears as a regular USB device and no network interface is added. Do you know why that might be the case? And is there anything else I could test?

nicolascolla, 2024-07-11:
```
$ lsmod
Module                  Size  Used by
...
88x2bu               3657728  0
...
```
It seems like, even if the module is loaded, it's not being used by the dongle.

RinCat, 2024-07-11:
Did you deleted old module and rebuild it from the new source?
Run `dmesg -W` as root, then plug in the USB dongle should shows something

nicolascolla, 2024-07-11:
> Did you deleted old module and rebuild it from the new source? Run `dmesg -W` as root, then plug in the USB dongle should shows something

I ran ```sudo make uninstall``` from the source folder, then cloned the new version of the repo and recompiled.

This is what dmesg shows when the dongle is plugged in:

```
[  323.872334] usb 1-1.2: new high-speed USB device number 3 using ehci-pci
[  323.982236] usb 1-1.2: New USB device found, idVendor=2c4e, idProduct=0107, bcdDevice= 2.10
[  323.982255] usb 1-1.2: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[  323.982261] usb 1-1.2: Product: 802.11ac NIC
[  323.982266] usb 1-1.2: Manufacturer: Realtek
[  323.982270] usb 1-1.2: SerialNumber: 123456
[ 1316.726831] usb 1-1.2: USB disconnect, device number 3
[ 1321.322587] usb 1-1.2: new high-speed USB device number 4 using ehci-pci
[ 1321.436576] usb 1-1.2: New USB device found, idVendor=2c4e, idProduct=0107, bcdDevice= 2.10
[ 1321.436595] usb 1-1.2: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[ 1321.436601] usb 1-1.2: Product: 802.11ac NIC
[ 1321.436606] usb 1-1.2: Manufacturer: Realtek
[ 1321.436610] usb 1-1.2: SerialNumber: 123456
```

RinCat, 2024-07-11:
Did you removed the old loaded module by run `modprobe -r 88x2bu`?

nicolascolla, 2024-07-11:
> Did you removed the old loaded module by run `modprobe -r 88x2bu`?

Instead of doing that, I just restarted the machine.

RinCat, 2024-07-11:
remove this file and rebuild again `/lib/modules/$(uname -r)/kernel/drivers/net/wireless/88x2bu.ko`

nicolascolla, 2024-07-11:
No luck, unfortunately. Steps taken:

```
$ sudo rm /lib/modules/$(uname -r)/kernel/drivers/net/wireless/88x2bu.ko
$ cd src/RTL88x2BU-Linux-Driver/
$ make -j4
...
$ sudo make install
install -p -m 644 88x2bu.ko  /lib/modules/5.15.0-113-generic/kernel/drivers/net/wireless/
/sbin/depmod -a 5.15.0-113-generic
$ sudo modprobe -r 88x2bu
```

The dongle still shows up as a regular USB device, as in https://github.com/RinCat/RTL88x2BU-Linux-Driver/issues/237#issuecomment-2224117648

RinCat, 2024-07-11:
Can you run `grep -r 0x0107` in the driver dir and see it report the usb id there?

nicolascolla, 2024-07-11:
Yes, the changes of the last commit are there:
```
$ grep -r 0x0107
include/rtw_btcoex.h:	HCI_BT_COEX_NOTIFY								= 0x0107,
os_dep/linux/usb_intf.c:	{USB_DEVICE_AND_INTERFACE_INFO(0x2c4e, 0x0107, 0xff, 0xff, 0xff), .driver_info = RTL8822B}, /* Mercusys MA30H V2 */
```

RinCat, 2024-07-11:
which system are you using? Also the installed kernel is the same as current running?

nicolascolla, 2024-07-11:
I'm running Linux Mint 21.3 Virginia, and the kernel running is the same as the one that is installed, and the one the driver was compiled for: 5.15.0-113-generic

```
$ uname -sr
Linux 5.15.0-113-generic
```

RinCat, 2024-07-12:
What's the result after run `modprobe 88x2bu` in `dmesg`?

nicolascolla, 2024-07-12:
> What's the result after run `modprobe 88x2bu` in `dmesg`?

```
[ 3032.437845] RTW: module init start
[ 3032.437850] RTW: rtl88x2bu v5.13.1-20-gbd7c7eb9d.20210702_COEX20210316-18317b7b
[ 3032.437851] RTW: rtl88x2bu BT-Coex version = COEX20210316-18317b7b
[ 3032.437895] usbcore: registered new interface driver rtl88x2bu
[ 3032.437896] RTW: module init ret=0
```

RinCat, 2024-07-12:
You are still somehow using the old version, I am not sure what happened.
```
[179036.732333] RTW: module init start
[179036.732336] RTW: rtl88x2bu v5.13.1-30-g37e60b26a.20220819_COEX20220812-18317b7b
[179036.732338] RTW: rtl88x2bu BT-Coex version = COEX20220812-18317b7b
[179036.732409] RTW: [HALMAC]55772M
                HALMAC_MAJOR_VER = 1
                HALMAC_PROTOTYPE_VER = 6
                HALMAC_MINOR_VER = 6
                HALMAC_PATCH_VER = 23
```

nicolascolla, 2024-07-12:
Just to be sure, are these steps the correct way of running the new version? Do you see anything wrong being done here?

```
$ cd RTL88x2BU-Linux-Driver/
$ sudo make uninstall
rm -f /lib/modules/5.15.0-113-generic/kernel/drivers/net/wireless//88x2bu.ko
/sbin/depmod -a 5.15.0-113-generic
$ sudo rm /lib/modules/$(uname -r)/kernel/drivers/net/wireless/88x2bu.ko
rm: no se puede borrar '/lib/modules/5.15.0-113-generic/kernel/drivers/net/wireless/88x2bu.ko': No existe el archivo o el directorio ["File not found", the module has already been deleted]
$ cd ..
$ rm -rf RTL88x2BU-Linux-Driver/
$ git clone https://github.com/RinCat/RTL88x2BU-Linux-Driver.git
$ cd RTL88x2BU-Linux-Driver
$ make clean
$ make -j4
$ sudo make install
$ sudo reboot
```

nicolascolla, 2024-07-12:
There's something that appears after the compilation has finished:

```
$ make -j4
...
  CC [M]  /home/nicolas/src/RTL88x2BU-Linux-Driver/hal/btc/halbtc8822b1ant.o
  CC [M]  /home/nicolas/src/RTL88x2BU-Linux-Driver/hal/btc/halbtc8822b2ant.o
  CC [M]  /home/nicolas/src/RTL88x2BU-Linux-Driver/platform/platform_ops.o
  CC [M]  /home/nicolas/src/RTL88x2BU-Linux-Driver/core/rtw_mp.o
  LD [M]  /home/nicolas/src/RTL88x2BU-Linux-Driver/88x2bu.o
  MODPOST /home/nicolas/src/RTL88x2BU-Linux-Driver/Module.symvers
  CC [M]  /home/nicolas/src/RTL88x2BU-Linux-Driver/88x2bu.mod.o
  LD [M]  /home/nicolas/src/RTL88x2BU-Linux-Driver/88x2bu.ko
  BTF [M] /home/nicolas/src/RTL88x2BU-Linux-Driver/88x2bu.ko
Skipping BTF generation for /home/nicolas/src/RTL88x2BU-Linux-Driver/88x2bu.ko due to unavailability of vmlinux
```
Is this normal? Could it be the root of the problem?

RinCat, 2024-07-12:
I don't see any issue here.
But you can try run `find /lib/modules -name "*88x2bu*"` and see if there are multiple modules that should not exist.

RinCat, 2024-07-12:
BTF can be safe ignored, it is used in kernel debug.

nicolascolla, 2024-07-12:
> I don't see any issue here. But you can try run `find /lib/modules -name "*88x2bu*"` and see if there are multiple modules that should not exist.

THAT WAS IT! Thank you very much!

```
$ find /lib/modules -name "*88x2bu*"
/lib/modules/5.15.0-113-generic/updates/dkms/88x2bu.ko
```
After deleting this file, the dongle seems to work, albeit with an error (dmesg output):

```
[ 3944.662212] usb 1-1.2: new high-speed USB device number 7 using ehci-pci
[ 3944.772210] usb 1-1.2: New USB device found, idVendor=2c4e, idProduct=0107, bcdDevice= 2.10
[ 3944.772229] usb 1-1.2: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[ 3944.772235] usb 1-1.2: Product: 802.11ac NIC
[ 3944.772240] usb 1-1.2: Manufacturer: Realtek
[ 3944.772244] usb 1-1.2: SerialNumber: 123456
[ 3944.814472] RTW: module init start
[ 3944.814476] RTW: rtl88x2bu v5.13.1-30-g37e60b26a.20220819_COEX20220812-18317b7b
[ 3944.814478] RTW: rtl88x2bu BT-Coex version = COEX20220812-18317b7b
[ 3944.814559] RTW: [HALMAC]55772M
               HALMAC_MAJOR_VER = 1
               HALMAC_PROTOTYPE_VER = 6
               HALMAC_MINOR_VER = 6
               HALMAC_PATCH_VER = 23
[ 3944.816447] RTW: ERROR [HALMAC][ERR]Dump efuse in suspend
[ 3945.358349] RTW: HW EFUSE
[ 3945.358356] RTW: 0x000: 29 81 00 3C  09 00 A1 00  B6 04 64 10  00 00 A3 00  
[ 3945.358366] RTW: 0x010: 2B 2B 2C 2C  2C 2C 2B 2B  2B 2B 2B 14  00 00 FF FF  
[ 3945.358375] RTW: 0x020: FF FF 29 28  27 27 26 25  24 23 24 25  26 25 25 25  
[ 3945.358384] RTW: 0x030: 13 00 FF FF  00 FF 1C 00  FF FF 2C 2C  2D 2D 2D 2D  
[ 3945.358392] RTW: 0x040: 2C 2C 2C 2C  2C 14 00 00  FF FF FF FF  2B 2B 2B 2B  
[ 3945.358401] RTW: 0x050: 2A 29 28 27  28 28 29 29  29 29 13 00  FF FF 00 FF  
[ 3945.358409] RTW: 0x060: 1C 00 FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF  
[ 3945.358418] RTW: 0x070: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF  
[ 3945.358426] RTW: 0x080: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF  
[ 3945.358435] RTW: 0x090: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF  
[ 3945.358443] RTW: 0x0A0: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF  
[ 3945.358452] RTW: 0x0B0: FF FF FF FF  FF FF FF FF  A6 2F 1F 00  FF FF FF FF  
[ 3945.358460] RTW: 0x0C0: FF 01 00 11  00 00 00 00  00 FF 03 FF  FF FF FF FF  
[ 3945.358469] RTW: 0x0D0: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF  
[ 3945.358477] RTW: 0x0E0: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF  
[ 3945.358486] RTW: 0x0F0: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF  
[ 3945.358494] RTW: 0x100: 4E 2C 07 01  E1 67 02 30  16 9D 2C 71  E2 09 03 52  
[ 3945.358503] RTW: 0x110: 65 61 6C 74  65 6B 0E 03  38 30 32 2E  31 31 61 63  
[ 3945.358511] RTW: 0x120: 20 4E 49 43  08 03 31 32  33 34 35 36  FF FF FF FF  
[ 3945.358520] RTW: 0x130: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF  
[ 3945.358528] RTW: 0x140: 33 0A 1F 01  00 00 21 0F  FF FF FF FF  FF FF FF FF  
[ 3945.358537] RTW: 0x150: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF  
[ 3945.358545] RTW: 0x160: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF  
[ 3945.358554] RTW: 0x170: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF  
[ 3945.358562] RTW: 0x180: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF  
[ 3945.358571] RTW: 0x190: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF  
[ 3945.358579] RTW: 0x1A0: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF  
[ 3945.358588] RTW: 0x1B0: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF  
[ 3945.358597] RTW: 0x1C0: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF  
[ 3945.358605] RTW: 0x1D0: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF  
[ 3945.358614] RTW: 0x1E0: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF  
[ 3945.358622] RTW: 0x1F0: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF  
[ 3945.358630] RTW: 0x200: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF  
[ 3945.358639] RTW: 0x210: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF  
[ 3945.358647] RTW: 0x220: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF  
[ 3945.358656] RTW: 0x230: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF  
[ 3945.358664] RTW: 0x240: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF  
[ 3945.358673] RTW: 0x250: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF  
[ 3945.358681] RTW: 0x260: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF  
[ 3945.358690] RTW: 0x270: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF  
[ 3945.358698] RTW: 0x280: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF  
[ 3945.358707] RTW: 0x290: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF  
[ 3945.358715] RTW: 0x2A0: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF  
[... 12 lines cut ...]
[ 3945.514210] RTW: ERROR [HALMAC][ERR]Dump efuse in suspend
[ 3945.514480] RTW: ERROR [HALMAC][ERR]Dump efuse in suspend
[ 3945.514829] RTW: ERROR [HALMAC][ERR]Dump efuse in suspend
[ 3945.515091] RTW: [RF_PATH] ver_id.RF_TYPE:RF_2T2R
[ 3945.515096] RTW: [RF_PATH] HALSPEC's rf_reg_trx_path_bmp:0x33, rf_reg_path_avail_num:2, max_tx_cnt:2
[ 3945.515101] RTW: [RF_PATH] PG's trx_path_bmp:0x00, max_tx_cnt:0
[ 3945.515105] RTW: [RF_PATH] Registry's trx_path_bmp:0x00, tx_path_lmt:0, rx_path_lmt:0
[ 3945.515108] RTW: [RF_PATH] HALDATA's trx_path_bmp:0x33, max_tx_cnt:2
[ 3945.515111] RTW: [RF_PATH] HALDATA's rf_type:RF_2T2R, NumTotalRFPath:2
[ 3945.515115] RTW: [TRX_Nss] HALSPEC - tx_nss:2, rx_nss:2
[ 3945.515118] RTW: [TRX_Nss] Registry - tx_nss:0, rx_nss:0
[... cut ...]
[ 3945.517641] RTW: rtw_ndev_init(wlan0) if1 mac_addr=30:16:9d:2c:71:e2
[ 3945.520555] usbcore: registered new interface driver rtl88x2bu
[ 3945.520560] RTW: module init ret=0
[ 3945.546635] rtl88x2bu 1-1.2:1.0 wlx30169d2c71e2: renamed from wlan0
[ 3945.607691] RTW: [HALMAC][ALWAYS]shall R reg twice!!
[ 3946.642337] start_addr=(0x20000), end_addr=(0x40000), buffer_size=(0x20000), smp_number_max=(16384)
```

But the green led is blinking and everything works like a charm. Thank you very much for your time again!

closed 2024-07-12 by nicolascolla

RinCat, 2024-07-12:
Those errors are normal, safe ignore.

BTW, you are using dkms, so you better remove them and install this driver dkms, or you will create a conflict each time you upgrade your kernel.