RinCat/RTL88x2BU-Linux-Driver #240: Upstreaming

adanhawth, 2024-08-18:
Any new news on the work done to include this in the Kernel?

RinCat, 2024-08-18:
Hi, the project in the kernel has no relationship with this project. The kernel project is trying to use the same code for all REALTEK WiFi cards, but this one only for 2 models. The REALTEK drivers are different mainlines and branches on different devices, so they cannot be simply merged or upsteamed.

adanhawth, 2024-08-18:
If/when that patchset is accepted, will it deprecate this repo?

What is the status of that patchset?  It looked to fall off the map in 2022.  Am I missing something?

RinCat, 2024-08-18:
It already merged. However, it looks like there are still performance and connectivity issues. So this one will not deprecate yet.

adanhawth, 2024-08-19:
Ok.  Thanks.