RinCat/RTL88x2BU-Linux-Driver #242: Driver causes SATA errors leading to filesystem aborts and crashes

RadiantUwU, 2024-10-05:


RinCat, 2024-10-05:
Hi, can you provide any info?

RadiantUwU, 2024-10-05:
Recently after getting the driver i noticed crashes like these.
[image]

RadiantUwU, 2024-10-05:
This also occurs when resuming from sleep mode.

RinCat, 2024-10-06:
This one look more like your disk is dying, could you check the result of `smartctl -a /dev/sdb`?

RadiantUwU, 2024-10-06:
> This one look more like your disk is dying, could you check the result of `smartctl -a /dev/sdb`?

```
SMART Attributes Data Structure revision number: 16
Vendor Specific SMART Attributes with Thresholds:
ID# ATTRIBUTE_NAME          FLAG     VALUE WORST THRESH TYPE      UPDATED  WHEN_FAILED RAW_VALUE
  1 Raw_Read_Error_Rate     0x0032   100   100   000    Old_age   Always       -       0
  9 Power_On_Hours          0x0032   100   100   000    Old_age   Always       -       2849
 12 Power_Cycle_Count       0x0032   100   100   000    Old_age   Always       -       436
148 Unknown_Attribute       0x0000   100   100   000    Old_age   Offline      -       0
149 Unknown_Attribute       0x0000   100   100   000    Old_age   Offline      -       0
167 Write_Protect_Mode      0x0000   100   100   000    Old_age   Offline      -       0
168 SATA_Phy_Error_Count    0x0012   100   100   000    Old_age   Always       -       0
169 Bad_Block_Rate          0x0000   100   100   000    Old_age   Offline      -       12
170 Bad_Blk_Ct_Lat/Erl      0x0000   100   100   010    Old_age   Offline      -       0/10
172 Erase_Fail_Count        0x0032   100   100   000    Old_age   Always       -       0
173 MaxAvgErase_Ct          0x0000   100   100   000    Old_age   Offline      -       162 (Average 134)
181 Program_Fail_Count      0x0032   100   100   000    Old_age   Always       -       0
182 Erase_Fail_Count        0x0000   100   100   000    Old_age   Offline      -       0
187 Reported_Uncorrect      0x0032   100   100   000    Old_age   Always       -       0
192 Unsafe_Shutdown_Count   0x0012   100   100   000    Old_age   Always       -       46
194 Temperature_Celsius     0x0022   037   047   000    Old_age   Always       -       37 (Min/Max 16/47)
196 Reallocated_Event_Count 0x0032   100   100   000    Old_age   Always       -       0
199 SATA_CRC_Error_Count    0x0032   100   100   000    Old_age   Always       -       720896
218 CRC_Error_Count         0x0032   100   100   000    Old_age   Always       -       0
231 SSD_Life_Left           0x0000   091   091   000    Old_age   Offline      -       91
233 Flash_Writes_GiB        0x0032   100   100   000    Old_age   Always       -       54048
241 Lifetime_Writes_GiB     0x0032   100   100   000    Old_age   Always       -       31114
242 Lifetime_Reads_GiB      0x0032   100   100   000    Old_age   Always       -       40509
244 Average_Erase_Count     0x0000   100   100   000    Old_age   Offline      -       134
245 Max_Erase_Count         0x0000   100   100   000    Old_age   Offline      -       162
246 Total_Erase_Count       0x0000   100   100   000    Old_age   Offline      -       482640
```

RadiantUwU, 2024-10-06:
We should take note how SATA errors are being reported in the logs, but inside the SSD no actual error has ever been reported.

RinCat, 2024-10-06:
The error is reported from SATA module and EXT4 module, why you think this driver cause the error? It gives the I/O timeout error on the logic block number on sdb.

RadiantUwU, 2024-10-06:
> The error is reported from SATA module and EXT4 module, why you think this driver cause the error? It gives the I/O timeout error on the logic block number on sdb.

Never had the issue without this driver, and i just removed and everything is now normal again.

RinCat, 2024-10-06:
> > The error is reported from SATA module and EXT4 module, why you think this driver cause the error? It gives the I/O timeout error on the logic block number on sdb.
> 
> Never had the issue without this driver, and i just removed and everything is now normal again.

Have you considered the possibility that it's EMI?

Nobody reported this before and I had 6 SATA SSDs connected right now.

RadiantUwU, 2024-11-14:
Unsuspected hardware issue (the motherboard is new, bought in August)

The SATA controller died on me [image]