RinCat/RTL88x2BU-Linux-Driver #243: DKMS fails - mainline kernel 6.13, Fedora

onlyreportingissues, 2024-11-24:
DKMS make.log for rtl88x2bu-git for kernel 6.13.0-0.rc0.20241124gt9f16d5e6.407.vanilla.fc41.x86_64 (x86_64)
So 24. Nov 13:58:32 CET 2024
make ARCH=x86_64 CROSS_COMPILE= -C /lib/modules/6.13.0-0.rc0.20241124gt9f16d5e6.407.vanilla.fc41.x86_64/build M=/var/lib/dkms/rtl88x2bu/git/build  modules
make[1]: Verzeichnis „/usr/src/kernels/6.13.0-0.rc0.20241124gt9f16d5e6.407.vanilla.fc41.x86_64“ wird betreten
  CC [M]  /var/lib/dkms/rtl88x2bu/git/build/core/rtw_cmd.o
  CC [M]  /var/lib/dkms/rtl88x2bu/git/build/core/rtw_security.o
  CC [M]  /var/lib/dkms/rtl88x2bu/git/build/core/rtw_debug.o
  CC [M]  /var/lib/dkms/rtl88x2bu/git/build/core/rtw_io.o
  CC [M]  /var/lib/dkms/rtl88x2bu/git/build/core/rtw_ioctl_query.o
  CC [M]  /var/lib/dkms/rtl88x2bu/git/build/core/rtw_ioctl_set.o
  CC [M]  /var/lib/dkms/rtl88x2bu/git/build/core/rtw_ieee80211.o
  CC [M]  /var/lib/dkms/rtl88x2bu/git/build/core/rtw_mlme.o
  CC [M]  /var/lib/dkms/rtl88x2bu/git/build/core/rtw_mlme_ext.o
  CC [M]  /var/lib/dkms/rtl88x2bu/git/build/core/rtw_mi.o
  CC [M]  /var/lib/dkms/rtl88x2bu/git/build/core/rtw_wlan_util.o
  CC [M]  /var/lib/dkms/rtl88x2bu/git/build/core/rtw_vht.o
In Datei, eingebunden von /var/lib/dkms/rtl88x2bu/git/build/include/drv_types.h:114,
                 von /var/lib/dkms/rtl88x2bu/git/build/core/rtw_cmd.c:17:
/var/lib/dkms/rtl88x2bu/git/build/include/rtw_mlme_ext.h:1181:4: Warnung: kein vorheriger Prototyp für »rtw_mesh_set_plink_state_cmd_hdl« [-Wmissing-prototypes]
 1181 | u8 rtw_mesh_set_plink_state_cmd_hdl(_adapter *adapter, u8 *parmbuf) { return H2C_CMD_FAIL; };
      |    ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
/var/lib/dkms/rtl88x2bu/git/build/core/rtw_ieee80211.c:489:6: Warnung: kein vorheriger Prototyp für »rtw_filter_suppport_rateie« [-Wmissing-prototypes]
  489 | void rtw_filter_suppport_rateie(WLAN_BSSID_EX *pbss_network, u8 keep)
      |      ^~~~~~~~~~~~~~~~~~~~~~~~~~
/var/lib/dkms/rtl88x2bu/git/build/core/rtw_ieee80211.c:766:5: Warnung: kein vorheriger Prototyp für »rtw_get_akm_suite_bitmap« [-Wmissing-prototypes]
  766 | u32 rtw_get_akm_suite_bitmap(u8 *s)
      |     ^~~~~~~~~~~~~~~~~~~~~~~~
/var/lib/dkms/rtl88x2bu/git/build/core/rtw_ieee80211.c:1830:6: Warnung: kein vorheriger Prototyp für »dump_ht_cap_ie« [-Wmissing-prototypes]
 1830 | void dump_ht_cap_ie(void *sel, const u8 *ie, u32 ie_len)
      |      ^~~~~~~~~~~~~~
/var/lib/dkms/rtl88x2bu/git/build/core/rtw_ieee80211.c:1849:6: Warnung: kein vorheriger Prototyp für »dump_ht_op_ie_content« [-Wmissing-prototypes]
 1849 | void dump_ht_op_ie_content(void *sel, const u8 *buf, u32 buf_len)
      |      ^~~~~~~~~~~~~~~~~~~~~
/var/lib/dkms/rtl88x2bu/git/build/core/rtw_ieee80211.c:1863:6: Warnung: kein vorheriger Prototyp für »dump_ht_op_ie« [-Wmissing-prototypes]
 1863 | void dump_ht_op_ie(void *sel, const u8 *ie, u32 ie_len)
      |      ^~~~~~~~~~~~~
  CC [M]  /var/lib/dkms/rtl88x2bu/git/build/core/rtw_pwrctrl.o
/var/lib/dkms/rtl88x2bu/git/build/core/rtw_vht.c:43:6: Warnung: kein vorheriger Prototyp für »dump_vht_cap_ie_content« [-Wmissing-prototypes]
   43 | void dump_vht_cap_ie_content(void *sel, const u8 *buf, u32 buf_len)
      |      ^~~~~~~~~~~~~~~~~~~~~~~
/var/lib/dkms/rtl88x2bu/git/build/core/rtw_vht.c:83:6: Warnung: kein vorheriger Prototyp für »dump_vht_op_ie_content« [-Wmissing-prototypes]
   83 | void dump_vht_op_ie_content(void *sel, const u8 *buf, u32 buf_len)
      |      ^~~~~~~~~~~~~~~~~~~~~~
/var/lib/dkms/rtl88x2bu/git/build/core/rtw_mlme.c:26:6: Warnung: kein vorheriger Prototyp für »rtw_init_mlme_timer« [-Wmissing-prototypes]
   26 | void rtw_init_mlme_timer(_adapter *padapter)
      |      ^~~~~~~~~~~~~~~~~~~
/var/lib/dkms/rtl88x2bu/git/build/core/rtw_mlme.c:43:9: Warnung: kein vorheriger Prototyp für »_rtw_init_mlme_priv« [-Wmissing-prototypes]
   43 | sint    _rtw_init_mlme_priv(_adapter *padapter)
      |         ^~~~~~~~~~~~~~~~~~~
/var/lib/dkms/rtl88x2bu/git/build/core/rtw_vht.c:370:6: Warnung: kein vorheriger Prototyp für »update_sta_vht_info_apmode_bf_cap« [-Wmissing-prototypes]
  370 | void update_sta_vht_info_apmode_bf_cap(_adapter *padapter, struct sta_info *psta)
      |      ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
/var/lib/dkms/rtl88x2bu/git/build/core/rtw_mlme.c:329:6: Warnung: kein vorheriger Prototyp für »_rtw_free_mlme_priv« [-Wmissing-prototypes]
  329 | void _rtw_free_mlme_priv(struct mlme_priv *pmlmepriv)
      |      ^~~~~~~~~~~~~~~~~~~
/var/lib/dkms/rtl88x2bu/git/build/core/rtw_mlme.c:357:9: Warnung: kein vorheriger Prototyp für »_rtw_enqueue_network« [-Wmissing-prototypes]
  357 | sint    _rtw_enqueue_network(_queue *queue, struct wlan_network *pnetwork)
      |         ^~~~~~~~~~~~~~~~~~~~
/var/lib/dkms/rtl88x2bu/git/build/core/rtw_mlme_ext.c:627:5: Warnung: kein vorheriger Prototyp für »rtw_chset_get_ch_non_ocp_ms« [-Wmissing-prototypes]
  627 | u32 rtw_chset_get_ch_non_ocp_ms(RT_CHANNEL_INFO *ch_set, u8 ch, u8 bw, u8 offset)
      |     ^~~~~~~~~~~~~~~~~~~~~~~~~~~
/var/lib/dkms/rtl88x2bu/git/build/core/rtw_mi.c:877:4: Warnung: kein vorheriger Prototyp für »_rtw_mi_busy_traffic_check« [-Wmissing-prototypes]
  877 | u8 _rtw_mi_busy_traffic_check(_adapter *padapter, void *data)
      |    ^~~~~~~~~~~~~~~~~~~~~~~~~~
/var/lib/dkms/rtl88x2bu/git/build/core/rtw_cmd.c:1634:6: Warnung: kein vorheriger Prototyp für »free_assoc_resources_hdl« [-Wmissing-prototypes]
 1634 | void free_assoc_resources_hdl(_adapter *padapter, u8 lock_scanned_queue)
      |      ^~~~~~~~~~~~~~~~~~~~~~~~
/var/lib/dkms/rtl88x2bu/git/build/core/rtw_mlme.c:2417:6: Warnung: kein vorheriger Prototyp für »rtw_reset_rx_info« [-Wmissing-prototypes]
 2417 | void rtw_reset_rx_info(_adapter *adapter)
      |      ^~~~~~~~~~~~~~~~~
[... 459 lines cut ...]
      |     ^~~~~~~~~~~~~~~~
  CC [M]  /var/lib/dkms/rtl88x2bu/git/build/core/rtw_odm.o
/var/lib/dkms/rtl88x2bu/git/build/core/rtw_xmit.c:6634:6: Warnung: kein vorheriger Prototyp für »rtw_sctx_chk_waring_status« [-Wmissing-prototypes]
 6634 | bool rtw_sctx_chk_waring_status(int status)
      |      ^~~~~~~~~~~~~~~~~~~~~~~~~~
  CC [M]  /var/lib/dkms/rtl88x2bu/git/build/core/rtw_rm.o
/var/lib/dkms/rtl88x2bu/git/build/core/rtw_br_ext.c:588:6: Warnung: kein vorheriger Prototyp für »nat25_db_expire« [-Wmissing-prototypes]
  588 | void nat25_db_expire(_adapter *priv)
      |      ^~~~~~~~~~~~~~~
/var/lib/dkms/rtl88x2bu/git/build/core/rtw_br_ext.c:704:5: Warnung: kein vorheriger Prototyp für »nat25_db_handle« [-Wmissing-prototypes]
  704 | int nat25_db_handle(_adapter *priv, struct sk_buff *skb, int method)
      |     ^~~~~~~~~~~~~~~
/var/lib/dkms/rtl88x2bu/git/build/core/rtw_br_ext.c:1114:5: Warnung: kein vorheriger Prototyp für »nat25_handle_frame« [-Wmissing-prototypes]
 1114 | int nat25_handle_frame(_adapter *priv, struct sk_buff *skb)
      |     ^~~~~~~~~~~~~~~~~~
/var/lib/dkms/rtl88x2bu/git/build/core/rtw_br_ext.c:1245:6: Warnung: kein vorheriger Prototyp für »dhcp_flag_bcast« [-Wmissing-prototypes]
 1245 | void dhcp_flag_bcast(_adapter *priv, struct sk_buff *skb)
      |      ^~~~~~~~~~~~~~~
/var/lib/dkms/rtl88x2bu/git/build/core/rtw_br_ext.c:1286:7: Warnung: kein vorheriger Prototyp für »scdb_findEntry« [-Wmissing-prototypes]
 1286 | void *scdb_findEntry(_adapter *priv, unsigned char *macAddr,
      |       ^~~~~~~~~~~~~~
  CC [M]  /var/lib/dkms/rtl88x2bu/git/build/core/rtw_rm_fsm.o
/var/lib/dkms/rtl88x2bu/git/build/core/rtw_sreset.c:104:6: Warnung: kein vorheriger Prototyp für »sreset_restore_security_station« [-Wmissing-prototypes]
  104 | void sreset_restore_security_station(_adapter *padapter)
      |      ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~

onlyreportingissues, 2024-11-24:
/var/lib/dkms/rtl88x2bu/git/build/core/rtw_sreset.c:140:6: Warnung: kein vorheriger Prototyp für »sreset_restore_network_station« [-Wmissing-prototypes]
  140 | void sreset_restore_network_station(_adapter *padapter)
      |      ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
/var/lib/dkms/rtl88x2bu/git/build/core/rtw_sreset.c:200:6: Warnung: kein vorheriger Prototyp für »sreset_restore_network_status« [-Wmissing-prototypes]
  200 | void sreset_restore_network_status(_adapter *padapter)
      |      ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~
  CC [M]  /var/lib/dkms/rtl88x2bu/git/build/core/rtw_ft.o
  CC [M]  /var/lib/dkms/rtl88x2bu/git/build/core/rtw_wnm.o
/var/lib/dkms/rtl88x2bu/git/build/core/rtw_btcoex.c:409:6: Warnung: kein vorheriger Prototyp für »rtw_btcoex_SetBtPatchVersion« [-Wmissing-prototypes]
  409 | void rtw_btcoex_SetBtPatchVersion(PADAPTER padapter, u16 btHciVer, u16 btPatchVer)
      |      ^~~~~~~~~~~~~~~~~~~~~~~~~~~~
/var/lib/dkms/rtl88x2bu/git/build/core/rtw_btcoex.c:414:6: Warnung: kein vorheriger Prototyp für »rtw_btcoex_SetHciVersion« [-Wmissing-prototypes]
  414 | void rtw_btcoex_SetHciVersion(PADAPTER  padapter, u16 hciVersion)
      |      ^~~~~~~~~~~~~~~~~~~~~~~~
/var/lib/dkms/rtl88x2bu/git/build/core/rtw_btcoex.c:419:6: Warnung: kein vorheriger Prototyp für »rtw_btcoex_StackUpdateProfileInfo« [-Wmissing-prototypes]
  419 | void rtw_btcoex_StackUpdateProfileInfo(void)
      |      ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
  CC [M]  /var/lib/dkms/rtl88x2bu/git/build/core/rtw_mbo.o
  CC [M]  /var/lib/dkms/rtl88x2bu/git/build/core/rtw_rm_util.o
  CC [M]  /var/lib/dkms/rtl88x2bu/git/build/core/efuse/rtw_efuse.o
  CC [M]  /var/lib/dkms/rtl88x2bu/git/build/core/rtw_roch.o
/var/lib/dkms/rtl88x2bu/git/build/core/rtw_odm.c:67:6: Warnung: kein vorheriger Prototyp für »rtw_odm_adaptivity_ver_msg« [-Wmissing-prototypes]
   67 | void rtw_odm_adaptivity_ver_msg(void *sel, _adapter *adapter)
      |      ^~~~~~~~~~~~~~~~~~~~~~~~~~
/var/lib/dkms/rtl88x2bu/git/build/core/rtw_odm.c:75:6: Warnung: kein vorheriger Prototyp für »rtw_odm_adaptivity_en_msg« [-Wmissing-prototypes]
   75 | void rtw_odm_adaptivity_en_msg(void *sel, _adapter *adapter)
      |      ^~~~~~~~~~~~~~~~~~~~~~~~~
/var/lib/dkms/rtl88x2bu/git/build/core/rtw_odm.c:92:6: Warnung: kein vorheriger Prototyp für »rtw_odm_adaptivity_mode_msg« [-Wmissing-prototypes]
   92 | void rtw_odm_adaptivity_mode_msg(void *sel, _adapter *adapter)
      |      ^~~~~~~~~~~~~~~~~~~~~~~~~~~
  CC [M]  /var/lib/dkms/rtl88x2bu/git/build/core/crypto/aes-internal.o
  CC [M]  /var/lib/dkms/rtl88x2bu/git/build/core/crypto/aes-internal-enc.o
  CC [M]  /var/lib/dkms/rtl88x2bu/git/build/core/crypto/aes-gcm.o
  CC [M]  /var/lib/dkms/rtl88x2bu/git/build/core/crypto/aes-ccm.o
  CC [M]  /var/lib/dkms/rtl88x2bu/git/build/core/crypto/aes-omac1.o
  CC [M]  /var/lib/dkms/rtl88x2bu/git/build/core/crypto/ccmp.o
  CC [M]  /var/lib/dkms/rtl88x2bu/git/build/core/crypto/gcmp.o
  CC [M]  /var/lib/dkms/rtl88x2bu/git/build/core/crypto/aes-siv.o
  CC [M]  /var/lib/dkms/rtl88x2bu/git/build/core/crypto/aes-ctr.o
  CC [M]  /var/lib/dkms/rtl88x2bu/git/build/core/crypto/sha256-internal.o
  CC [M]  /var/lib/dkms/rtl88x2bu/git/build/core/crypto/sha256.o
/var/lib/dkms/rtl88x2bu/git/build/core/efuse/rtw_efuse.c:60:9: Warnung: kein vorheriger Prototyp für »efuse_IsBT_Masked« [-Wmissing-prototypes]
   60 | BOOLEAN efuse_IsBT_Masked(PADAPTER pAdapter, u16 Offset)
      |         ^~~~~~~~~~~~~~~~~
/var/lib/dkms/rtl88x2bu/git/build/core/efuse/rtw_efuse.c: In Funktion »rtw_efuse_analyze«:
/var/lib/dkms/rtl88x2bu/git/build/core/efuse/rtw_efuse.c:971:17: Warnung: diese »if«-Klausel beschützt nicht … [-Wmisleading-indentation]
  971 |                 if (i % 16 == 0)
      |                 ^~
In Datei, eingebunden von /var/lib/dkms/rtl88x2bu/git/build/include/drv_types.h:60,
                 von /var/lib/dkms/rtl88x2bu/git/build/core/efuse/rtw_efuse.c:17:
/var/lib/dkms/rtl88x2bu/git/build/include/rtw_debug.h:261:9: Anmerkung: … diese Anweisung, aber letztere ist verwirrend eingerückt, als wäre sie abhängig von »if«
  261 |         do {\
      |         ^~
/var/lib/dkms/rtl88x2bu/git/build/core/efuse/rtw_efuse.c:973:25: Anmerkung: bei Substitution des Makros »_RTW_PRINT_SEL«
  973 |                         _RTW_PRINT_SEL(RTW_DBGDUMP, "%02X%s"
      |                         ^~~~~~~~~~~~~~
/var/lib/dkms/rtl88x2bu/git/build/core/efuse/rtw_efuse.c: Auf höchster Ebene:
/var/lib/dkms/rtl88x2bu/git/build/core/efuse/rtw_efuse.c:985:6: Warnung: kein vorheriger Prototyp für »efuse_PreUpdateAction« [-Wmissing-prototypes]
  985 | void efuse_PreUpdateAction(
      |      ^~~~~~~~~~~~~~~~~~~~~
/var/lib/dkms/rtl88x2bu/git/build/core/efuse/rtw_efuse.c:1014:6: Warnung: kein vorheriger Prototyp für »efuse_PostUpdateAction« [-Wmissing-prototypes]
 1014 | void efuse_PostUpdateAction(
      |      ^~~~~~~~~~~~~~~~~~~~~~
  CC [M]  /var/lib/dkms/rtl88x2bu/git/build/core/crypto/sha256-prf.o
  CC [M]  /var/lib/dkms/rtl88x2bu/git/build/core/crypto/rtw_crypto_wrap.o
  CC [M]  /var/lib/dkms/rtl88x2bu/git/build/core/rtw_swcrypto.o
  CC [M]  /var/lib/dkms/rtl88x2bu/git/build/os_dep/osdep_service.o
  CC [M]  /var/lib/dkms/rtl88x2bu/git/build/os_dep/linux/os_intfs.o
/var/lib/dkms/rtl88x2bu/git/build/core/crypto/ccmp.c:179:6: Warnung: kein vorheriger Prototyp für »ccmp_get_pn« [-Wmissing-prototypes]
  179 | void ccmp_get_pn(u8 *pn, const u8 *data)
[... 393 lines cut ...]
[... cut ...]
/var/lib/dkms/rtl88x2bu/git/build/hal/hal_com_phycfg.c:5880:4: Warnung: kein vorheriger Prototyp für »phy_get_txpwr_tpc« [-Wmissing-prototypes]
 5880 | s8 phy_get_txpwr_tpc(_adapter *adapter, struct hal_spec_t *hal_spec)
      |    ^~~~~~~~~~~~~~~~~
make[2]: *** [/usr/src/kernels/6.13.0-0.rc0.20241124gt9f16d5e6.407.vanilla.fc41.x86_64/Makefile:1938: /var/lib/dkms/rtl88x2bu/git/build] Fehler 2
make[1]: *** [Makefile:226: __sub-make] Fehler 2
make[1]: Verzeichnis „/usr/src/kernels/6.13.0-0.rc0.20241124gt9f16d5e6.407.vanilla.fc41.x86_64“ wird verlassen
make: *** [Makefile:2470: modules] Fehler 2

onlyreportingissues, 2024-11-26:
This fixes it: [d8d9ef02c73833a9d83f82f45aa77cfed86d6b8c](https://github.com/cilynx/rtl88x2bu/compare/5.8.7.1_35809.20191129_COEX20191120-7777...fix-linux-6.12)

RinCat, 2024-11-26:
Thank you, but my upstream does not have 6.13 yet. I will update that after I did tests.

RinCat, 2025-01-13:
https://github.com/RinCat/RTL88x2BU-Linux-Driver/commit/2092b8d80d7fb7fa953d9dd3d666e5679c99a2f8