RinCat/RTL88x2BU-Linux-Driver #250: out of boundary of array

0TripleFish0, 2025-02-18:
hal/phydm)/phydm_dfs.c, line 1286

dfs->pw_long_hold_sum[i] = dfs->pw_long_hold_sum[i] +
			dfs->pw_hold[(dfs->hist_long_idx + j) % 4][i];

size of dfs->pw_long_hold_sum is 6, but i=6 herer

RinCat, 2025-02-18:
Hmmm, it seems Realtek does not touch that in the past 5+ years.

That j loop seems should be nested?

```
	for (i = 0; i < 6; i++) {
		dfs->pw_hold[dfs->hist_idx][i] = (u8)g_pw[i];
		dfs->pri_hold[dfs->hist_idx][i] = (u8)g_pri[i];
		/*@collect whole histogram report may take some time
		 *so we add the counter of 2 time slots in FCC and ETSI
		 */
		dfs->pw_hold_sum[i] = dfs->pw_hold_sum[i] +
			dfs->pw_hold[(dfs->hist_idx + 1) % 3][i] +
			dfs->pw_hold[(dfs->hist_idx + 2) % 3][i];
		dfs->pri_hold_sum[i] = dfs->pri_hold_sum[i] +
			dfs->pri_hold[(dfs->hist_idx + 1) % 3][i] +
			dfs->pri_hold[(dfs->hist_idx + 2) % 3][i];

		/*@For long radar type*/
		for (j = 1; j < 4; j++) {
			dfs->pw_long_hold_sum[i] = dfs->pw_long_hold_sum[i] +
				dfs->pw_hold[(dfs->hist_long_idx + j) % 4][i];
			dfs->pri_long_hold_sum[i] = dfs->pri_long_hold_sum[i] +
				dfs->pri_hold[(dfs->hist_long_idx + j) % 4][i];
		}
	}

```

RinCat, 2025-02-18:
https://github.com/RinCat/RTL88x2BU-Linux-Driver/commit/f0644acaa8b906a94a5176d7afc9efc988239fcb

closed 2025-02-18 by RinCat