RinCat/RTL88x2BU-Linux-Driver #252: Hello, Rnschat,  The driver has been a great help in my work, but I have a question about MP_tool.

innerlife, 2025-03-06:
Hello,

I successfully managed to operate the USB WiFi dongle using your driver, and I sincerely appreciate your work. 
However, I am facing an issue where the 'mp_tool' function is not working. 
I would like to know whether this function is supported and how I can enable it.

This feature is essential for conducting RF tests required for product certification, so your assistance would be greatly appreciated. I look forward to your response.

Best regards,
innerlife

RinCat, 2025-03-07:
Hi, this driver does not include the `mp_tool`, so it never been tested. If you can post error log I may have a look.

innerlife, 2025-03-09:
Thank you for your response.

Since the CONFIG_MP_INCLUDED option is set to 'y' in the Makefile, I assumed that the MP tool would be included. However, after probing, the related commands (e.g., mp_start, mp_channel, etc.) are not listed, which is why I am reaching out for clarification.

Could you kindly guide me on which part of the source code I should update to test this myself? Your assistance would be greatly appreciated.

root@linaro-alip:~# iwpriv wlan0
wlan0     Available private ioctls :
          write            (8BE0) : set 2047 char  & get   0
          read             (8BE1) : set 2047 char  & get  16 char
          driver_ext       (8BE2) : set   0       & get   0
          mp_ioctl         (8BE3) : set   0       & get   0
          apinfo           (8BE4) : set   1 int   & get   0
          setpid           (8BE5) : set   2 int   & get   0
          wps_start        (8BE6) : set   1 int   & get   0
          get_sensitivity  (8BE7) : set   1 int   & get   0
          wps_prob_req_ie  (8BE8) : set   1 int   & get   0
          wps_assoc_req_ie (8BE9) : set   1 int   & get   0
          channel_plan     (8BEA) : set   1 int   & get   0
          dbg              (8BEB) : set   2 int   & get   0
          rfw              (8BEC) : set   3 int   & get   0
          rfr              (8BED) : set   2 int   & get  16 char
          p2p_set          (8BF0) : set 1024 char  & get   0
          p2p_get          (8BF1) : set 1024 char  & get 2047 char
          NULL             (8BF2) : set   0       & get   0
          p2p_get2         (8BF3) : set  64 char  & get  64 char
          tdls             (8BF4) : set  64 char  & get   0
          tdls_get         (8BF5) : set 1024 char  & get 1024 char
          pm_set           (8BF6) : set  64 char  & get   0
          NULL             (8BF7) : set 1024 char  & get   0
          NULL             (8BF8) : set   0       & get   0
          NULL             (8BFA) : set 1024 char  & get   0
          NULL             (8BFB) : set 128 char  & get 2047 char
          test             (8BFD) : set  40 char  & get 2047 char

RinCat, 2025-03-11:
Hi, the MP tools does not included, the `CONFIG_MP_INCLUDED` only enabled some functions for the tool to call during test. I've never seen Realtek provide this tool itself in these drivers, so I think it's probably proprietary. And after some search, I believe the tool called `rtwpriv`, but I never used it. You need to test it yourself.