RinCat/RTL88x2BU-Linux-Driver #253: Nintendo Switch support?

Talonflame193, 2025-03-07:
I just installed Ubuntu noble on my switch and I tried hours installing drivers for my Ac53 dongle but it doesn't work. Is it even possible?

RinCat, 2025-03-07:
Hi, Nintendo Switch uses ARM so it in theory it could work. And logs?

Talonflame193, 2025-03-07:
I am a newbie so I think its definitely me doing smth wrong. I can send you logs but can you tell me which exactly? After some research I saw that the switch uses kernel 4.9.x but I dont know if it works and where I can get kernel headers

RinCat, 2025-03-07:
You need attach logs first. Where did you got the kernel?

RinCat, 2025-03-07:
run `sudo apt install switch-bsp` and see if it works

RinCat, 2025-03-07:
OK, then post the build logs.

Talonflame193, 2025-03-07:
Please tell me how to get them

RinCat, 2025-03-07:
run `make . 2>&1 | tee output.txt` in the driver dir and upload the `output.txt`

Talonflame193, 2025-03-07:
It shows this error on every file I try to make

[output.txt](https://github.com/user-attachments/files/19133832/output.txt)

RinCat, 2025-03-07:
`/bin/sh: 1: scripts/basic/fixdep: Exec format error` this is caused by the kernel build system issue. There is nothing I can do unless you want build the kernel yourself. You need report this to their discord for their to fix it.

Talonflame193, 2025-03-08:
How can I build the kernel myself?

RinCat, 2025-03-08:
If you ask then it means that it is outside your current skill set. The kernel version indicates that it is an unusual kernel with custom patches. And the `exec format error' means someone built it using cross-compile, which is more complex than a normal kernel build. You need to use the exact same build environment to restore this. Or you can ask its author if he can build one for you.

Talonflame193, 2025-03-08:
I’ll ask him but anyway ty for your help, I appreciate it!

makinbacon21, 2025-03-16:
switch guy here!

the issue is this entire build system is based on the assumption that it is being built on the original kernel build host, or at least that it is being built on the same arch as the original build host. fixdep is running and failing because fixdep is a host tool for linux build and therefore is whatever the host arch is. these assumptions are incorrect for a module that can be built on target with dkms etc. the on-target build directory is also /lib/modules for some reason, which is kinda unhinged lol.

> The kernel version indicates that it is an unusual kernel with custom patches

irrelevant here, not the issue and shouldn't affect the build

> And the exec format error means someone built it using cross-compile, which is more complex than a normal kernel build. You need to use the exact same build environment to restore this.

imo, if you support dkms etc. you should have the makefile actually account for cross compiled kernels (every other module I've seen does this). most arm kernels are not in fact compiled on arm machines, at least for now, unless you have some ampere machine lying around 😛.

for now @Talonflame193 your best option is as you have been told in l4s server, build the kernel yourself (pretty easy using kernel build scripts, https://gitlab.com/l4t-community/kernel/l4t-kernel-build-scripts)

@RinCat highly recommend someone redo that makefile, or dkms/on target compile support should not be advertised or be accompanied by a strong warning. all a user should need to compile a dkms kmod is headers and basic build tools, and no assumptions should be made regarding original kernel build host arch.

RinCat, 2025-03-16:
@makinbacon21 welcome here. 🤣

That Makefile from upstream, i doubt it if someone want to spend time to rewrite it for a EOL product.

But I think it can be used for cross compile if set correct override variables.