RinCat/RTL88x2BU-Linux-Driver #254: USB 3.0 Mode does not seem to work correctly in new kernel version 6.13.6 (6.13.6-arch1-1)

RAIACAD0564, 2025-03-12:
After rebooting following a kernel update seeming to be the only one relevant to this issue, it seems the adapter can no longer connect correctly if the driver is set to USB 3.0 mode (rtw_switch_usb_mode=1).

This setup has already been working for months without issue, as per the readme, the motherboard, USB port, and WiFi adapter all fully support USB 3.0, and it had been working fine up until the updated kernel.

The adapter works fine if I run it in USB 2.0 mode, both without defining any options and if I manually add the option (rtw_switch_usb_mode=2). This isn't a big loss for me as my internet plan doesn't exceed USB 2.0 speeds anyways, but I thought I'd report it in case fixes need to be made or if it persists into future kernel versions.

The following are the logs from journalctl. networkmanager logs are unnecessary as the issue doesn't allow the adapter to make it to the step where networkmanager can recognize it:

**Modprobing in USB 2.0/3.0 mode (Both do not show any differences in loading, only when connecting the adapter, however, I thought to include it in case it's relevant.):**

```
Mar 11 23:35:17 Rukia kernel: RTW: rtl88x2bu v5.13.1-30-g37e60b26a.20220819_COEX20220812-18317b7b
Mar 11 23:35:17 Rukia kernel: RTW: rtl88x2bu BT-Coex version = COEX20220812-18317b7b
Mar 11 23:35:17 Rukia kernel: usbcore: registered new interface driver rtl88x2bu
Mar 11 23:35:17 Rukia kernel: RTW: module init ret=0
Mar 11 23:35:17 Rukia sudo[44750]: pam_unix(sudo:session): session closed for user root
```

**Connecting adapter to port in USB 3.0 mode (failure):**

```
Mar 11 23:28:57 Rukia kernel: usb 3-2: new high-speed USB device number 22 using xhci_hcd
Mar 11 23:28:57 Rukia kernel: usb 3-2: New USB device found, idVendor=2357, idProduct=0115, bcdDevice= 2.10
Mar 11 23:28:58 Rukia kernel: usb 3-2: New USB device strings: Mfr=1, Product=2, SerialNumber=3
Mar 11 23:28:58 Rukia kernel: usb 3-2: Product: 802.11ac NIC
Mar 11 23:28:58 Rukia kernel: usb 3-2: Manufacturer: Realtek
Mar 11 23:28:58 Rukia kernel: usb 3-2: SerialNumber: 123456
Mar 11 23:28:58 Rukia kernel: RTW: [HALMAC]55772M
                              HALMAC_MAJOR_VER = 1
                              HALMAC_PROTOTYPE_VER = 6
                              HALMAC_MINOR_VER = 6
                              HALMAC_PATCH_VER = 23
Mar 11 23:28:58 Rukia kernel: RTW: ERROR [HALMAC][ERR]Dump efuse in suspend
Mar 11 23:28:58 Rukia kernel: RTW: HW EFUSE
Mar 11 23:28:58 Rukia kernel: RTW: 0x000: 29 81 00 3C  09 00 A1 00  B6 04 64 10  00 00 A3 00
Mar 11 23:28:58 Rukia kernel: RTW: 0x010: 2F 2F 30 30  30 30 29 29  2A 2A 2A 13  EE EE FF FF
Mar 11 23:28:58 Rukia kernel: RTW: 0x020: FF FF 15 15  14 14 14 13  12 11 10 0E  14 12 12 12
Mar 11 23:28:58 Rukia kernel: RTW: 0x030: 13 EE FF FF  00 FF 0C E0  FF FF 30 30  31 31 31 31
Mar 11 23:28:58 Rukia kernel: RTW: 0x040: 2C 2C 2C 2C  2C 13 EE 00  FF FF FF FF  1C 1B 1A 1A
Mar 11 23:28:58 Rukia kernel: RTW: 0x050: 17 15 14 12  11 0F 15 12  12 12 13 EE  FF FF 00 FF
Mar 11 23:28:58 Rukia kernel: RTW: 0x060: 0C E0 FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF
Mar 11 23:28:58 Rukia kernel: RTW: 0x070: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF
Mar 11 23:28:58 Rukia kernel: RTW: 0x080: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF
Mar 11 23:28:58 Rukia kernel: RTW: 0x090: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF
Mar 11 23:28:58 Rukia kernel: RTW: 0x0A0: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF
Mar 11 23:28:58 Rukia kernel: RTW: 0x0B0: FF FF FF FF  FF FF FF FF  B4 27 1B 00  FF FF FF FF
Mar 11 23:28:58 Rukia kernel: RTW: 0x0C0: FF 09 00 11  00 00 00 00  00 FF 02 FF  FF FF FF FF
Mar 11 23:28:58 Rukia kernel: RTW: 0x0D0: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF
Mar 11 23:28:58 Rukia kernel: RTW: 0x0E0: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF
Mar 11 23:28:58 Rukia kernel: RTW: 0x0F0: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF
Mar 11 23:28:58 Rukia kernel: RTW: 0x100: 57 23 15 01  C1 67 02 50  3E AA 28 B3  2A 09 03 52
Mar 11 23:28:58 Rukia kernel: RTW: 0x110: 65 61 6C 74  65 6B 0E 03  38 30 32 2E  31 31 61 63
Mar 11 23:28:58 Rukia kernel: RTW: 0x120: 20 4E 49 43  08 03 31 32  33 34 35 36  FF FF FF FF
Mar 11 23:28:58 Rukia kernel: RTW: 0x130: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF
Mar 11 23:28:58 Rukia kernel: RTW: 0x140: 33 0A 1F 01  00 00 21 0F  FF FF FF FF  FF FF FF FF
Mar 11 23:28:58 Rukia kernel: RTW: 0x150: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF
Mar 11 23:28:58 Rukia kernel: RTW: 0x160: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF
Mar 11 23:28:58 Rukia kernel: RTW: 0x170: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF
Mar 11 23:28:58 Rukia kernel: RTW: 0x180: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF
Mar 11 23:28:58 Rukia kernel: RTW: 0x190: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF
Mar 11 23:28:58 Rukia kernel: RTW: 0x1A0: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF
Mar 11 23:28:58 Rukia kernel: RTW: 0x1B0: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF
Mar 11 23:28:58 Rukia kernel: RTW: 0x1C0: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF
Mar 11 23:28:58 Rukia kernel: RTW: 0x1D0: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF
Mar 11 23:28:58 Rukia kernel: RTW: 0x1E0: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF
Mar 11 23:28:58 Rukia kernel: RTW: 0x1F0: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF
Mar 11 23:28:58 Rukia kernel: RTW: 0x200: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF
Mar 11 23:28:58 Rukia kernel: RTW: 0x210: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF
Mar 11 23:28:58 Rukia kernel: RTW: 0x220: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF
Mar 11 23:28:58 Rukia kernel: RTW: 0x230: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF
[... 140 lines cut ...]
Mar 11 23:37:05 Rukia kernel: start_addr=(0x20000), end_addr=(0x40000), buffer_size=(0x20000), smp_number_max=(16384)
Mar 11 23:37:05 Rukia kernel: RTW: WARN _beamforming_leave: 00:00:00:00:00:00 is neither beamforming ee or er!!
Mar 11 23:37:05 Rukia kernel: RTW: ERROR Free disconnecting network of scanned_queue failed due to pwlan == NULL
Mar 11 23:37:05 Rukia NetworkManager[995]: <info>  [1741750625.5209] device (wlan1): set-hw-addr: set MAC address to 2A:0D:C4:B0:B3:B5 (scanning)
Mar 11 23:37:05 Rukia NetworkManager[995]: <info>  [1741750625.5513] device (wlan1): supplicant interface state: internal-starting -> disconnected
Mar 11 23:37:05 Rukia NetworkManager[995]: <info>  [1741750625.5513] Wi-Fi P2P device controlled by interface wlan1 created
Mar 11 23:37:05 Rukia NetworkManager[995]: <info>  [1741750625.5514] manager: (p2p-dev-wlan1): new 802.11 Wi-Fi P2P device (/org/freedesktop/NetworkManager/Devices/12)
Mar 11 23:37:05 Rukia NetworkManager[995]: <info>  [1741750625.5515] device (p2p-dev-wlan1): state change: unmanaged -> unavailable (reason 'managed', managed-type: 'external')
Mar 11 23:37:05 Rukia NetworkManager[995]: <info>  [1741750625.5518] device (wlan1): state change: unavailable -> disconnected (reason 'supplicant-available', managed-type: 'full')
Mar 11 23:37:05 Rukia NetworkManager[995]: <info>  [1741750625.5521] device (p2p-dev-wlan1): state change: unavailable -> disconnected (reason 'none', managed-type: 'full')
Mar 11 23:37:08 Rukia systemd[1]: systemd-rfkill.service: Deactivated successfully.
```

It seems to me the trouble line in USB 3.0 mode with the newest kernel is:

```
Mar 11 23:28:58 Rukia kernel: RTW: ERROR _halmac_reg_write_32: I/O FAIL!
Mar 11 23:28:58 Rukia kernel: rtl88x2bu 3-2:1.0: Runtime PM usage count underflow!
Mar 11 23:28:58 Rukia kernel: RTW: WARN free_recv_skb_queue not empty, 8
Mar 11 23:28:58 Rukia kernel: usb 3-2: USB disconnect, device number 22
```

That's all I have. I will keep you posted if the issue persists after the next kernel update as well. I might be a bit slow to reply to any requests for additional information however as the driver still works in 2.0 mode without issue for me. I will try to be as much help as possible though.

Thanks!

RinCat, 2025-03-12:
Hi, thank you for the detail report. I just tested with 6.13.6 and USB 3.0 mode but unable to reproduce.
The error happened at `RTW: ERROR _halmac_reg_write_32: I/O FAIL!` cause by switching to USB 3.0 which will need to reconnect, thus you will see it disconnect and reconnect like this:

```
[ 2426.942722] usb 1-3: new high-speed USB device number 6 using xhci_hcd      // USB 2.0
[ 2427.068990] usb 1-3: New USB device found, idVendor=2357, idProduct=0138, bcdDevice= 2.10
[ 2427.068994] usb 1-3: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[ 2427.068996] usb 1-3: Product: 802.11ac NIC
[ 2427.068998] usb 1-3: Manufacturer: Realtek

...

[ 2429.054780] RTW: rtw_regsty_chk_target_tx_power_valid return _FALSE for band:0, path:0, rs:0, t:-1
[ 2429.064722] RTW: ERROR _halmac_reg_write_32: I/O FAIL!
[ 2429.064726] rtl88x2bu 1-3:1.0: Runtime PM usage count underflow!
[ 2429.064791] RTW: WARN free_recv_skb_queue not empty, 8
[ 2429.064912] usb 1-3: USB disconnect, device number 6
[ 2429.596746] usb 2-3: new SuperSpeed USB device number 5 using xhci_hcd      // USB 3.0
[ 2429.609083] usb 2-3: New USB device found, idVendor=2357, idProduct=0138, bcdDevice= 3.00
[ 2429.609086] usb 2-3: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[ 2429.609088] usb 2-3: Product: 802.11ac NIC
[ 2429.609090] usb 2-3: Manufacturer: Realtek
```

Do you mean you see
```
Mar 11 23:28:58 Rukia kernel: RTW: rtw_regsty_chk_target_tx_power_valid return _FALSE for band:0, path:0, rs:0, t:-1
Mar 11 23:28:58 Rukia kernel: RTW: ERROR _halmac_reg_write_32: I/O FAIL!
Mar 11 23:28:58 Rukia kernel: rtl88x2bu 3-2:1.0: Runtime PM usage count underflow!
Mar 11 23:28:58 Rukia kernel: RTW: WARN free_recv_skb_queue not empty, 8
Mar 11 23:28:58 Rukia kernel: usb 3-2: USB disconnect, device number 22
```
And the device never come back?

RAIACAD0564, 2025-03-13:
Okay, stranger still, I looked further through the journal for the past 7 days, and it appears that USB 3.0 mode continued to work after the kernel update, but it was actually a reboot following an update with mostly KDE packages oddly enough. Here is the full list. I looked between when it worked correctly and when the issue presented itself and these were the only packages that were updated or installed between that time. I don't appear to have installed or changed any packages myself between that time either:

`abseil-cpp-20250127.0-2 babl-0.1.112-1 bluedevil-1:6.3.3-1 breeze-6.3.3-1 breeze-gtk-6.3.3-1 chaotic-mirrorlist-20240724-3 chromium-134.0.6998.88-1 conky-1.22.0-2 discover-6.3.3-1 drkonqi1-6.3.3-1 faac-1.31.1-1 flatpak-kcm-6.3.3-1 gtest-1.16.0-1 kactivitymanagerd-6.3.3-1 kde-cli-tools-6.3.3-1 kde-gtk-config-6.3.3-1 kdecoration-6.3.3-1 kdeplasma-addons-6.3.3-1 kgamma-6.3.3-1 kglobalacceld-6.3.3-1 kinfocenter-6.3.3-1 kmenuedit-6.3.3-1 kpipewire-6.3.3-1 krdp-6.3.3-1 kscreen-6.3.3-1 kscreenlocker-6.3.3-1 ksshaskpass-6.3.3-1 ksystemstats-6.3.3-1 kwallet-pam-6.3.3-1 kwayland-6.3.3-1 kwayland-integration-6.3.3-1 kwin-6.3.3-1 kwrited-6.3.3-1 layer-shell-qt-6.3.3-1 lensfun-git-0.3.2.r3056.92f694bd5-1 lib32-libcap-2.75-1 libcamera-0.3.2-3 libcamera-ipa-0.3.2-3 libcap-2.75-1 libkscreen-6.3.3-1 libksysguard-6.3.3-1 libplasma-6.3.3-1 milou-6.3.3-1 nvm-0.40.2-1 ocean-sound-theme-6.3.3-1 oxygen-6.3.3-1 oxygen-sounds-6.3.3-1 pango-1:1.56.2-1 plasma-activities-6.3.3-1 plasma-activities-stats-6.3.3-1 plasma-browser-integration-6.3.3-1 plasma-desktop-6.3.3-1 plasma-disks-6.3.3-1 plasma-firewall-6.3.3-1 plasma-integration-6.3.3-1 plasma-thunderbolt-6.3.3-1 plasma-nm-6.3.3-1 plasma-pa-6.3.3-1 plasma-systemmonitor-6.3.3-1 plasma-vault-6-3.3-1 plasma-welcome-6.3.3-1 plasma-workspace-6.3.3-1 plasma-workspace-wallpapers-6.3.3-1 powerdevil-6.3.3-1 print-manager-1:6.3.3-1 plasma5support-6.3.3-1 polkit-kde-agent-6.3.3-1 qqc2-breeze-style-6.3.3-1 sddm-kcm-6.3.3-1 spectacle-1:6.3.3-1 systemsettings-6.3.3-1 usd-25.02.a-1 xdg-desktop-portal-kde-6.3.3-1`

The driver wasn't touched by the upgrade as I use the pkgbuild and manage it myself, however, I did attempt a rebuild after this update to see if it would solve the issue and it's unchanged, working in 2.0 only.

I also realized I did the kernel update during a previous boot and reboot, so that's not actually the cause of this strange behaviour as it worked fine until this update and boot in particular.

To answer your question, however, that is precisely the issue. It will disconnect as seen with the line `Mar 11 23:28:58 Rukia kernel: usb 3-2: USB disconnect, device number 22` and never come back.

This revelation makes things even more odd. I'm not sure what else might be causing it. Looking over the packages that updated before the issue occurred, none of them involve any USB/XHCI drivers, FUSE, kernel, or networkmanager/wireless packages. The only ones that would seem to change anything is bluedevil or KDE wrappers for nm, disks and other packages, which still seems unlikely. However, I will leave you this list and you can test for yourself if any of these are the culprit. I'm sorry I can't be of more help as the logs I sent you don't point towards what the actual problem package is.

RAIACAD0564, 2025-03-13:
Alright, I figured it out and apologize. It seems that the USB 3.0 extension cable has worn out/gotten frayed over time. I swapped the cable and tried it in different ports and it works now.

Sorry for the time waster haha, it was a classic gotcha. Thanks for all of your help though!

closed 2025-03-13 by RAIACAD0564