RinCat/RTL88x2BU-Linux-Driver #255: Build fails!

danergo, 2025-03-12:
I pulled the current code to my repo, and tried to make it, but it keeps failing, see below.

My kernel: 6.1.99
My OS: Debian

Log:
```
/home/pi/work/wifi/RTL88x2BU-Linux-Driver/core/rtw_mlme_ext.c: In function ‘rtw_txpwr_init_regd’:
/home/pi/work/wifi/RTL88x2BU-Linux-Driver/core/rtw_mlme_ext.c:299:6: error: this statement may fall through [-Werror=implicit-fallthrough=]
  299 |   if (rfctl->regd_name)
      |      ^
/home/pi/work/wifi/RTL88x2BU-Linux-Driver/core/rtw_mlme_ext.c:302:2: note: here
  302 |  default:
      |  ^~~~~~~
/home/pi/work/wifi/RTL88x2BU-Linux-Driver/core/rtw_mlme_ext.c: In function ‘mgt_dispatcher’:
/home/pi/work/wifi/RTL88x2BU-Linux-Driver/core/rtw_mlme_ext.c:1670:17: error: this statement may fall through [-Werror=implicit-fallthrough=]
 1670 |    ptable->func = &OnAuthClient;
      |    ~~~~~~~~~~~~~^~~~~~~~~~~~~~~
/home/pi/work/wifi/RTL88x2BU-Linux-Driver/core/rtw_mlme_ext.c:1672:2: note: here
 1672 |  case WIFI_ASSOCREQ:
      |  ^~~~
cc1: all warnings being treated as errors
make[2]: *** [scripts/Makefile.build:250: /home/pi/work/wifi/RTL88x2BU-Linux-Driver/core/rtw_mlme_ext.o] Error 1
make[1]: *** [Makefile:2053: /home/pi/work/wifi/RTL88x2BU-Linux-Driver] Error 2
make[1]: Leaving directory '/usr/src/linux-headers-6.1.99'
make: *** [Makefile:2470: modules] Error 2

```

danergo, 2025-03-12:
Had to enable

`EXTRA_CFLAGS += -Wimplicit-fallthrough=1` in `Makefile`

RinCat, 2025-03-12:
Hi, your system set warning as error, there is not much I can to fix that except you add the CFLAGS. The fallthroughs are different in different gcc / clang versions, and because the code needs to support a wider range of kernel versions, I cannot add it, as unrecognized compilers will report errors.

closed 2025-03-12 by danergo