RinCat/RTL88x2BU-Linux-Driver #256: Driver not loaded after compilation - TP-Link TX20UPlus - Kubuntu 24.04.2

pablocaron, 2025-03-17:
Hi RinCat,

I can't get the driver to load after compilation, either through Manual or DKMS install.
Below I copy some information

$ lsusb
Bus 003 Device 005: ID 2357:013f TP-Link 802.11ac WLAN Adapter

$ lsmod | grep r88
"empty output"

I compiled the driver, restarted the system but the driver is not loaded and I can't get the Tp-link modem working.

I would appreciate any help you can give me.

Regards

Pablo

RinCat, 2025-03-18:
Hi , that most likely due to secure boot where you cannot 3rd party kernel modules. You need either disable it or sign your own modules.

pablocaron, 2025-03-19:
Hi RinCat, thank you for your reply.
My notebook doesn't support SecureBoot (It's a Samsung Ultrabook from 2013). Bellow I copy the output of the commands that confirm this.

` $ mokutil --sb-state`
`This system doesn't support Secure Boot`

`$ sudo dmesg | grep -i secure`
`[sudo] password for pablo: `
`[    0.000000] secureboot: Secure boot disabled`
`[    0.023065] secureboot: Secure boot disabled`
`[    0.679654] Loaded X.509 cert 'Canonical Ltd. Secure Boot Signing: 61482aa2830d0ab2ad5af10b7250da9033ddcef0'`
`[    0.679683] Loaded X.509 cert 'Canonical Ltd. Secure Boot Signing (2017): 242ade75ac4a15e50d50c84b0d45ff3eae707a03'`
`[    0.679711] Loaded X.509 cert 'Canonical Ltd. Secure Boot Signing (ESM 2018): 365188c1d374d6b07c3c8f240f8ef722433d6a8b'`
`[    0.679739] Loaded X.509 cert 'Canonical Ltd. Secure Boot Signing (2019): c0746fd6c5da3ae827864651ad66ae47fe24b3e8'`
`[    0.679767] Loaded X.509 cert 'Canonical Ltd. Secure Boot Signing (2021 v1): a8d54bbb3825cfb94fa13c9f8a594a195c107b8d'`
`[    0.679795] Loaded X.509 cert 'Canonical Ltd. Secure Boot Signing (2021 v2): 4cf046892d6fd3c9a5b03f98d845f90851dc6a8c'`
`[    0.679825] Loaded X.509 cert 'Canonical Ltd. Secure Boot Signing (2021 v3): 100437bb6de6e469b581e61cd66bce3ef4ed53af'`
`[    0.679855] Loaded X.509 cert 'Canonical Ltd. Secure Boot Signing (Ubuntu Core 2019): c1d57b8f6b743f23ee41f4f7ee292f06eecadfb9'`

Perhaps I did something wrong during the compilation of the module?

Regards

Pablo

pablocaron, 2025-03-19:
Below I copy the output

`$ sudo modprobe 88xbu`
`modprobe: FATAL: Module 88xbu not found in directory /lib/modules/6.8.0-55-generic`

The `sudo dmesh`gives a very long output so I grep it with 88xbu
`$ sudo dmesg | grep 88xbu`
`empty output`

RinCat, 2025-03-19:
Did you run `sudo make install`? or you current kernel is not what you installed?

pablocaron, 2025-03-19:
In order to be sure, I repeated the installation from scratch

1. `make clean`
2. `make` the output is attached to this message
3. `sudo make install` the output is attached to this message
4. Restart the system

The module is still not loaded.
`$ sudo modprobe 88xbu`
`empty output`

I attached the output from the make commands. Perhaps you find the issue there. 

[log.tar.gz](https://github.com/user-attachments/files/19347750/log.tar.gz)

RinCat, 2025-03-19:
Both had no issues. What's the output of `uname -a`? The log shows it installed for `6.8.0-55-generic`.

pablocaron, 2025-03-19:
`$ uname -a`
`Linux woody 6.8.0-55-generic #57-Ubuntu SMP PREEMPT_DYNAMIC Wed Feb 12 23:42:21 UTC 2025 x86_64 x86_64 x86_64 GNU/Linux`

RinCat, 2025-03-19:
I would say check the dmesg logs, nothing seems wrong for all logs you provides.

RinCat, 2025-03-19:
Wait, Archer TX20U Plus? That's not this driver.

pablocaron, 2025-03-19:
After pluging in the modem in the usb port this is what I get from the dmesg

`[ 2882.742421] usb 3-1: new high-speed USB device number 4 using xhci_hcd`
`[ 2882.868825] usb 3-1: New USB device found, idVendor=0bda, idProduct=1a2b, bcdDevice= 0.00`
`[ 2882.868851] usb 3-1: New USB device strings: Mfr=1, Product=2, SerialNumber=0`
`[ 2882.868859] usb 3-1: Product: DISK`
`[ 2882.868865] usb 3-1: Manufacturer: Realtek`
`[ 2882.871209] usb-storage 3-1:1.0: USB Mass Storage device detected`
`[ 2882.871622] scsi host6: usb-storage 3-1:1.0`
`[ 2883.511605] usb 3-1: USB disconnect, device number 4`
`[ 2884.238418] usb 3-1: new high-speed USB device number 5 using xhci_hcd`
`[ 2884.364954] usb 3-1: New USB device found, idVendor=2357, idProduct=013f, bcdDevice= 0.00`
`[ 2884.364980] usb 3-1: New USB device strings: Mfr=1, Product=2, SerialNumber=3`
`[ 2884.364988] usb 3-1: Product: 802.11ac WLAN Adapter`
`[ 2884.364995] usb 3-1: Manufacturer: Realtek`
`[ 2884.365000] usb 3-1: SerialNumber: 00e04c000001`

I don't see any Error message or any issue. But I still can't see the modem in the tray icon.

pablocaron, 2025-03-19:
Oh no. This is so embarrassing. Please, forgive me.
I was trying several drivers and at some point I though that this driver was ok for my modem.

Again, apologies.

RinCat, 2025-03-19:
Although lsusb shows it is 802.11ac, it is actually 802.11ax. You need to use RTL8852AU driver.

pablocaron, 2025-03-19:
For further reference. The correct driver that is working is this
[https://github.com/lwfinger/rtl8852au](https://github.com/lwfinger/rtl8852au)