RinCat/RTL88x2BU-Linux-Driver #257: Driver usage issues

guoyanzhang0817, 2025-03-18:
I am using the Linux 6.1 kernel with the rt patch. Can it be adapted?
The chip I use is Realtek RTL8832BU. Can it be adapted?

RinCat, 2025-03-18:
Hi,  RTL8832BU is 802.11ax WiFi device and cannot be used with this driver. This one only support RTL8812BU and RTL8822BU.

guoyanzhang920817, 2025-03-19:
**"But if it doesn't work or is unstable"**

The introduction mentioned that the RTW88 that comes with Linux is unstable, but Linux 6.1 uses RTW89. I'm not sure what this instability you are talking about is? Has RTW89 solved this instability?  thanks.

RinCat, 2025-03-19:
> I'm not sure what this instability you are talking about is? 

You will get random disconnect and low speed.

I do not think it solved the instability. Also I do not think it support RTL8832BU neither.

closed 2025-03-24 by guoyanzhang0817