RinCat/RTL88x2BU-Linux-Driver #259: Error! Bad return status for module build on kernel: 6.12.20+rpt-rpi-2712 (aarch64)

cfuture81, 2025-04-02:
Hello

sadly the current debian / arm kernel update is failing:

Building module:
Cleaning build area...
'make' -j 4 KVER=6.12.20+rpt-rpi-2712 src=/usr/src/rtl88x2bu-5.8.7.1...........(bad exit status: 2)
Error! Bad return status for module build on kernel: 6.12.20+rpt-rpi-2712 (aarch64)
Consult /var/lib/dkms/rtl88x2bu/5.8.7.1/build/make.log for more information.
Error! One or more modules failed to install during autoinstall.
Refer to previous errors for more information.
dkms: autoinstall for kernel: 6.12.20+rpt-rpi-2712 failed!
run-parts: /etc/kernel/postinst.d/dkms exited with return code 11

RinCat, 2025-04-02:
Hi, you need post `/var/lib/dkms/rtl88x2bu/5.8.7.1/build/make.log` that actually show the error.

cfuture81, 2025-04-02:
cat /var/lib/dkms/rtl88x2bu/5.8.7.1/build/make.log

`
DKMS make.log for rtl88x2bu-5.8.7.1 for kernel 6.12.20+rpt-rpi-v8 (aarch64)
Wed Apr  2 05:15:11 PM CEST 2025
make ARCH=aarch64 CROSS_COMPILE= -C /lib/modules/6.12.20+rpt-rpi-v8/build M=/var/lib/dkms/rtl88x2bu/5.8.7.1/build  modules
make[1]: Entering directory '/usr/src/linux-headers-6.12.20+rpt-rpi-v8'
  CC [M]  /var/lib/dkms/rtl88x2bu/5.8.7.1/build/core/rtw_cmd.o
  CC [M]  /var/lib/dkms/rtl88x2bu/5.8.7.1/build/core/rtw_security.o
  CC [M]  /var/lib/dkms/rtl88x2bu/5.8.7.1/build/core/rtw_debug.o
  CC [M]  /var/lib/dkms/rtl88x2bu/5.8.7.1/build/core/rtw_io.o
/var/lib/dkms/rtl88x2bu/5.8.7.1/build/core/rtw_cmd.c:2071:6: warning: no previous prototype for ‘free_assoc_resources_hdl’ [-Wmissing-prototypes]
 2071 | void free_assoc_resources_hdl(_adapter *padapter, u8 lock_scanned_queue)
      |      ^~~~~~~~~~~~~~~~~~~~~~~~
/var/lib/dkms/rtl88x2bu/5.8.7.1/build/core/rtw_cmd.c:2234:4: warning: no previous prototype for ‘_rtw_set_chplan_cmd’ [-Wmissing-prototypes]
 2234 | u8 _rtw_set_chplan_cmd(_adapter *adapter, int flags, u8 chplan, const struct country_chplan *country_ent, u8 swconfig)
      |    ^~~~~~~~~~~~~~~~~~~
/var/lib/dkms/rtl88x2bu/5.8.7.1/build/core/rtw_cmd.c:2506:4: warning: no previous prototype for ‘rtw_ssmps_wk_hdl’ [-Wmissing-prototypes]
 2506 | u8 rtw_ssmps_wk_hdl(_adapter *adapter, struct ssmps_cmd_parm *ssmp_param)
      |    ^~~~~~~~~~~~~~~~
/var/lib/dkms/rtl88x2bu/5.8.7.1/build/core/rtw_cmd.c:3282:6: warning: no previous prototype for ‘rtw_dynamic_chk_wk_hdl’ [-Wmissing-prototypes]
 3282 | void rtw_dynamic_chk_wk_hdl(_adapter *padapter)
      |      ^~~~~~~~~~~~~~~~~~~~~~
/var/lib/dkms/rtl88x2bu/5.8.7.1/build/core/rtw_cmd.c:3327:6: warning: no previous prototype for ‘lps_ctrl_wk_hdl’ [-Wmissing-prototypes]
 3327 | void lps_ctrl_wk_hdl(_adapter *padapter, u8 lps_ctrl_type, u8 *buf)
      |      ^~~~~~~~~~~~~~~
/var/lib/dkms/rtl88x2bu/5.8.7.1/build/core/rtw_cmd.c:3506:6: warning: no previous prototype for ‘rtw_dm_in_lps_hdl’ [-Wmissing-prototypes]
 3506 | void rtw_dm_in_lps_hdl(_adapter *padapter)
      |      ^~~~~~~~~~~~~~~~~
/var/lib/dkms/rtl88x2bu/5.8.7.1/build/core/rtw_cmd.c:3547:6: warning: no previous prototype for ‘rtw_lps_change_dtim_hdl’ [-Wmissing-prototypes]
 3547 | void rtw_lps_change_dtim_hdl(_adapter *padapter, u8 dtim)
      |      ^~~~~~~~~~~~~~~~~~~~~~~
/var/lib/dkms/rtl88x2bu/5.8.7.1/build/core/rtw_cmd.c:3726:6: warning: no previous prototype for ‘rtw_dm_ra_mask_hdl’ [-Wmissing-prototypes]
 3726 | void rtw_dm_ra_mask_hdl(_adapter *padapter, struct sta_info *psta)
      |      ^~~~~~~~~~~~~~~~~~
/var/lib/dkms/rtl88x2bu/5.8.7.1/build/core/rtw_cmd.c:3768:6: warning: no previous prototype for ‘power_saving_wk_hdl’ [-Wmissing-prototypes]
 3768 | void power_saving_wk_hdl(_adapter *padapter)
      |      ^~~~~~~~~~~~~~~~~~~
/var/lib/dkms/rtl88x2bu/5.8.7.1/build/core/rtw_cmd.c:3774:6: warning: no previous prototype for ‘reset_securitypriv_hdl’ [-Wmissing-prototypes]
 3774 | void reset_securitypriv_hdl(_adapter *padapter)
      |      ^~~~~~~~~~~~~~~~~~~~~~
/var/lib/dkms/rtl88x2bu/5.8.7.1/build/core/rtw_cmd.c:4200:4: warning: no previous prototype for ‘rtw_dfs_rd_hdl’ [-Wmissing-prototypes]
 4200 | u8 rtw_dfs_rd_hdl(_adapter *adapter)
      |    ^~~~~~~~~~~~~~
/var/lib/dkms/rtl88x2bu/5.8.7.1/build/core/rtw_cmd.c:4591:6: warning: no previous prototype for ‘btinfo_evt_dump’ [-Wmissing-prototypes]
 4591 | void btinfo_evt_dump(void *sel, void *buf)
      |      ^~~~~~~~~~~~~~~
/var/lib/dkms/rtl88x2bu/5.8.7.1/build/core/rtw_cmd.c:5064:4: warning: no previous prototype for ‘rtw_c2h_wk_cmd’ [-Wmissing-prototypes]
 5064 | u8 rtw_c2h_wk_cmd(PADAPTER padapter, u8 *pbuf, u16 length, u8 type)
      |    ^~~~~~~~~~~~~~
/var/lib/dkms/rtl88x2bu/5.8.7.1/build/core/rtw_cmd.c:5204:4: warning: no previous prototype for ‘session_tracker_cmd’ [-Wmissing-prototypes]
 5204 | u8 session_tracker_cmd(_adapter *adapter, u8 cmd, struct sta_info *sta, u8 *local_naddr, u8 *local_port, u8 *remote_naddr, u8 *remote_port)
      |    ^~~~~~~~~~~~~~~~~~~
/var/lib/dkms/rtl88x2bu/5.8.7.1/build/core/rtw_cmd.c:5270:6: warning: no previous prototype for ‘session_tracker_chk_for_sta’ [-Wmissing-prototypes]
 5270 | void session_tracker_chk_for_sta(_adapter *adapter, struct sta_info *sta)
      |      ^~~~~~~~~~~~~~~~~~~~~~~~~~~
/var/lib/dkms/rtl88x2bu/5.8.7.1/build/core/rtw_cmd.c:5352:6: warning: no previous prototype for ‘session_tracker_chk_for_adapter’ [-Wmissing-prototypes]
 5352 | void session_tracker_chk_for_adapter(_adapter *adapter)
      |      ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
/var/lib/dkms/rtl88x2bu/5.8.7.1/build/core/rtw_cmd.c:5384:6: warning: no previous prototype for ‘session_tracker_cmd_hdl’ [-Wmissing-prototypes]
 5384 | void session_tracker_cmd_hdl(_adapter *adapter, struct st_cmd_parm *parm)
      |      ^~~~~~~~~~~~~~~~~~~~~~~
/var/lib/dkms/rtl88x2bu/5.8.7.1/build/core/rtw_cmd.c:5546:6: warning: no previous prototype for ‘rtw_ac_parm_cmd_hdl’ [-Wmissing-prototypes]
 5546 | void rtw_ac_parm_cmd_hdl(_adapter *padapter, u8 *_ac_parm_buf, int ac_type)
      |      ^~~~~~~~~~~~~~~~~~~
  CC [M]  /var/lib/dkms/rtl88x2bu/5.8.7.1/build/core/rtw_ioctl_query.o
  CC [M]  /var/lib/dkms/rtl88x2bu/5.8.7.1/build/core/rtw_ioctl_set.o
  CC [M]  /var/lib/dkms/rtl88x2bu/5.8.7.1/build/core/rtw_ieee80211.o
  CC [M]  /var/lib/dkms/rtl88x2bu/5.8.7.1/build/core/rtw_mlme.o
  CC [M]  /var/lib/dkms/rtl88x2bu/5.8.7.1/build/core/rtw_mlme_ext.o
[... 469 lines cut ...]
[... cut ...]
/usr/src/rtl88x2bu-5.8.7.1/include/usb_ops_linux.h:43:49: note: in definition of macro ‘usb_read_port_complete’
   43 | #define usb_read_port_complete(purb, regs)      usb_read_port_complete(purb)
      |                                                 ^~~~~~~~~~~~~~~~~~~~~~
make[2]: *** [/usr/src/linux-headers-6.12.20+rpt-common-rpi/Makefile:1969: /var/lib/dkms/rtl88x2bu/5.8.7.1/build] Error 2
make[1]: *** [/usr/src/linux-headers-6.12.20+rpt-common-rpi/Makefile:236: __sub-make] Error 2
make[1]: Leaving directory '/usr/src/linux-headers-6.12.20+rpt-rpi-v8'
make: *** [Makefile:2392: modules] Error 2
`

RinCat, 2025-04-02:
You are using 5.8.7.1? That a 5 years old version. Update the version should works.

cfuture81, 2025-04-02:
> You are using 5.8.7.1? That a 5 years old version. Update the version should works.

hi.  this was part of the system image for the raspberry.  so should I manually remove the current installed drivers with 
`sudo dkms remove -m  ..`

and clone down the repo and install fresh like described here?
https://askubuntu.com/questions/1288547/rtl88x2bu-driver-for-new-kernel-version

thanks a lot

RinCat, 2025-04-02:
Yes, you can also check the README

cfuture81, 2025-04-02:
> Yes, you can also check the README

Thanks. Worked like a charm. Thanks a lot