RinCat/RTL88x2BU-Linux-Driver #262: Bug Report: TP-Link Archer T4U v3 (2357:0115) Issues with 88x2bu Driver on Arch Linux

RayZ3R0, 2025-04-17:
## System Information
- **OS**: Arch Linux x86_64
- **Kernel**: 6.12.23-1-lts
- **USB Device**: TP-Link Archer T4U ver.3 (0x2357:0x0115)
- **Driver**: 88x2bu (rincat driver, loaded successfully)
- **Other hardware**: AMD Ryzen 5 5600G, B550MX/E PRO motherboard

## Issue Description
The TP-Link Archer T4U v3 WiFi adapter is recognized by the system and the 88x2bu driver loads successfully, but the interface is unstable, frequently becoming unavailable. When attempting to scan for wireless networks, the scan fails with either "scan aborted!" or "Cannot find device wlan0" errors. The adapter works correctly on Linux Mint 22 and Windows, confirming this is likely a driver integration issue with Arch Linux.

## Steps to Reproduce
1. Insert TP-Link Archer T4U v3 adapter into USB port
2. Load the 88x2bu driver (`sudo modprobe 88x2bu`)
3. Try to scan for networks (`sudo iw dev wlan0 scan`)
4. Observe scan failure or device unavailability

## Current Status
- Device is detected by USB subsystem
- 88x2bu driver loads successfully
- Network interface (wlan0) appears in `ip link show` output
- Interface MAC address matches the driver's assigned address
- Scanning fails with "scan aborted!" or "No such device" errors
- NetworkManager reports the device as "unavailable"

## Logs and Diagnostic Output

### USB Device Information
```
Bus 002 Device 002: ID 2357:0115 TP-Link Archer T4U ver.3
Negotiated speed: SuperSpeed (5Gbps)
```

### Driver Status
```
$ lsmod | grep 88x2
88x2bu               5132288  0
cfg80211             1396736  1 88x2bu
```

### Network Interfaces
```
$ ip link show
1: lo: <LOOPBACK,UP,LOWER_UP> mtu 65536 qdisc noqueue state UNKNOWN mode DEFAULT group default qlen 1000
    link/loopback 00:00:00:00:00:00 brd 00:00:00:00:00:00
2: enp6s0: <NO-CARRIER,BROADCAST,MULTICAST,UP> mtu 1500 qdisc fq_codel state DOWN mode DEFAULT group default qlen 1000
    link/ether f4:b5:20:56:c2:f5 brd ff:ff:ff:ff:ff:ff
    altname enxf4b52056c2f5
4: enp4s0f0u2: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc fq_codel state UNKNOWN mode DEFAULT group default qlen 1000
    link/ether 02:3e:0d:0e:75:7b brd ff:ff:ff:ff:ff:ff
    altname enx023e0d0e757b
6: wlan0: <NO-CARRIER,BROADCAST,MULTICAST,UP> mtu 1500 qdisc mq state DOWN mode DEFAULT group default qlen 1000
    link/ether ec:75:0c:e9:f3:89 brd ff:ff:ff:ff:ff:ff
```

### Wireless Interface Status
```
$ iwconfig wlan0
wlan0     unassociated  Nickname:"<WIFI@REALTEK>"
          Mode:Managed  Frequency=2.412 GHz  Access Point: Not-Associated   
          Sensitivity:0/0  
          Retry:off   RTS thr:off   Fragment thr:off
          Power Management:off
          Link Quality:0  Signal level:0  Noise level:0
          Rx invalid nwid:0  Rx invalid crypt:0  Rx invalid frag:0
          Tx excessive retries:0  Invalid misc:0   Missed beacon:0
```

### Scanning Failures
```
$ sudo iw dev wlan0 scan
scan aborted!

# Sometimes:
$ sudo iw dev wlan0 scan
Cannot find device "wlan0"
command failed: No such device (-19)
```

### Driver Log Excerpts
```
[  236.715082] RTW: rtw_macaddr_cfg mac addr:ec:75:0c:e9:f3:89
[  236.716245] RTW: rtw_ndev_init(wlan0) if1 mac_addr=ec:75:0c:e9:f3:89
[  236.716305] RTW: rtw_ndev_notifier_call(wlan0) state:17
[  236.716451] RTW: cfg80211_rtw_get_txpower(wlan0) total max: -10000 mbm
```

## Attempted Solutions
1. Cleaned up conflicting module configuration files
2. Created new configurations with various driver parameters:
   ```
   options 88x2bu rtw_power_mgnt=0 rtw_ips_mode=0 rtw_enusbss=0 rtw_drv_log_level=4
   ```
3. Added udev rule to manage device initialization
4. Disabled NetworkManager to test direct interface control
5. Tried various USB ports (both USB 3.0 and 2.0)
6. Blacklisted competing modules:
   ```
   blacklist rtw88_8822bu
   blacklist rtw88_8822b
   blacklist rtw88_usb
   ```

None of these solutions resolved the issue. Despite successful driver loading, the adapter remains unstable and fails to scan for networks.

## Additional Notes
- The device consistently initializes (logs show proper initialization)
- The MAC address is correctly assigned
- Power management settings are applied but don't solve the scanning issues
- The device intermittently disappears from the system
- Device works fine in other operating systems

## Requested Help
Any insights or suggestions for fixing these issues on Arch Linux would be greatly appreciated. The adapter works on other Linux distributions, suggesting compatibility with the Linux kernel, but something specific to the Arch environment or this kernel version is causing problems.

rennanfelipe, 2025-04-23:
Same problem here on this very own adapter version, I'm using Ubuntu 20.04.

RinCat, 2025-04-24:
This has been reported multiple times that in some system / hardware, scan cause it somehow offline / disconnect. I do not know the root cause of this. But maybe you can try iwd and see if it work?

RayZ3R0, 2025-04-24:
Iirc I did test with iwd, couldnt fix it. I actually fixed this by just going to the mainline kernel with iwd as the back end for NetworkManager, works fine now without any third party drivers

kintaris3, 2026-01-02:
Having the same issue on Nobara 43 with kernel 6.8. Also disabled the power saving measures for USB devices. Blacklisted all stock wireless drivers but the system just refuses to properly use the newly installed driver despite forcing it.