RinCat/RTL88x2BU-Linux-Driver #263: [Deepin 23] Error when compiling

LMS5413, 2025-04-21:
```
LMS@LMS-PC:~/RTL88x2BU-Linux-Driver$ make
make ARCH=x86_64 CROSS_COMPILE= -C /lib/modules/6.6.84-amd64-desktop-hwe/build M=/home/LMS/RTL88x2BU-Linux-Driver  modules
make[1]: Entrando no diretório '/usr/src/linux-headers-6.6.84-amd64-desktop-hwe'

warning: the compiler differs from the one used to build the kernel
  The kernel was built by: gcc (Deepin 12.3.0-17deepin8) 12.3.0
  You are using:           gcc (Deepin 12.3.0-17deepin12) 12.3.0
  CC [M]  /home/LMS/RTL88x2BU-Linux-Driver/core/rtw_cmd.o
  CC [M]  /home/LMS/RTL88x2BU-Linux-Driver/core/rtw_security.o
  CC [M]  /home/LMS/RTL88x2BU-Linux-Driver/core/rtw_debug.o
  CC [M]  /home/LMS/RTL88x2BU-Linux-Driver/core/rtw_io.o
  CC [M]  /home/LMS/RTL88x2BU-Linux-Driver/core/rtw_ioctl_query.o
  CC [M]  /home/LMS/RTL88x2BU-Linux-Driver/core/rtw_ioctl_set.o
  CC [M]  /home/LMS/RTL88x2BU-Linux-Driver/core/rtw_ieee80211.o
  CC [M]  /home/LMS/RTL88x2BU-Linux-Driver/core/rtw_mlme.o
  CC [M]  /home/LMS/RTL88x2BU-Linux-Driver/core/rtw_mlme_ext.o
/home/LMS/RTL88x2BU-Linux-Driver/core/rtw_mlme_ext.c: In function ‘rtw_txpwr_init_regd’:
/home/LMS/RTL88x2BU-Linux-Driver/core/rtw_mlme_ext.c:299:20: warning: this statement may fall through [-Wimplicit-fallthrough=]
  299 |                 if (rfctl->regd_name)
      |                    ^
/home/LMS/RTL88x2BU-Linux-Driver/core/rtw_mlme_ext.c:302:9: note: here
  302 |         default:
      |         ^~~~~~~
/home/LMS/RTL88x2BU-Linux-Driver/core/rtw_mlme_ext.c: In function ‘mgt_dispatcher’:
/home/LMS/RTL88x2BU-Linux-Driver/core/rtw_mlme_ext.c:1670:38: warning: this statement may fall through [-Wimplicit-fallthrough=]
 1670 |                         ptable->func = &OnAuthClient;
      |                         ~~~~~~~~~~~~~^~~~~~~~~~~~~~~
/home/LMS/RTL88x2BU-Linux-Driver/core/rtw_mlme_ext.c:1672:9: note: here
 1672 |         case WIFI_ASSOCREQ:
      |         ^~~~
  CC [M]  /home/LMS/RTL88x2BU-Linux-Driver/core/rtw_mi.o
  CC [M]  /home/LMS/RTL88x2BU-Linux-Driver/core/rtw_wlan_util.o
  CC [M]  /home/LMS/RTL88x2BU-Linux-Driver/core/rtw_vht.o
  CC [M]  /home/LMS/RTL88x2BU-Linux-Driver/core/rtw_pwrctrl.o
  CC [M]  /home/LMS/RTL88x2BU-Linux-Driver/core/rtw_rf.o
  CC [M]  /home/LMS/RTL88x2BU-Linux-Driver/core/rtw_chplan.o
  CC [M]  /home/LMS/RTL88x2BU-Linux-Driver/core/monitor/rtw_radiotap.o
  CC [M]  /home/LMS/RTL88x2BU-Linux-Driver/core/rtw_recv.o
/home/LMS/RTL88x2BU-Linux-Driver/core/rtw_recv.c: In function ‘validate_mp_recv_frame’:
/home/LMS/RTL88x2BU-Linux-Driver/core/rtw_recv.c:3842:25: warning: this ‘for’ clause does not guard... [-Wmisleading-indentation]
 3842 |                         for (i = 0; i < precv_frame->u.hdr.len; i = i + 8)
      |                         ^~~
In file included from /home/LMS/RTL88x2BU-Linux-Driver/include/drv_types.h:60,
                 from /home/LMS/RTL88x2BU-Linux-Driver/core/rtw_recv.c:17:
/home/LMS/RTL88x2BU-Linux-Driver/include/rtw_debug.h:167:9: note: ...this statement, but the latter is misleadingly indented as if it were guarded by the ‘for’
  167 |         do {\
      |         ^~
/home/LMS/RTL88x2BU-Linux-Driver/core/rtw_recv.c:3845:33: note: in expansion of macro ‘RTW_INFO’
 3845 |                                 RTW_INFO("#############################\n");
      |                                 ^~~~~~~~
  CC [M]  /home/LMS/RTL88x2BU-Linux-Driver/core/rtw_sta_mgt.o
/home/LMS/RTL88x2BU-Linux-Driver/core/rtw_sta_mgt.c: In function ‘rtw_mfree_stainfo’:
/home/LMS/RTL88x2BU-Linux-Driver/core/rtw_sta_mgt.c:391:27: warning: the comparison will always evaluate as ‘true’ for the address of ‘lock’ will never be NULL [-Waddress]
  391 |         if (&(psta->lock) != NULL)
      |                           ^~
In file included from /home/LMS/RTL88x2BU-Linux-Driver/include/drv_types.h:112,
                 from /home/LMS/RTL88x2BU-Linux-Driver/core/rtw_sta_mgt.c:17:
/home/LMS/RTL88x2BU-Linux-Driver/include/sta_info.h:267:17: note: ‘lock’ declared here
  267 |         _lock   lock;
      |                 ^~~~
  CC [M]  /home/LMS/RTL88x2BU-Linux-Driver/core/rtw_ap.o
  CC [M]  /home/LMS/RTL88x2BU-Linux-Driver/core/wds/rtw_wds.o
  CC [M]  /home/LMS/RTL88x2BU-Linux-Driver/core/mesh/rtw_mesh.o
  CC [M]  /home/LMS/RTL88x2BU-Linux-Driver/core/mesh/rtw_mesh_pathtbl.o
  CC [M]  /home/LMS/RTL88x2BU-Linux-Driver/core/mesh/rtw_mesh_hwmp.o
  CC [M]  /home/LMS/RTL88x2BU-Linux-Driver/core/rtw_xmit.o
  CC [M]  /home/LMS/RTL88x2BU-Linux-Driver/core/rtw_p2p.o
  CC [M]  /home/LMS/RTL88x2BU-Linux-Driver/core/rtw_rson.o
  CC [M]  /home/LMS/RTL88x2BU-Linux-Driver/core/rtw_tdls.o
  CC [M]  /home/LMS/RTL88x2BU-Linux-Driver/core/rtw_br_ext.o
  CC [M]  /home/LMS/RTL88x2BU-Linux-Driver/core/rtw_iol.o
  CC [M]  /home/LMS/RTL88x2BU-Linux-Driver/core/rtw_sreset.o
  CC [M]  /home/LMS/RTL88x2BU-Linux-Driver/core/rtw_btcoex_wifionly.o
  CC [M]  /home/LMS/RTL88x2BU-Linux-Driver/core/rtw_btcoex.o
  CC [M]  /home/LMS/RTL88x2BU-Linux-Driver/core/rtw_beamforming.o
  CC [M]  /home/LMS/RTL88x2BU-Linux-Driver/core/rtw_odm.o
  CC [M]  /home/LMS/RTL88x2BU-Linux-Driver/core/rtw_rm.o
  CC [M]  /home/LMS/RTL88x2BU-Linux-Driver/core/rtw_rm_fsm.o
  CC [M]  /home/LMS/RTL88x2BU-Linux-Driver/core/rtw_ft.o
  CC [M]  /home/LMS/RTL88x2BU-Linux-Driver/core/rtw_wnm.o
  CC [M]  /home/LMS/RTL88x2BU-Linux-Driver/core/rtw_mbo.o
  CC [M]  /home/LMS/RTL88x2BU-Linux-Driver/core/rtw_rm_util.o
  CC [M]  /home/LMS/RTL88x2BU-Linux-Driver/core/efuse/rtw_efuse.o
/home/LMS/RTL88x2BU-Linux-Driver/core/efuse/rtw_efuse.c: In function ‘rtw_efuse_analyze’:
/home/LMS/RTL88x2BU-Linux-Driver/core/efuse/rtw_efuse.c:971:17: warning: this ‘if’ clause does not guard... [-Wmisleading-indentation]
  971 |                 if (i % 16 == 0)
      |                 ^~
In file included from /home/LMS/RTL88x2BU-Linux-Driver/include/drv_types.h:60,
                 from /home/LMS/RTL88x2BU-Linux-Driver/core/efuse/rtw_efuse.c:17:
/home/LMS/RTL88x2BU-Linux-Driver/include/rtw_debug.h:261:9: note: ...this statement, but the latter is misleadingly indented as if it were guarded by the ‘if’
  261 |         do {\
      |         ^~
/home/LMS/RTL88x2BU-Linux-Driver/core/efuse/rtw_efuse.c:973:25: note: in expansion of macro ‘_RTW_PRINT_SEL’
  973 |                         _RTW_PRINT_SEL(RTW_DBGDUMP, "%02X%s"
      |                         ^~~~~~~~~~~~~~
  CC [M]  /home/LMS/RTL88x2BU-Linux-Driver/core/rtw_roch.o
  CC [M]  /home/LMS/RTL88x2BU-Linux-Driver/core/crypto/aes-internal.o
  CC [M]  /home/LMS/RTL88x2BU-Linux-Driver/core/crypto/aes-internal-enc.o
  CC [M]  /home/LMS/RTL88x2BU-Linux-Driver/core/crypto/aes-gcm.o
  CC [M]  /home/LMS/RTL88x2BU-Linux-Driver/core/crypto/aes-ccm.o
  CC [M]  /home/LMS/RTL88x2BU-Linux-Driver/core/crypto/aes-omac1.o
  CC [M]  /home/LMS/RTL88x2BU-Linux-Driver/core/crypto/ccmp.o
  CC [M]  /home/LMS/RTL88x2BU-Linux-Driver/core/crypto/gcmp.o
  CC [M]  /home/LMS/RTL88x2BU-Linux-Driver/core/crypto/aes-siv.o
  CC [M]  /home/LMS/RTL88x2BU-Linux-Driver/core/crypto/aes-ctr.o
  CC [M]  /home/LMS/RTL88x2BU-Linux-Driver/core/crypto/sha256-internal.o
  CC [M]  /home/LMS/RTL88x2BU-Linux-Driver/core/crypto/sha256.o
  CC [M]  /home/LMS/RTL88x2BU-Linux-Driver/core/crypto/sha256-prf.o
  CC [M]  /home/LMS/RTL88x2BU-Linux-Driver/core/crypto/rtw_crypto_wrap.o
  CC [M]  /home/LMS/RTL88x2BU-Linux-Driver/core/rtw_swcrypto.o
  CC [M]  /home/LMS/RTL88x2BU-Linux-Driver/os_dep/osdep_service.o
  CC [M]  /home/LMS/RTL88x2BU-Linux-Driver/os_dep/linux/os_intfs.o
  CC [M]  /home/LMS/RTL88x2BU-Linux-Driver/os_dep/linux/usb_intf.o
/home/LMS/RTL88x2BU-Linux-Driver/os_dep/linux/usb_intf.c:354:17: error: ‘struct usb_driver’ has no member named ‘drvwrap’
  354 |         .usbdrv.drvwrap.driver.shutdown = rtw_dev_shutdown,
      |                 ^~~~~~~
/home/LMS/RTL88x2BU-Linux-Driver/os_dep/linux/usb_intf.c:354:43: error: initialization of ‘const char *’ from incompatible pointer type ‘void (*)(struct device *)’ [-Werror=incompatible-pointer-types]
  354 |         .usbdrv.drvwrap.driver.shutdown = rtw_dev_shutdown,
      |                                           ^~~~~~~~~~~~~~~~
/home/LMS/RTL88x2BU-Linux-Driver/os_dep/linux/usb_intf.c:354:43: note: (near initialization for ‘usb_drv.usbdrv.name’)
cc1: some warnings being treated as errors
make[3]: *** [scripts/Makefile.build:243: /home/LMS/RTL88x2BU-Linux-Driver/os_dep/linux/usb_intf.o] Error 1
make[2]: *** [/usr/src/linux-headers-6.6.84-amd64-desktop-hwe/Makefile:1929: /home/LMS/RTL88x2BU-Linux-Driver] Error 2
make[1]: *** [Makefile:234: __sub-make] Error 2
make[1]: Saindo do diretório '/usr/src/linux-headers-6.6.84-amd64-desktop-hwe'

make: *** [Makefile:2470: modules] Error 2
```

RinCat, 2025-04-21:
Hi, change https://github.com/RinCat/RTL88x2BU-Linux-Driver/commit/7bdc911e1c14cac9448c3b9f68bf5392cc318849 `KERNEL_VERSION(6, 8, 0))` to `KERNEL_VERSION(6, 6, 0))`, it seems got back ported form new version.

LMS5413, 2025-04-21:
Hey! its worked, but i when executed "make install" and reboot my pc, doesnt work, you think if have deepin issue?

RinCat, 2025-04-21:
run `sudo modprobe 88x2bu` and see results and `dmesg` output.

LMS5413, 2025-04-21:
```
LMS@LMS-PC:~/RTL88x2BU-Linux-Driver$ sudo modprobe 88x2bu && dmesg --ctime | grep -iE '88x2bu|rtw|wlan|usbcore'
[seg abr 21 19:38:19 2025] usbcore: registered new interface driver usbfs
[seg abr 21 19:38:19 2025] usbcore: registered new interface driver hub
[seg abr 21 19:38:19 2025] usbcore: registered new device driver usb
[seg abr 21 19:38:19 2025] usbcore: registered new interface driver bcm5974
[seg abr 21 19:38:21 2025] usb 3-1: Product: 802.11ac WLAN Adapter
[seg abr 21 19:38:22 2025] usbcore: registered new interface driver usbhid
[seg abr 21 19:38:22 2025] usbcore: registered new interface driver usbkbd
[seg abr 21 19:38:22 2025] usbcore: registered new interface driver usbmouse
[seg abr 21 19:38:24 2025] usbcore: registered new interface driver rtl8812au
[seg abr 21 19:39:10 2025] usbcore: registered new interface driver snd-usb-audio
[seg abr 21 19:40:16 2025] usbcore: registered new interface driver cdc_ether
[seg abr 21 19:40:16 2025] usbcore: registered new interface driver rndis_host
[seg abr 21 19:44:44 2025] RTW: module init start
[seg abr 21 19:44:44 2025] RTW: rtl88x2bu v5.13.1-30-g37e60b26a.20220819_COEX20220812-18317b7b
[seg abr 21 19:44:44 2025] RTW: rtl88x2bu BT-Coex version = COEX20220812-18317b7b
[seg abr 21 19:44:44 2025] usbcore: registered new interface driver rtl88x2bu
[seg abr 21 19:44:44 2025] RTW: module init ret=0
```

RinCat, 2025-04-21:
`0bda:8832` is not 8812bu or 8822bu. You need a different driver. The lsusb result is wrong,  it is a Realtek 802.11a**x** device.

LMS5413, 2025-04-21:
Oh... Do you know any other drivers? When compatible for this?

RinCat, 2025-04-21:
Maybe this? https://github.com/lwfinger/rtl8852bu

LMS5413, 2025-04-22:
I did it! I used this repository https://github.com/lwfinger/rtl8852au and made the same modification suggested, thank you very much

closed 2025-04-22 by LMS5413