RinCat/RTL88x2BU-Linux-Driver #264: Enhancement: Latest DKMS deprecated feature: CLEAN

RAIACAD0564, 2025-04-27:
I titled this as an enhancement as it's not particularly an error, at least yet.

In the last two kernel updates, DKMS builds the driver successfully, however, throws this warning:

```
==> dkms install --no-depmod rtl88x2bu/5.13.1.r204.fda90b7 -k 6.14.4-arch1-1
Deprecated feature: CLEAN (/usr/src/rtl88x2bu-5.13.1.r204.fda90b7/dkms.conf)
```

The now deprecated line in said file is:
`CLEAN="make clean"`

My version of DKMS and kernel (headers) are as follows:

linux-headers 6.14.4.arch1-1
dkms 3.1.8-1

I thought I'd mention this as the driver still builds successfully, however, now that CLEAN is deprecated, I'm not sure how long that will last. 

Thanks!

johnpiers, 2025-05-08:
It all appeared to update correctly, so I'm presuming there is nothing to worry about? It's the first time I've seen it on my Arch install. Will removing line 8:  CLEAN="'make' clean" in those particular dkms.confs cause any problems or is it the solution to clearing the message? Thank you. 

dkms install --no-depmod rtl8821cu/5.12.0.4.202.d74134a -k 6.14.5-arch1-1 Deprecated feature: CLEAN (/usr/src/rtl8821cu-5.12.0.4.202.d74134a/dkms.conf) and a similar one for the Zen Kernel with a different path: Deprecated feature: CLEAN (/var/lib/dkms/rtl8821cu/5.12.0.4.202.d74134a/source/dkms.conf)

RinCat, 2025-05-09:
OK, it is safe to just drop it.

https://github.com/RinCat/RTL88x2BU-Linux-Driver/commit/9624bd8943dc9998fc09355ea725ef7d59f80cb2

closed 2025-05-09 by RinCat